Ysurac/openmptcprouter #2887: omr port forwarding broken

Kalimeiro, 2023-07-19:
port forwarding is totaly broken on latest snapshot since 18/07/2023.

## Specifications

  - OpenMPTCProuter version: 0.59.2 alpha snapshot release 18/07/2023
  - OpenMPTCProuter VPS version: 0.1029-test snapshot 18/07/2023
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI4 - x86_64 - ...
  - Country:  FR

Kalimeiro, 2023-07-19:
from tcpdump on the router part, i can see incoming packets, but no output packets, the connection is in looping mode and my application wait for the connection and never come :

13:16:47.205115 IP 123.123.123.123.10543 > 10.255.255.2.10080: Flags [S], seq 2806557094, win 42340, options [mss 1460,sackOK,TS val 2485784215 ecr 0,nop,wscale 11,mptcp capable {0x81b89fd3f0a0bfda}], length 0
13:16:47.301113 IP 123.123.123.123.10553 > 10.255.255.2.10080: Flags [S], seq 2202596940, win 42340, options [mss 1460,sackOK,TS val 2485784311 ecr 0,nop,wscale 11,mptcp capable {0xcbfc7cfe60f9d580}], length 0
13:16:47.301167 IP 123.123.123.123.10555 > 10.255.255.2.10080: Flags [S], seq 2745540026, win 42340, options [mss 1460,sackOK,TS val 2485784311 ecr 0,nop,wscale 11,mptcp capable {0x6fad091fcc7669d}], length 0
13:16:47.333138 IP 123.123.123.123.10559 > 10.255.255.2.10080: Flags [S], seq 720396653, win 42340, options [mss 1460,sackOK,TS val 2485784343 ecr 0,nop,wscale 11,mptcp capable {0x6a74757710adb4fa}], length 0
13:16:51.461278 IP 123.123.123.123.10555 > 10.255.255.2.10080: Flags [S], seq 2745540026, win 42340, options [mss 1460,sackOK,TS val 2485788471 ecr 0,nop,wscale 11,mptcp capable {0x6fad091fcc7669d}], length 0
13:16:51.461373 IP 123.123.123.123.10559 > 10.255.255.2.10080: Flags [S], seq 720396653, win 42340, options [mss 1460,sackOK,TS val 2485788471 ecr 0,nop,wscale 11,mptcp capable {0x6a74757710adb4fa}], length 0
13:16:51.461410 IP 123.123.123.123.10553 > 10.255.255.2.10080: Flags [S], seq 2202596940, win 42340, options [mss 1460,sackOK,TS val 2485788471 ecr 0,nop,wscale 11,mptcp capable {0xcbfc7cfe60f9d580}], length 0
13:16:51.461445 IP 123.123.123.123.10543 > 10.255.255.2.10080: Flags [S], seq 2806557094, win 42340, options [mss 1460,sackOK,TS val 2485788471 ecr 0,nop,wscale 11,mptcp capable {0x81b89fd3f0a0bfda}], length 0
13:16:56.634721 IP 10.255.255.1 > 10.255.255.2: ICMP echo request, id 50786, seq 1, length 64
13:16:56.634807 IP 10.255.255.2 > 10.255.255.1: ICMP echo reply, id 50786, seq 1, length 64
13:16:57.636448 IP 10.255.255.1 > 10.255.255.2: ICMP echo request, id 50786, seq 2, length 64
13:16:57.636516 IP 10.255.255.2 > 10.255.255.1: ICMP echo reply, id 50786, seq 2, length 64
13:16:58.637955 IP 10.255.255.1 > 10.255.255.2: ICMP echo request, id 50786, seq 3, length 64
13:16:58.638017 IP 10.255.255.2 > 10.255.255.1: ICMP echo reply, id 50786, seq 3, length 64
13:16:59.640125 IP 10.255.255.1 > 10.255.255.2: ICMP echo request, id 50786, seq 4, length 64
13:16:59.640219 IP 10.255.255.2 > 10.255.255.1: ICMP echo reply, id 50786, seq 4, length 64
13:16:59.653249 IP 123.123.123.123.10543 > 10.255.255.2.10080: Flags [S], seq 2806557094, win 42340, options [mss 1460,sackOK,TS val 2485796663 ecr 0,nop,wscale 11,mptcp capable {0x81b89fd3f0a0bfda}], length 0
13:16:59.653324 IP 123.123.123.123.10553 > 10.255.255.2.10080: Flags [S], seq 2202596940, win 42340, options [mss 1460,sackOK,TS val 2485796663 ecr 0,nop,wscale 11,mptcp capable {0xcbfc7cfe60f9d580}], length 0
13:16:59.653363 IP 123.123.123.123.10559 > 10.255.255.2.10080: Flags [S], seq 720396653, win 42340, options [mss 1460,sackOK,TS val 2485796663 ecr 0,nop,wscale 11,mptcp capable {0x6a74757710adb4fa}], length 0
13:16:59.653400 IP 123.123.123.123.10555 > 10.255.255.2.10080: Flags [S], seq 2745540026, win 42340, options [mss 1460,sackOK,TS val 2485796663 ecr 0,nop,wscale 11,mptcp capable {0x6fad091fcc7669d}], length 0
13:17:00.372605 IP 123.123.123.123.14707 > 10.255.255.2.10080: Flags [S], seq 407518371, win 42340, options [mss 1460,sackOK,TS val 2485797382 ecr 0,nop,wscale 11,mptcp capable {0x98dab0826c1b3c1e}], length 0
13:17:00.457479 IP 123.123.123.123.14709 > 10.255.255.2.10080: Flags [S], seq 3048979734, win 42340, options [mss 1460,sackOK,TS val 2485797467 ecr 0,nop,wscale 11,mptcp capable {0x5a811cec584a537}], length 0
13:17:00.463056 IP 123.123.123.123.14713 > 10.255.255.2.10080: Flags [S], seq 2286143285, win 42340, options [mss 1460,sackOK,TS val 2485797473 ecr 0,nop,wscale 11,mptcp capable {0xe566f902747bdbef}], length 0
13:17:00.510466 IP 123.123.123.123.14729 > 10.255.255.2.10080: Flags [S], seq 1670794823, win 42340, options [mss 1460,sackOK,TS val 2485797520 ecr 0,nop,wscale 11,mptcp capable {0x55089d9bdc89dc72}], length 0
13:17:00.641461 IP 10.255.255.1 > 10.255.255.2: ICMP echo request, id 50786, seq 5, length 64
13:17:00.641531 IP 10.255.255.2 > 10.255.255.1: ICMP echo reply, id 50786, seq 5, length 64
13:17:01.381203 IP 123.123.123.123.14707 > 10.255.255.2.10080: Flags [S], seq 407518371, win 42340, options [mss 1460,sackOK,TS val 2485798391 ecr 0,nop,wscale 11,mptcp capable {0x98dab0826c1b3c1e}], length 0
13:17:01.477014 IP 123.123.123.123.14709 > 10.255.255.2.10080: Flags [S], seq 3048979734, win 42340, options [mss 1460,sackOK,TS val 2485798487 ecr 0,nop,wscale 11,mptcp capable {0x5a811cec584a537}], length 0
13:17:01.477073 IP 123.123.123.123.14713 > 10.255.255.2.10080: Flags [S], seq 2286143285, win 42340, options [mss 1460,sackOK,TS val 2485798487 ecr 0,nop,wscale 11,mptcp capable {0xe566f902747bdbef}], length 0
13:17:01.541256 IP 123.123.123.123.14729 > 10.255.255.2.10080: Flags [S], seq 1670794823, win 42340, options [mss 1460,sackOK,TS val 2485798551 ecr 0,nop,wscale 11,mptcp capable {0x55089d9bdc89dc72}], length 0
13:17:03.397054 IP 123.123.123.123.14707 > 10.255.255.2.10080: Flags [S], seq 407518371, win 42340, options [mss 1460,sackOK,TS val 2485800407 ecr 0,nop,wscale 11,mptcp capable {0x98dab0826c1b3c1e}], l [... cut ...]
13:17:07.589558 IP 123.123.123.123.14713 > 10.255.255.2.10080: Flags [S], seq 2286143285, win 42340, options [mss 1460,sackOK,TS val 2485804599 ecr 0,nop,wscale 11,mptcp capable {0xe566f902747bdbef}], length 0
13:17:07.589609 IP 123.123.123.123.14709 > 10.255.255.2.10080: Flags [S], seq 3048979734, win 42340, options [mss 1460,sackOK,TS val 2485804599 ecr 0,nop,wscale 11,mptcp capable {0x5a811cec584a537}], length 0
13:17:07.589648 IP 123.123.123.123.14707 > 10.255.255.2.10080: Flags [S], seq 407518371, win 42340, options [mss 1460,sackOK,TS val 2485804599 ecr 0,nop,wscale 11,mptcp capable {0x98dab0826c1b3c1e}], length 0

Kalimeiro, 2023-07-19:
```
qdisc noqueue 0: dev lo root refcnt 2
qdisc mq 0: dev eth0 root
qdisc fq_codel 0: dev eth0 parent :5 limit 10240p flows 1024 quantum 1514 target 5ms interval 100ms memory_limit 4Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth0 parent :4 limit 10240p flows 1024 quantum 1514 target 5ms interval 100ms memory_limit 4Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth0 parent :3 limit 10240p flows 1024 quantum 1514 target 5ms interval 100ms memory_limit 4Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth0 parent :2 limit 10240p flows 1024 quantum 1514 target 5ms interval 100ms memory_limit 4Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth0 parent :1 limit 10240p flows 1024 quantum 1514 target 5ms interval 100ms memory_limit 4Mb ecn drop_batch 64
qdisc noqueue 0: dev eth0.6 root refcnt 2
qdisc noqueue 0: dev wan1 root refcnt 2
qdisc noqueue 0: dev eth0.5 root refcnt 2
qdisc noqueue 0: dev wan2 root refcnt 2
qdisc noqueue 0: dev eth0.4 root refcnt 2
qdisc noqueue 0: dev wan3 root refcnt 2
qdisc noqueue 0: dev eth0.8 root refcnt 2
qdisc fq_codel 0: dev tun0 root refcnt 2 limit 10240p flows 1024 quantum 1500 target 5ms interval 100ms memory_limit 4Mb ecn drop_batch 64

```

Ysurac, 2023-07-19:
I'm not able to reproduce the issue.
What is the result of ```uci show firewall``` on the router via SSH ?

Kalimeiro, 2023-07-19:
i send you the content by mail.

Ysurac, 2023-07-20:
As it's working for me with MTU 9000, I think the problem is elsewhere.
I think this is due to the fact that source IP is set, I will fix that asap.

Emtek-Industrial, 2023-07-21:
Same happened to me 48 hours ago. I was setting up port-forwarding for my CCTV camera's. Everything was working fine with the initial setup but then I decided to use different ports and after changeing the ports it stopped forwarding all ports. Its a relativly new installation so easier for me to wipe down and re-install. Ysurac, if it helps, I'm happy to give you access to my system if you are still unable to reproduce the situation.

Ysurac, 2023-07-21:
The problem here is only on Snaspshot images and if a source IP is set. This will be fixed today.

Kalimeiro, 2023-07-23:
After testing the latest snapshot, the problem didn't solved.

Ysurac, 2023-07-23:
In fact, I fixed an error when a source destination IP is set, in your case it's source IP of packets.
For now I'm not able to reproduce the issue, still working on that.

Kalimeiro, 2023-07-23:
Désolé de te contredire, mais tu ne test pas correctement/assez tes push. Je ne suis pas un cas isolé.

closed 2023-07-23 by Kalimeiro

Ysurac, 2023-07-23:
I don't understand what you mean here... I push something that was not made to fix your problem so it's the case...

Kalimeiro, 2023-07-23:
On va en discuter en dehors de github.