Ysurac/openmptcprouter #2889: le noyau 6.1 ne permet pas l'agrégation

Kalimeiro, 2023-07-19:
Bonjour Ysurac,

J'écris en Français afin de pouvoir m'exprimer clairement.

J'essaie, en vain, de faire fonctionner OMR (routeur et VPS) avec le noyau 6.1 avec l'agrégation, dont tu as affirmé que ça fonctionne, et ça fait des mois que je fais des essais de temps en temps afin de simplement "vérifier" que ça fonctionne, alors voilà quelques captures d'écrans, qui montre tout l'inverse :

[image]

[image]

que ce soit en mptcp version 0 ou 1, ça ne fonctionne pas, seul la liaison maître est utilisé.

lorsque la liaison maître est débranché, je n'ai juste plus internet du tout.

quels sont les recommandations pour utiliser le noyau 6.1 et que les liens soient agrégés ?

Ysurac, 2023-07-19:
Est-ce qu'il s'agit d'une mise à jour ou d'une nouvelle installation ? Le VPS tourne sous quel OS/version ?

Kalimeiro, 2023-07-19:
J'ai essayé les deux méthodes : mise à jour, ça ne fonctionne pas. J'ai donc tout réinstallé à zéro, pareil.

le VPS est sous Debian, j'ai essayé avec la 10/11 et actuellement je suis sur la 12.

Ysurac, 2023-07-19:
Que donne, via SSH sur le routeur, ```ip mptcp endpoint``` ? Même chose côté VPS.

Kalimeiro, 2023-07-19:
sur le routeur :

192.168.10.253 id 10 subflow fullmesh dev eth1
192.168.11.253 id 11 subflow fullmesh dev eth4
192.168.12.253 id 12 subflow fullmesh dev eth3

sur le vps : 

51.xxx.xxx.xxx id 1 signal dev eth0

Ysurac, 2023-07-19:
Tout ça est bon.
Quel est le contenu de ```ip r``` ? Le proxy est Shadowsocks ?

Kalimeiro, 2023-07-19:
le proxy est v2ray, je vais essayer shadowsocks.

le contenu de "ip r" : 

```
default via 10.255.254.1 dev tun0 initcwnd 10 initrwnd 10
default metric 1
        nexthop via 192.168.10.254 dev eth1 weight 100
        nexthop via 192.168.11.254 dev eth4 weight 1
        nexthop via 192.168.12.254 dev eth3 weight 1
default via 192.168.10.254 dev eth1 metric 3 initcwnd 10 initrwnd 10
default via 192.168.12.254 dev eth3 metric 4 initcwnd 10 initrwnd 10
default via 192.168.11.254 dev eth4 metric 10 initcwnd 10 initrwnd 10
default via 10.255.254.1 dev tun0 metric 1200 initcwnd 10 initrwnd 10
10.255.254.1 dev tun0 proto kernel scope link src 10.255.254.2
10.255.254.2 dev tun0 scope link metric 1200 initcwnd 10 initrwnd 10
51.77.220.248 metric 1
        nexthop via 192.168.10.254 dev eth1 weight 100
        nexthop via 192.168.11.254 dev eth4 weight 1
        nexthop via 192.168.12.254 dev eth3 weight 1
127.0.0.0/8 dev lo proto static scope link metric 8
192.168.10.0/24 dev eth1 scope link metric 3 initcwnd 10 initrwnd 10
192.168.11.0/24 dev eth4 scope link metric 10 initcwnd 10 initrwnd 10
192.168.12.0/24 dev eth3 scope link metric 4 initcwnd 10 initrwnd 10

```

Kalimeiro, 2023-07-23:
Je viens de tester avec ShadowSocks au lieu de V2RAY et je confirme que l'agrégation fonctionne bien avec le noyau 6.1

Donc, V2RAY n'est pas compatible, ce qui est dommage car il est bien meilleur en tout point que shadowsocks (débit, latence, support de UDP)

Kalimeiro, 2023-07-23:
Par contre, il manque BBR2 sur le noyau 6.1.

Kalimeiro, 2023-07-23:
Aussi pour information j'ai testé en désactivant le proxy afin que tout passe par Glorytun qui est configuré en UDP chez moi, l'agrégation fonctionne parfaitement ! par contre Glorytun est à la ramasse en performance, je ne dépasse pas les 350Mbps en téléchargement et ~150Mbps en émission.

Kalimeiro, 2023-07-23:
Dernier test effectué : j'ai 3 connexions dont 2 fibres et une 4G LTE, j'ai du désactiver la 4G LTE qui est moins bien que les connexions fibres, simplement parce qu'il y a un problème avec la gestion de la congestion et le débit est à la ramasse malgré qu'il soit bien agrégé. Je suis preneur d'info sur ce sujet (BBR/CUBIC/?)

closed 2023-07-23 by Kalimeiro