Ysurac/openmptcprouter #2891: No interface aggregation on Gateworks Ventana

ThisIsNotAnAvailableUsername, 2023-07-21:
## Expected Behavior

The communication should be split between all of the 4 cellular interfaces.
## Current Behavior

Upon checking the bandwidth (under Network -> MPTCP), and examining tcpdump, it seems that only one interface is actually used (the one configured to be MPTCP master).
The following line appears repeatedly in the system logs:

> user.notice post-tracking-post-tracking: Can't find omr-tracker-ss, restart omr-tracker...

The Router was configured automatically using the server key.
## Specifications

  - OpenMPTCProuter version: 0.59.2 alpha
  - OpenMPTCProuter VPS version: 0.1029-test
  - OpenMPTCProuter VPS provider: on-prem
  - OpenMPTCProuter platform: Gateworks Ventana imx6 armv7l

Ysurac, 2023-07-21:
All is green in the status page ? Shadowsocks is used as proxy ?
It's a custom compilation ?

JohnTar22, 2023-07-21:
Hi, It's something we work on together.
Yes, all green in status page and shadowsocks is used as proxy.
It's a custom compilation exactly as described in the wiki for unsupported platforms.
Thanks!

JohnTar22, 2023-07-21:
The lab setup now is: 
Gateworks Ventana GW5410 with Quecetel EC25 as Master LTE Modem. 
Another USB to ethernet adapter connected to Teltonika RUT955 LTE Router.
Another Teltonika RUT240 connected to a free ethernet port on board.

The desired setup is:
6 Mini Pcie LTE modems on board working together with a variety of sim cards from many ISP's.

Ysurac, 2023-07-22:
What is the result of ```uci show openmptcprouter``` ?

JohnTar22, 2023-07-22:
openmptcprouter.settings=settings
openmptcprouter.settings.enabled='1'
openmptcprouter.settings.sipalg='1'
openmptcprouter.settings.master='balancing'
openmptcprouter.settings.disable_ipv6='1'
openmptcprouter.settings.check_ipv4_website='http://ip.openmptcprouter.com'
openmptcprouter.settings.check_ipv6_website='http://ipv6.openmptcprouter.com'
openmptcprouter.settings.status_vps_timeout='1'
openmptcprouter.settings.status_getip_timeout='1'
openmptcprouter.settings.enable_nodelay='1'
openmptcprouter.settings.scaling_governor='performance'
openmptcprouter.settings.menu='OpenMPTCProuter'
openmptcprouter.settings.disableintfrename='1'
openmptcprouter.settings.tracebox='0'
openmptcprouter.settings.version='-deelop-75694cfc'
openmptcprouter.settings.allmptcpovervpn='0'
openmptcprouter.settings.serial='0000000000000000'
openmptcprouter.settings.mptcpovervpn='wireguard'
openmptcprouter.settings.dns64='0'
openmptcprouter.settings.proxy='shadowsocks'
openmptcprouter.settings.ha='0'
openmptcprouter.settings.vpn='glorytun_tcp'
openmptcprouter.settings.firstboot='0'
openmptcprouter.settings.apilc='1690042643'
openmptcprouter.settings.country='world'
openmptcprouter.vps=server
openmptcprouter.vps.username='openmptcprouter'
openmptcprouter.vps.master='1'
openmptcprouter.vps.backup='0'
openmptcprouter.vps.port='65500'
openmptcprouter.vps.set_firewall='0'
openmptcprouter.vps.get_config='0'
openmptcprouter.vps.current='1'
openmptcprouter.vps.password='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
openmptcprouter.vps.admin_error='0'
openmptcprouter.vps.redirect_ports='0'
openmptcprouter.vps.wgkey='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
openmptcprouter.vps.pihole='0'
openmptcprouter.vps.kernel='6.1.0-10-amd64'
openmptcprouter.vps.machine='x86_64'
openmptcprouter.vps.omr_version='0.1029-test'
openmptcprouter.vps.token_error='1'
openmptcprouter.vps.lastchange='1690042649'
openmptcprouter.vps.token='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
openmptcprouter.vps.ip='xx.xx.xx.xx'
openmptcprouter.vps.available_vpn='glorytun_tcp' 'glorytun_udp' 'dsvpn' 'openvpn' 'openvpn_bonding' 'mlvpn'
openmptcprouter.omr=router
openmptcprouter.omr.ss_sss0='up'
openmptcprouter.omr.detected_public_ipv4='xx.xx.xx.xx'
openmptcprouter.omr.detected_ss_ipv4='xx.xx.xx.xx'
openmptcprouter.wan1=interface
openmptcprouter.wan1.multipath='on'
openmptcprouter.wan1.metric='3'
openmptcprouter.wan1.multipathvpn='0'
openmptcprouter.wan1.state='up'
openmptcprouter.wan1.publicip='xxx.xxx.xxx.xxx'
openmptcprouter.wan1.latency='45'
openmptcprouter.wan1.asn='FIXED-IP-ADDRESS-FOR-RSINGEL-APPLICATION'
openmptcprouter.wan1.mtu='1500'
openmptcprouter.wan1.mptcp_status='MPTCP enabled'
openmptcprouter.wan1.latency_previous='35'
openmptcprouter.wan1.lc='1690046119'
openmptcprouter.wan2=interface
openmptcprouter.wan2.multipath='on'
openmptcprouter.wan2.metric='4'
openmptcprouter.wan2.multipathvpn='0'
openmptcprouter.wan2.state='up'
openmptcprouter.wan2.publicip='xxx.xxx.xxx.xxx'
openmptcprouter.wan2.latency='52'
openmptcprouter.wan2.asn='FIXED-IP-ADDRESS-FOR-RSINGEL-APPLICATION'
openmptcprouter.wan2.mtu='1500'
openmptcprouter.wan2.mptcp_status='MPTCP enabled'
openmptcprouter.wan2.manufacturer='huawei'
openmptcprouter.wan2.latency_previous='49'
openmptcprouter.wan2.lc='1690046206'
openmptcprouter.loopback=interface
openmptcprouter.loopback.multipath='off'
openmptcprouter.loopback.metric='5'
openmptcprouter.lan=interface
openmptcprouter.lan.multipath='off'
openmptcprouter.lan.metric='6'
openmptcprouter.lan.multipathvpn='0'
openmptcprouter.omrvpn=interface
openmptcprouter.omrvpn.multipath='off'
openmptcprouter.omrvpn.metric='1200'
openmptcprouter.omr6in4=interface
openmptcprouter.omr6in4.multipath='off'
openmptcprouter.omr6in4.metric='1201'
openmptcprouter.latest_versions=latest_versions
openmptcprouter.latest_versions.omr='0.59.1-5.15'
openmptcprouter.latest_versions.vps='0.1028'
openmptcprouter.latest_versions.lc='1690042641'
openmptcprouter.wan3=interface
openmptcprouter.wan3.multipath='master'
openmptcprouter.wan3.metric='7'
openmptcprouter.wan3.multipathvpn='0'
openmptcprouter.wan3.state='up'
openmptcprouter.wan3.publicip='xxx.xxx.xxx.xxx'
openmptcprouter.wan3.latency='34'
openmptcprouter.wan3.asn='xxxxxxxxx'
openmptcprouter.wan3.mtu='1500'
openmptcprouter.wan3.mptcp_status='MPTCP enabled'
openmptcprouter.wan3.manufacturer='huawei'
openmptcprouter.wan3.latency_previous='33'
openmptcprouter.wan3.lc='1690046211'

JohnTar22, 2023-07-22:
The above example is running 5.15 kernel

Ysurac, 2023-07-22:
All seems ok, you can try to run ```/bin/omr-tracker-ss``` and check if you have any output error.

JohnTar22, 2023-07-22:
nothing happens, no output at all.

Ysurac, 2023-07-22:
And no output in ```logread``` about tracker ? ```ps w | grep tracker``` show trackers ?

JohnTar22, 2023-07-22:
ps w | grep tracker:
19424 root      1368 S    {omr-tracker} /bin/sh /bin/omr-tracker wan3
19425 root      1368 S    {omr-tracker} /bin/sh /bin/omr-tracker wan2
19426 root      1376 S    {omr-tracker} /bin/sh /bin/omr-tracker omrvpn
19427 root      1208 S    {omr-tracker-ss} /bin/sh /bin/omr-tracker-ss tracker_sss0
20744 root      2100 S    /usr/bin/ss-local -c /var/etc/shadowsocks-libev/ss_local.tracker_sss0.json
29397 root      1696 S    {omr-tracker} /bin/sh /bin/omr-tracker wan2
29737 root      1176 R    grep tracker

logread | grep -i tracker: 
Jul 22 18:56:17 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/omr-tracker reload dependency on /etc/config/network
Jul 22 18:56:19 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/omr-tracker reload dependency on /etc/config/shadowsocks-libev
Jul 22 18:56:33 OpenMPTCProuter user.notice omr-tracker: Launching...
Jul 22 18:56:41 OpenMPTCProuter user.notice omr-tracker: Launched

JohnTar22, 2023-07-22:
it's very unstable, sometimes it downloads on 1.0 Mb rate but uploads on 15Mb, sometimes it  doesn't work at all, and sometimes the upload and download rates switch places.

Ysurac, 2023-07-22:
Can you put more ouput of logread ? Maybe some issue with shadowsocks...
5.15 kernel is not tested, so I have no idea if it work correctly or not, no release based on it will be done. Next release will be based on kernel 6.1.

JohnTar22, 2023-07-22:
i'm not sure that Gateworks will release kernel 6.1 for ventana, i can go back to 5.4 and try again. 
Jul 22 18:56:00 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.36.1
Jul 22 18:56:00 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S13openssl: Generating engines.cnf
Jul 22 18:56:00 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S13openssl: Generating providers.cnf
Jul 22 18:56:01 OpenMPTCProuter user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Jul 22 18:56:01 OpenMPTCProuter user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Jul 22 18:56:01 OpenMPTCProuter user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5353
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Jul 22 18:56:01 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Jul 22 18:56:01 OpenMPTCProuter authpriv.info dropbear[3272]: Not backgrounding
Jul 22 18:56:02 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 22 18:56:02 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jul 22 18:56:02 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jul 22 18:56:02 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jul 22 18:56:02 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jul 22 18:56:03 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jul 22 18:56:03 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jul 22 18:56:03 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jul 22 18:56:03 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Include '/etc/firewall.omr-bypass' failed with exit code 1
Jul 22 18:56:03 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Command failed: ubus call service delete { "name": "miniupnpd" } (Not found)
Jul 22 18:56:03 OpenMPTCProuter user.notice Shadowsocks: Rules UP
Jul 22 18:56:04 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Flush terminated
Jul 22 18:56:04 OpenMPTCProuter user.notice Shadowsocks: Reload omr-bypass rules
Jul 22 18:56:05 OpenMPTCProuter daemon.notice wpa_supplicant[4307]: Successfully initialized wpa_supplicant
Jul 22 18:56:05 OpenMPTCProuter user.notice : Added device handler type: 8021ad
Jul 22 18:56:05 OpenMPTCProuter user.notice : Added device handler type: 8021q
Jul 22 18:56:05 OpenMPTCProuter user.notice : Added device handler type: macvlan
Jul 22 18:56:05 OpenMPTCProuter user.notice : Added device handler type: veth
Jul 22 18:56:05 OpenMPTCProuter user.notice : Added device handler type: bridge
Jul 22 18:56:05 OpenMPTCProuter user.notice : Added device handler type: Network device
Jul 22 18:56:05 OpenMPTCProuter user.notice : Added device handler type: tunnel
Jul 22 18:56:07 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S50sqm: Starting SQM script: simple.qos on wwan0, in: 95000 Kbps, out: 95000 Kbps
Jul 22 18:56:08 OpenMPTCProuter daemon.warn netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?
Jul 22 18:56:08 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Jul 22 18:56:08 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Jul 22 18:56:08 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Jul 22 18:56:08 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is enabled
Jul 22 18:56:08 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is setting up now
Jul 22 18:56:08 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is now up
Jul 22 18:56:08 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is setting up now
[... cut ...]
Jul 22 19:36:53 OpenMPTCProuter user.notice post-tracking-post-tracking: Check API configuration...
Jul 22 19:36:53 OpenMPTCProuter user.notice post-tracking-post-tracking: Check API configuration... Done
Jul 22 19:44:58 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1690055098.026352 Warning: Configured global ping response timeout: 300 seconds exceeded.
Jul 22 19:54:43 OpenMPTCProuter user.notice cake-autorate.omrvpn: ERROR: 1690055683.497641 Received request to change maintain_pingers state to existing state.

Ysurac, 2023-07-23:
You can go back to 5.4 kernel and test yes, I never had any good result with 5.15 (I will stop compiling for it).
Maybe possible to port 5.15 patches to compile on 6.1 ?

JohnTar22, 2023-07-25:
Hi, thanks for the reply, went back to 5.4, now it looks good, i'll keep testing it in the next few days to see if it's stable.
I don't have a good answer about compiling kernel 6.1 for this specific device family, i guess i'll do it if i have to. 
Please keep this issue open for the next few days so i can make further testing.
Thanks again for the help and this great project!

JohnTar22, 2023-07-26:
did a few tests, trying to stream live HD video over OMR, when using cubic after 2 minutes (give or take) it just halts and kills the connection, when i changed to bbr and changed the tcp syn to 8 (kernel 5.4) it holds fine, doesn't really use the entire bandwidth but at least it;s stable. 
another thing is that even when i join 4 lte modems (2 cat 4 and 2 cat 20) i still get a laggy stream on the host.
when i run speedtest-cli --secure from a pc that's connected to the OMR i get 90mb download and 45mb upload. 
what am i missing? 
thanks!

Ysurac, 2023-07-26:
Did you use SQM with download/upload speed set on each interface ?

JohnTar22, 2023-07-26:
Iv'e checked the box but didn't set any values in the text boxes, how do i calculate the limits? On lte sometimes there's good reception and sometimes it falls back to 3g.

JohnTar22, 2023-07-29:
I saw something about using "tc" command, any "best practice" about the limit rates?

Ysurac, 2023-07-29:
This is done via the wizard or via Network->SQM.
You only need to put values that work in your case, put values and lower them if it's not working correctly.
SQM-Autorate should do that automatically but this need more tests as it's not always working correctly.

closed 2023-11-01 by github-actions[bot]