Ysurac/openmptcprouter #2892: (Temporary) Disable (LTE) interface with high ping

h45h3ncryp73d, 2023-07-22:
**Is your feature request related to a problem? Please describe.**
From time to time I experience extremely high ping on one of the LTE interfaces for unknown reason (I guess insufficient network capacity and inadequate LTE infrastructure).

**Describe the solution you'd like**
Maybe it would be possible to introduce an option allowing users to set a ping value after which certain interface would not be used until ping comes down below it again.

**Describe alternatives you've considered**
Actually haven't got any (yet).

**Additional context**
n/a

Ysurac, 2023-07-22:
This can already be done via Services->OMR-Tracker with specific setting on an interface, but can be simplified and done in a better way for this.

h45h3ncryp73d, 2023-07-22:
Thanks, definitely not a very friendly way to do it via OMR-Tracker ... If I got all the settings right minimum ping must be at least 1000ms for this to work, but I could easily be wrong ...

closed 2023-10-25 by github-actions[bot]