Ysurac/openmptcprouter #2893: LOG time is incorrect

fareign, 2023-07-23:
## Expected Behavior
Jul 23 11:54:59 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Jul 23 11:54:59 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Jul 23 11:54:59 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Jul 23 11:54:59 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Jul 23 11:54:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Jul 23 03:54:59 OpenMPTCProuter daemon.notice netifd: Interface 'wwan' is enabled
Jul 23 03:54:59 OpenMPTCProuter daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
Jul 23 03:54:59 OpenMPTCProuter daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Jul 23 03:54:59 OpenMPTCProuter daemon.notice netifd: bridge 'br-lan' link is up

## Current Behavior

IF I set system UTC,then some LOG is system time ,and some is 0 timezone time.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.59.2 Alpha
  - OpenMPTCProuter VPS version: 0.1029test
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86-64
  - Country:

Ysurac, 2023-07-23:
True, even after a reboot I still have log in 2 timezone...
I will check how this can happen and if there is a fix.

vempire-ghost, 2023-10-23:
Has this issue been fixed? My log is still in a different timezone than the one configured in Luci.

closed 2024-01-27 by github-actions[bot]