Ysurac/openmptcprouter #2895: V2ray service can't start by some setting

fareign, 2023-07-24:
## Expected Behavior

if I change socks to shadowsocks. it may cause v2rayservice cound not start.

## Current Behavior

VPS syslgo

Jul 24 14:02:48 iZwz9ecl90hv4wg2bwwqqmZ systemd[1]: Reloaded Shorewall IPv4 firewall.
Jul 24 14:02:49 iZwz9ecl90hv4wg2bwwqqmZ systemd[1]: Stopping V2Ray Service...
Jul 24 14:02:49 iZwz9ecl90hv4wg2bwwqqmZ systemd[1]: v2ray.service: Succeeded.
Jul 24 14:02:49 iZwz9ecl90hv4wg2bwwqqmZ systemd[1]: Stopped V2Ray Service.
Jul 24 14:02:49 iZwz9ecl90hv4wg2bwwqqmZ systemd[1]: v2ray.service: Start request repeated too quickly.
Jul 24 14:02:49 iZwz9ecl90hv4wg2bwwqqmZ systemd[1]: v2ray.service: Failed with result 'start-limit-hit'.
Jul 24 14:02:49 iZwz9ecl90hv4wg2bwwqqmZ systemd[1]: Failed to start V2Ray Service.
Jul 24 14:02:50 iZwz9ecl90hv4wg2bwwqqmZ ss-server[610]: getpeername: Transport endpoint is not connected
Jul 24 14:02:52 iZwz9ecl90hv4wg2bwwqqmZ ss-server[610]: getpeername: Transport endpoint is not connected
Jul 24 14:02:56 iZwz9ecl90hv4wg2bwwqqmZ ss-server[610]: getpeername: Transport endpoint is not connected

## Possible Solution

from the log stopped V2RAY twice then the V2ray service not start.

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.59.2 alpha
  - OpenMPTCProuter VPS version: 0.1029 test
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:x86_64
  - Country: cn

Ysurac, 2023-07-24:
Issue when V2Ray is used for port redirection is now fixed in VPS script, you can relaunch VPS script and make any change in Firewall router configuration to fix V2Ray config.
If you didn't made any port redirection, what is the result of ```journalctl -u v2ray```

fareign, 2023-07-24:
in firewall port forwarding if I check USE v2ray . it also can cause this issue.
Use V2Ray
Forward a port (not a range) from server using V2Ray proxy (if enabled) instead of VPN

raidzx, 2023-07-31:
I had an issue similar to this, had port forwarding enabled for v2ray and v2ray would not start. I did what @Kalimeiro suggested and that fixed it. Using the VPS script on this page: https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS

Are you sure its fixed in VPS script @Ysurac? Apologies if I am hijacking.

Ysurac, 2023-07-31:
It should be fixed in VPS snapshot script: ```wget -O - http://www.openmptcprouter.com/server-test/debian-x86_64.sh | sh```

closed 2023-08-29 by fareign