Ysurac/openmptcprouter #2896: Raspberry Pi 4B 64-bit image not working

Keksstar, 2023-07-24:
Hi!

Downloaded the 64 bit Raspberry Pi 4B Image. Sadly it´s not running and the Pi stucks on the first image where it shows colors...
The 32bit works fine.

How to fix this?

Ysurac, 2023-07-24:
Please use the template when you open an issue.
What is the downloaded image exactly ? version, file,...

Keksstar, 2023-07-24:
Sorry for that!

File from the OpenMPTCProuter page:
[openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz](https://download.openmptcprouter.com/release/v0.59.1/rpi4/targets/bcm27xx/bcm2711/openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz)

Used balenaEtcher to copy the file.
The 32-bit for the Pi 4B works fine, it´s just the 64-bit that isn´t working.

Ysurac, 2023-07-24:
Did you check checksum ?
Did you try the squashfs version ?

simonemessina92, 2023-07-25:
Recently I've tried many times to install OMR on the rpi4 B and not have any success to make it work , neither with openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz nor with the beta v.0.59.2 alpha 5.4. A month ago it worked, and soon after some weeks on the boot up it couldn't connect to the server, and neither surf the web in standalone mode without the bonding... Still wonder why.  Furthermoroe there's still the problem with the network interfaces that has to settle the "force link" to not go on dhcp loop when dhcp is enabled on the wan... I really hope to try a new version release as soon as possible. Thanks for your work @Ysurac , I'm looking forward for something new and stable to update both the multilan pc and the rpi 4.

Ysurac, 2023-07-26:
It's not related to initial issue...
Use default gateway should be disabled everywhere.
You are using alpha release, did you update also the VPS script ?
Can you take a screenshot of the status page ?

simonemessina92, 2023-07-26:
> It's not related to initial issue... Use default gateway should be disabled everywhere. You are using alpha release, did you update also the VPS script ? Can you take a screenshot of the status page ?

I'm using the alpha release, the server had of course been updated ( actually was a fresh install with the beta release ) at the time. You said the default gateway should be disabled everywhere, but all the times I've installed the OMR on the pi the lan comes with the flagged use default gateway, so I don't know if the alpha on the latest release of 23 July ( https://snapshots.openmptcprouter.com/5.4/rpi4/targets/bcm27xx/bcm2711/openmptcprouter-v0.59.2alpha-5.4-r0+16819-1b6e9b3f64-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz )  got some diffrences from the one you were suggested to use months ago, but turns out that the OMR itself wasn't functional neither without the bonding activated. First of all whenever I start the OMR I always settle every WAN interfaces, test all works and than connect to the VPS server, you know just to be sure everything is fine. I've tried both balena etcher and raspberrypi imager, because for some stupid reason i thought the software wasn't copied in the correct way on the pi, but it isn't seems something like that unfortunately. Even if on the lan I uncheck the use default gateway in the dashboard of OMR I don't have any connection, just using WAN 1 with an FTTH connected in DHCP... I don't know why

Ysurac, 2023-07-26:
I still need screenshot of system->OpenMPTCProuter status page.

simonemessina92, 2023-07-26:
> I still need screenshot of system->OpenMPTCProuter status page.

Ysurac at the time beeing there's nothing in the OpenMPTCProuter status page, because as I said I don't have connected the OMR router to the VPS , if you wish I'll try and sent you the screenshot, but as I said I'm worried that neither as a standalone router it works as it should... 

here's the terminal of the OMR on the pi you can see the wan got ip address, and now as you've suggested I've disabled the "use default gateway on the lan"

OpenMPTCProuter login: root

BusyBox v1.33.2 (2023-01-06 16:34:46 UTC) built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _           
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _ 
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|  
       |_|                                                               
------------------------------------------------------------------------------
   (r0+16819-1b6e9b3f64)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.59.2alpha
 TARGET:      rpi4
 ARCH:        aarch64_cortex-a72

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Sun Jul 23 09:40:36 UTC 2023
-----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenMPTCProuter:~# ifconfig
eth0      Link encap:Ethernet  HWaddr DC:A6:32:DC:D1:BD  
          inet addr:192.168.100.1  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:10805 errors:0 dropped:0 overruns:0 frame:0
          TX packets:15768 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:2000 
          RX bytes:1928772 (1.8 MiB)  TX bytes:14006702 (13.3 MiB)

eth1      Link encap:Ethernet  HWaddr 00:E0:4C:39:69:3C  
          inet addr:192.168.1.15  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:61869 errors:0 dropped:0 overruns:0 frame:0
          TX packets:55826 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:14499006 (13.8 MiB)  TX bytes:4645367 (4.4 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:7819 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7819 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:518641 (506.4 KiB)  TX bytes:518641 (506.4 KiB)

root@OpenMPTCProuter:~#

Ysurac, 2023-07-26:
What is the result of ```ip r``` ?

simonemessina92, 2023-07-26:
> What is the result of `ip r` ?

I've tried to connect to the server with the WAN1 but this is the screenshot of the status
[image]

and this is what I got from the terminal running the comand you told me 

root@OpenMPTCProuter:~# ip r
default via 192.168.1.1 dev eth1 metric 1 
default via 192.168.1.1 dev eth1 metric 3 initcwnd 10 initrwnd 10 
5.249.148.148 via 192.168.1.1 dev eth1 metric 1 
127.0.0.0/8 dev lo proto static scope link metric 5 
192.168.1.0/24 dev eth1 scope link metric 3 initcwnd 10 initrwnd 10 
root@OpenMPTCProuter:~#

Ysurac, 2023-07-26:
Can you try ```ping -c 1 1.1.1.1``` and  ```curl 1.1.1.1``` via SSH from the router to check internet connectivity ?
Can you also check Status->System Logs

simonemessina92, 2023-07-26:
> Can you try `ping -c 1 1.1.1.1` and `curl 1.1.1.1` via SSH from the router to check internet connectivity ? Can you also check Status->System Logs

root@OpenMPTCProuter:~# ping -c 1 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=53 time=7.83 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 7.832/7.832/7.832/0.000 ms
root@OpenMPTCProuter:~# curl 1.1.1.1
<html>
<head><title>301 Moved Permanently</title></head>
<body>
<center><h1>301 Moved Permanently</h1></center>
<hr><center>cloudflare</center>
</body>
</html>
root@OpenMPTCProuter:~#

Jul 25 22:56:55 OpenMPTCProuter daemon.info dnsmasq[2547]: using only locally-known addresses for domain local
Jul 25 22:56:55 OpenMPTCProuter daemon.info dnsmasq[2547]: using only locally-known addresses for domain invalid
Jul 25 22:56:55 OpenMPTCProuter daemon.info dnsmasq[2547]: using only locally-known addresses for domain bind
Jul 25 22:56:55 OpenMPTCProuter daemon.info dnsmasq[2547]: using only locally-known addresses for domain use-application-dns.net
Jul 25 22:56:55 OpenMPTCProuter daemon.info dnsmasq[2547]: using only locally-known addresses for domain lan
Jul 25 22:56:55 OpenMPTCProuter daemon.info dnsmasq[2547]: using nameserver 127.0.0.1#5353
Jul 25 22:56:55 OpenMPTCProuter daemon.info dnsmasq[2547]: using 1 more local addresses
Jul 25 22:56:55 OpenMPTCProuter daemon.info dnsmasq[2547]: read /etc/hosts - 4 addresses
Jul 25 22:56:55 OpenMPTCProuter daemon.info dnsmasq[2547]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Jul 25 22:56:56 OpenMPTCProuter authpriv.info dropbear[2707]: Not backgrounding
Jul 25 22:56:56 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jul 25 22:56:56 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Failed to connect to ubus
Jul 25 22:56:56 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_lan' cannot resolve device of network 'lan'
Jul 25 22:56:56 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan1'
Jul 25 22:56:56 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan2'
Jul 25 22:56:56 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omrvpn'
Jul 25 22:56:56 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'
Jul 25 22:56:56 OpenMPTCProuter daemon.notice wpa_supplicant[3080]: Successfully initialized wpa_supplicant
Jul 25 22:56:56 OpenMPTCProuter user.notice : Added device handler type: 8021ad
Jul 25 22:56:56 OpenMPTCProuter user.notice : Added device handler type: 8021q
Jul 25 22:56:56 OpenMPTCProuter user.notice : Added device handler type: macvlan
Jul 25 22:56:56 OpenMPTCProuter user.notice : Added device handler type: veth
Jul 25 22:56:56 OpenMPTCProuter user.notice : Added device handler type: bridge
Jul 25 22:56:56 OpenMPTCProuter user.notice : Added device handler type: Network device
Jul 25 22:56:56 OpenMPTCProuter user.notice : Added device handler type: tunnel
Jul 25 22:56:57 OpenMPTCProuter daemon.warn netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is enabled
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is setting up now
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is now up
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is enabled
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Network device 'eth0' link is up
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Interface 'lan' has link connectivity 
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Network device 'lo' link is up
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' has link connectivity 
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Network device 'eth1' link is up
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' has link connectivity 
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is setting up now
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: wan1 (4056): udhcpc: started, v1.33.2
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: wan1 (4056): udhcpc: sending discover
Jul 25 22:56:57 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of lan (eth0)
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: wan1 (4056): udhcpc: sending select for 192.168.1.15
Jul 25 22:56:57 OpenMPTCProuter daemon.notice netifd: wan1 (4056): udhcpc: lease of 192.168.1.15 obtained, lease time 4294967295
Jul 25 22:56:58 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is now up
Jul 25 22:56:58 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
[... cut ...]
Jul 25 23:33:06 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Jul 25 23:33:08 OpenMPTCProuter daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)
Jul 25 23:33:15 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Jul 25 23:33:19 OpenMPTCProuter user.notice omr-tracker: Launching...
Jul 25 23:33:24 OpenMPTCProuter user.notice omr-tracker: Launched
Jul 25 23:33:33 OpenMPTCProuter daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)

Ysurac, 2023-07-26:
Seems like if v2ray and shadowsocks are both running...
Did you change default proxy ?

simonemessina92, 2023-07-26:
> Seems like if v2ray and shadowsocks are both running... Did you change default proxy ?

I've always used V2ray without VPN on the server side, because I need both SRT and WebRTC and that settings seems only the one which works... I can try reboot the VPS, I'll do now!

Ysurac, 2023-07-26:
The problem is on router side here, with both proxy running. How did you enabled V2Ray ?
What is the result of ```uci show shadowsocks-libev``` and ```uci show openmptcprouter``` ?

simonemessina92, 2023-07-26:
> The problem is on router side here, with both proxy running. How did you enabled V2Ray ? What is the result of `uci show shadowsocks-libev` and `uci show openmptcprouter` ?

here's the results

root@OpenMPTCProuter:~#
root@OpenMPTCProuter:~#
root@OpenMPTCProuter:~# uci show shadowsocks-libev
shadowsocks-libev.hi1=ss_redir
shadowsocks-libev.hi1.server='sss0'
shadowsocks-libev.hi1.local_address='::'
shadowsocks-libev.hi1.local_port='1100'
shadowsocks-libev.hi1.mode='tcp_and_udp'
shadowsocks-libev.hi1.timeout='1000'
shadowsocks-libev.hi1.fast_open='1'
shadowsocks-libev.hi1.verbose='0'
shadowsocks-libev.hi1.syslog='1'
shadowsocks-libev.hi1.reuse_port='1'
shadowsocks-libev.hi1.mptcp='1'
shadowsocks-libev.hi1.ipv6_first='1'
shadowsocks-libev.hi1.no_delay='1'
shadowsocks-libev.hi1.ebpf='0'
shadowsocks-libev.hi2=ss_redir
shadowsocks-libev.hi2.local_port='1100'
shadowsocks-libev.hi2.mode='tcp_and_udp'
shadowsocks-libev.hi2.timeout='1000'
shadowsocks-libev.hi2.fast_open='1'
shadowsocks-libev.hi2.verbose='0'
shadowsocks-libev.hi2.syslog='1'
shadowsocks-libev.hi2.reuse_port='1'
shadowsocks-libev.hi2.mptcp='1'
shadowsocks-libev.hi2.ipv6_first='1'
shadowsocks-libev.hi2.no_delay='1'
shadowsocks-libev.hi2.local_address='::'
shadowsocks-libev.hi2.server='sss1'
shadowsocks-libev.hi2.ebpf='0'
shadowsocks-libev.ss_rules=ss_rules
shadowsocks-libev.ss_rules.disabled='0'
shadowsocks-libev.ss_rules.src_default='forward'
shadowsocks-libev.ss_rules.dst_default='forward'
shadowsocks-libev.ss_rules.local_default='forward'
shadowsocks-libev.ss_rules.server='sss0'
shadowsocks-libev.ss_rules.redir_tcp='all'
shadowsocks-libev.sss0=server
shadowsocks-libev.sss0.server_port='65101'
shadowsocks-libev.sss0.obfs='0'
shadowsocks-libev.sss0.server='5.249.148.148'
shadowsocks-libev.sss0.key='oXpWicXttKU0dmpuWV2iAOd/EoIvGlneaw7JWGAD+jY='
shadowsocks-libev.sss0.method='aes-256-gcm'
shadowsocks-libev.sss0.disabled='0'
shadowsocks-libev.sss0.obfs_host='host148-148-249-5.serverdedicati.aruba.it'
shadowsocks-libev.dns=ss_tunnel
shadowsocks-libev.dns.disabled='1'
shadowsocks-libev.dns.mode='tcp_and_udp'
shadowsocks-libev.dns.server='sss0'
shadowsocks-libev.dns.local_port='5353'
shadowsocks-libev.dns.tunnel_address='8.8.8.8:53'
shadowsocks-libev.hi3=ss_redir
shadowsocks-libev.hi3.server='sss0'
shadowsocks-libev.hi3.local_address='0.0.0.0'
shadowsocks-libev.hi3.local_port='1101'
shadowsocks-libev.hi3.mode='tcp_and_udp'
shadowsocks-libev.hi3.timeout='1000'
shadowsocks-libev.hi3.fast_open='1'
shadowsocks-libev.hi3.reuse_port='1'
shadowsocks-libev.hi3.syslog='1'
shadowsocks-libev.hi3.mptcp='1'
shadowsocks-libev.hi3.no_delay='1'
shadowsocks-libev.hi3.verbose='0'
shadowsocks-libev.hi3.ebpf='0'
shadowsocks-libev.hi4=ss_redir
[... 88 lines cut ...]
openmptcprouter.wan1.state='up'
openmptcprouter.wan1.multipathvpn='0'
openmptcprouter.wan1.mtu='1492'
openmptcprouter.wan1.mptcp_status='MPTCP enabled'
openmptcprouter.wan1.lc='1690329476'
openmptcprouter.wan1.latency='16'
openmptcprouter.wan1.latency_previous='16'
openmptcprouter.loopback=interface
openmptcprouter.loopback.multipath='off'
openmptcprouter.loopback.metric='5'
openmptcprouter.lan=interface
openmptcprouter.lan.multipath='off'
openmptcprouter.lan.metric='6'
openmptcprouter.lan.multipathvpn='0'
openmptcprouter.omrvpn=interface
openmptcprouter.omrvpn.multipath='off'
openmptcprouter.omrvpn.metric='1200'
openmptcprouter.omr6in4=interface
openmptcprouter.omr6in4.multipath='off'
openmptcprouter.omr6in4.metric='1201'
openmptcprouter.latest_versions=latest_versions
openmptcprouter.latest_versions.omr='0.59.1-5.4'
openmptcprouter.latest_versions.vps='0.1028'
openmptcprouter.latest_versions.lc='1690321869'
root@OpenMPTCProuter:~#

Ysurac, 2023-07-26:
```uci set shadowsocks-libev.sss0.disabled=1 && uci commit && reboot```

Ysurac, 2023-07-31:
You said it was working with squashfs image, it's not the case ?

Keksstar, 2023-07-31:
It seems to work in my eyes, but as you can see in the log of #2907 it seems to have errors?
So, how can we get this image up and running? After a lot of reading here, I shouldn´t be the only one with this problem...

Keksstar, 2023-07-31:
Reflashed the SD card with Rufus instead of BalenaEtcher. Now EXT4 is working. Thanks!

closed 2023-07-31 by Keksstar