Ysurac/openmptcprouter #2897: Getaway Down for DW5821e

madspeed, 2023-07-26:
Hello!
I am consistently having problem with USB modem DW5821e.

I have connected it using modemmanager, but it is always telling that the gateway is down. It the same time I have internet access, but when I connect the second modem the Dl speed reduces instead of increasing. I suppose that this problem may be caused as OMR thinks that main gateway is down.
[image]
[image]

Ysurac, 2023-07-26:
Gateway down is only a warning, only that gateway doesn't answer to ping. As you have connection, no issue here.
For the slowest speed when both modem connected, I need to know if you use same operator.
You also have an issue on VPS side: it's not using the good kernel, so no aggregation possible here. It's should use "5.4.*-mptcp" kernel. Did you reboot after VPS script launch ?

madspeed, 2023-07-26:
Thank you for a prompt reply!
I am using the same operator in both modems.
What should I do with VPS side? I have tried to reboot, still "Version 0.1028 5.10.0-21-amd64"

Ysurac, 2023-07-26:
On the VPS, check kernel with ```uname -a```, version displayed in status page can take some time before update.
If it's also 5.10, what is the OS installed on the VPS (should be a Debian) ? And what is the VPS provider ?

madspeed, 2023-07-26:
Linux localhost 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU
Debian 11
VPS is REG.RU

Ysurac, 2023-07-26:
I will not be able to test on this provider.
What do you have in ```/boot/grub/grub.cfg```, in ```/etc/default/grub``` and the result of ```ls /boot/``` ?

madspeed, 2023-07-26:
I can deploy a new server with debian 11 for you for tests, ok?
[image]
[image]
[image]

Ysurac, 2023-07-26:
I only need the full content of ```/boot/grub/grub.cfg``` file to check if entry "1>6" is the correct one or if it's the wrong kernel.

Ysurac, 2023-07-26:
Maybe you have a ```/boot/grub2``` directory ? But you must have a grub.cfg with more than that. Do a ```cat /boot/grub*/grub.cfg```

madspeed, 2023-07-27:
You are right, here is the full grub.cfg:
[image]

madspeed, 2023-07-27:
You were right, it was not full file. Here is full grub.cfg:
[grub.txt](https://github.com/Ysurac/openmptcprouter/files/12180034/grub.txt)

Ysurac, 2023-07-27:
The grub.cfg is correct, it should use the right entry.
Check if the VPS is configured to boot from kernel on disk, some VPS can boot from a network kernel.
Check if you can get a VPS Console access to see Grub boot menu when the VPS is starting.

madspeed, 2023-07-27:
could you please give commands to check it?

Ysurac, 2023-07-27:
No commands here, you need to check via your VPS provider interface.
Configuration on the VPS is correct.

madspeed, 2023-07-27:
Do I understand it correctly that VPS debian 11 kernel should be 5.4+smth...?

Ysurac, 2023-07-27:
Yes

madspeed, 2023-07-27:
Reinstalled the VPS, now everything is good with kernel.
Decided to test aggregation, but now have problems with DW5821e, the modemmanager fails after several minutes of work and does not reconnect. If I remove second modem (wan2), everything is good and dl speed is higher.
[image]

Ysurac, 2023-07-27:
Can you check in menu Status->System logs what ModemManager is saying ?

madspeed, 2023-07-27:
and after reconnect
[log.txt](https://github.com/Ysurac/openmptcprouter/files/12182360/log.txt)

Ysurac, 2023-07-27:
It's seems that there is some USB problems. Do you use a powered USB hub ?

madspeed, 2023-07-27:
modems are plugged into rpi4, no hubs

Ysurac, 2023-07-27:
I think it's the problem: RPI4 is really bad to give enough power to some USB device. You should try with a powered USB hub.

closed 2023-10-31 by github-actions[bot]