Ysurac/openmptcprouter #2898: The modems are not being aggregated

sanfweb, 2023-07-27:
Hi!

Please help me understand why the internet is not being aggregated.

I am using a Raspberry Pi 4 with openmptcprouter v0.59.1-54 r0+16594-ce92de8c8c installed and a VPS with 5.4.207-mptcp 0.1028. I have two Fibocom L850-GL modems connected via USB adapters 3.0

[image]

When I measure the speed of the first modem (omr-test-speed eth4), I get a result of 16M, sometimes a bit higher.

[image]

When I measure the speed of the second modem (omr-test-speed eth1), I get a result of 13M, sometimes slightly higher.

[image]

However, when I measure the aggregated speed (omr-test-speed), I get a result of 16M, sometimes a bit higher, and this doesn't seem like aggregation.

[image]

Please help me understand what I'm doing wrong and where the problem might be.
Thank you very much!

Ysurac, 2023-07-27:
To check if you have aggregation, look at bandwidth tab in System->MPTCP.
Modems are on different ISP ?

sanfweb, 2023-07-28:
I used two modems with the same providers, is it not possible to achieve aggregation in this case?

sanfweb, 2023-07-28:
I conducted a new experiment, I added one more modem, so now I have three, each with a different provider.

[image]

I went to the bandwidth section and initiated a speed test using speedtest in another window. In the bandwidth section, I see that two providers are being aggregated, but the third one is not.

[image]

Ysurac, 2023-07-28:
If your VPS provider can provide more than 300Mb/s, possible that this provider filter MPTCP, you can check by disabling another connection and test again.

sanfweb, 2023-07-28:
> 

I will clarify the limitation and conduct the experiment as you suggested.

sanfweb, 2023-07-28:
I mentioned earlier about using multiple modems with the same provider, does aggregation not occur in this case.

This happens only when there are two modems with the same provider, or if, for example, I take four modems, and two of them have the same provider, will it work?

Ysurac, 2023-07-28:
When using same provider the problem is the antenna that often have a limited bandwidth available shared between all customer. With 2 modems using same provider you are sharing same bandwidth.

sanfweb, 2023-08-02:
Seems like the issue indeed lies in MPTCP being blocked by one of the providers. Thank you!

closed 2023-08-02 by sanfweb