Ysurac/openmptcprouter #2899: R86S (x86_64 w/EFI): Needs updated IGC driver, plus REQ x86_64 target include kmod-mlx4-core

pipe2null, 2023-07-27:
The problem:
- Intel 2.5G pcie network adapter: VEN_8086 DEV_125C
- Handled by IGC kmod, hardware also used in some NUCs.

OpenWRT with kernel 5.10.176 fully supports, no known problems.
OMR with kernel 5.4:
- does not recognize device, forcing mod load for IGC driver with "echo 8086 125C > new_id" fails.
OMR with kernel 5.15.50:
- recognizes device and works with static IP, but using DHCP interface directly with ethX will constantly repeatedly reset the port after obtaining an IP for a few seconds.  Using a "Bridge Device" between the raw port and the DHCP interface works around the problem, but is clearly an ugly hack although trivial.
OMR with kernel 6.1:
- I attempted building an image, but failed.

**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
More or less yes.  See above.

**Describe why you are asking for this support?**
The R86S is becoming a very popular hardware choice for net appliances: 3x 2.5G (intel) and 2x 10G SFP+ (connectx) NICs, Intel quad core proc, built in mmblk + sdcard + M.2 NVME...  It is a very nice platform and very cheap for its feature set.

Due to the popularity of the R86S platform, it might be worthwhile adding it to the officially supported hardware list. 
Additional:  It would be a "nice to have" the kmod-mlx4-core driver included by default.

Ysurac, 2023-07-27:
You can try snapshots images:
* 5.4 kernel: https://snapshots.openmptcprouter.com/5.4/x86_64/targets/x86/64/
* 6.1 kernel: https://snapshots.openmptcprouter.com/6.1/x86_64/targets/x86/64/

I will add kmod-mlx4-core on x86_64 for next release.

pipe2null, 2023-07-27:
Awesome, thanks!

Re-tested:

openmptcprouter-v0.59.2alpha-5.4-r0+16819-1b6e9b3f64-x86-64-generic-ext4-combined-efi.img
- IGC driver recognizes 2.5G ports, and appears to work ok with static IP, but the port will continually get reset within seconds of getting a DHCP IP, same behavior I saw with the 5.15.50 kernel.
-  The hacky workaround works:  wrap the physical port ethX with a bridge, then use DHCP interface for the hacky bridge.
- Software repo: various issues with repo access from a snapshot build...  But you probably already know this.
- Ran into package dependency issues when trying to manually install mlx4-core, dependent packages kmod-ptp looking for out-of-date version of kmod-pps.  So, SFP+ ports were not tested with this build.

openmptcprouter-v0.59.2alpha-6.1-r0+23414-ed816f6ba8-x86-64-generic-ext4-combined-efi.img
- 2.5G ports: Same behavior as other builds, DHCP continuously resets the port but works for static IP or if you wrap with a bridge.
-  manual install of kmod-mlx4-core worked, and things look like they are working alright.

For use until next release (or other builds you'd like me to try): What build would you recommend as the most stable between released 5.15.50,  0.59.2alpha-5.4 (.225-1),  0.59.2alpha-6.1 (.34-1)  AKA the builds that at least recognize the 2.5G ports?

pipe2null, 2023-07-28:
Same 2.5G issue on:
openmptcprouter-v0.59.2alpha-5.15-r0+23227-106c83a1ea-x86-64-generic-ext4-combined-efi

One more workaround of sorts:
The 2.5G problem appears to ONLY apply to a DHCP interface for a raw untagged ethX port, where the port will get reset continuously.  Using static IP works, AND a DHCP interface for a tagged VLAN also works, as in eth1.10 works fine.  Only DHCP of the raw untagged port appears to be the problem, and if there are both tagged and untagged, the untagged will get the whole port reset which also drops the tagged interface as well.

Ysurac, 2023-07-29:
Did you try to check "force link" in Network->interfaces ?
There is some issues with I22* drivers.

pipe2null, 2023-07-30:
Reproed and yes, "force link" also works around the problem.

closed 2023-11-03 by github-actions[bot]