Ysurac/openmptcprouter #2900: Documentation request: How to enable openvpn on OMR router

ioogithub, 2023-07-27:
I would like to use a VPN on the OMR router. @Ysurac, you mentioned that "OpenVPN can be used on the router but **you have to configure routes and firewall for that**." Can you explain further what needs to be configured to get this working?

Here is my progress so far, I have three specific questions below:

### Import OpenVPN config:
- I have activated an openvpn from a *.ovpn file.
- Menu->VPN->OpenVPN shows `Enabled` and `Started`.
-Confirmed from Status->System Log:

```
Thu Jul 27 17:46:08 2023 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu Jul 27 17:46:08 2023 daemon.notice netifd: Network device 'tun0' link is up
Thu Jul 27 17:46:08 2023 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Thu Jul 27 17:46:08 2023 daemon.notice netifd: Interface 'omrvpn' is setting up now
Thu Jul 27 17:46:08 2023 daemon.notice openvpn(vpn)[21384]: net_iface_mtu_set: mtu 1500 for tun0
Thu Jul 27 17:46:08 2023 daemon.notice openvpn(vpn)[21384]: net_iface_up: set tun0 up
Thu Jul 27 17:46:08 2023 daemon.notice openvpn(vpn)[21384]: net_addr_v6_add: address/64 dev tun0
Thu Jul 27 17:46:08 2023 daemon.notice openvpn(vpn)[21384]: /usr/libexec/openvpn-hotplug up vpnnane tun0 1500 1553 ip2 255.255.255.0 init
Thu Jul 27 17:46:08 2023 daemon.notice netifd: Interface 'omrvpn' is now up
Thu Jul 27 17:46:08 2023 daemon.warn openvpn(vpn)[21384]: NOTE: unable to redirect IPv4 default gateway -- Cannot read current default gateway from system
Thu Jul 27 17:46:08 2023 daemon.notice openvpn(vpn)[21384]: Initialization Sequence Completed
Thu Jul 27 17:46:08 2023 daemon.warn [7092]: <warn>  [SQM_IFB_4c660] invalid sysfs path read for net/SQM_IFB_4c660
Thu Jul 27 17:46:08 2023 daemon.warn [7092]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_4c660 not found
Thu Jul 27 17:46:08 2023 daemon.warn [7092]: <warn>  [SQM_IFB_0469b] invalid sysfs path read for net/SQM_IFB_0469b
Thu Jul 27 17:46:08 2023 daemon.warn [7092]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_0469b not found
Thu Jul 27 17:46:08 2023 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Thu Jul 27 17:46:08 2023 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Jul 27 17:46:08 2023 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Thu Jul 27 17:46:09 2023 user.notice MPTCP: Add route ip1 via ip3 dev eth0
Thu Jul 27 17:46:09 2023 user.notice MPTCP: Flush route cache
Thu Jul 27 17:46:09 2023 user.notice MPTCP: Add route ip1 via ip3 dev eth0
```
**Question 1**: I see` tun0` and `omrvpn` is up but when I ssh into the OMR and do a `ip address` I do not see tun0 or omrvpn listed. Why?
**Question 2**: Is this error a problem: `NOTE: unable to redirect IPv4 default gateway -- Cannot read current default gateway from system`

### Firewall and Routes
**Question 3**: What do I have to do next with firewall and routes?
- Status->Routes I see `tun0` has target and ipv4 gateway. What other routes do  I need to configure?
- Status->Firewall I see `omr_vpn_import ` rules but I don't know how to configure this.

## Specifications
  - OpenMPTCProuter version: [v0.59.1-5.4 r0+16594-ce92de8c8c]
  - OpenMPTCProuter VPS version:
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:(x86_64)

I saw that a few others users in past issues were asking about similar functionality. If you can provide guidance on how to get it working I would be happy to write a tutorial to add to the wiki for future users.

Ysurac, 2023-07-27:
You must use another interface than tun0 for OpenVPN as it's already used by Glorytun VPN on OMR, this should solve already some problems.

ioogithub, 2023-07-27:
Thank you @Ysurac. I have modified my openvpn config to use tun1. I can confirm the vpn tunnel tun1 is up and has an ip address:
```
Thu Jul 27 19:25:25 2023 daemon.notice openvpn(vpn)[8497]: net_iface_up: set tun1 up
Thu Jul 27 19:25:25 2023 daemon.notice openvpn(vpn)[8497]: /usr/libexec/openvpn-hotplug up vpn tun1 1500 1553 vpnip 255.255.255.0 init
Thu Jul 27 19:25:25 2023 daemon.warn openvpn(vpn)[8497]: NOTE: unable to redirect IPv4 default gateway -- Cannot read current default gateway from system
Thu Jul 27 19:25:25 2023 daemon.notice openvpn(vpn)[8497]: Initialization Sequence Completed
```

On Status->Routes->Active IPv4 Routes: I see a entry added to route table:
```
(tun1)	vpnip/24	-	0	main	kernel
````

What is the next step to get the traffic coming in from the lan go go out over vpn on top of the multi-path? I guess I need to use firewall zones or add routes? This part I need help with until I understand better how it is all working.

Note, every time the tun1 interface is restarted, the IP address and gateway assigned to tun1 from the vpn server change so the solution needs to be dynamic.

ioogithub, 2023-07-28:
Hi @Ysurac, I have brought the interface up on tun1 and it is working. Can you suggest what settings I need to make to the firewall and route table to get the land traffic to go out over the vpn then over the multi-path connection?

Ysurac, 2023-07-29:
I would try a redirect rule from lan to the vpn.

closed 2023-11-01 by github-actions[bot]