Ysurac/openmptcprouter #2901: Agregation Ok sur le routeur mais pas sur le réseau

iitsmee, 2023-07-27:
Bonjour, 
je suis désolé je vais écrire en français car mon niveau en anglais ne me permet pas d’exprimer clairement ce que je souhaite

Je viens de refaire une installation au propre car je viens de recevoir une antenne Starlink qui va remplacer un forfait mobile + une box ADSL
Normalement je devrais donc selon les tests séparé un total de 400-450 en download et 50-70 en Upload

 Current Behavior

En réalité actuellement quand je fais des tests sur speed test ou Nperf sur un PC connecté sur le réseau je suis plutôt à 150 en download et 20 en Up ce qui correspond plus à Starlink seul, et en effet si je regarde l'état de mon routeur 4G Free Pendant un speed test il ne se passe rien il n'utilise aucune donnée.
Cependant si j'effectue la commande 
"omr-test-speed" le débit s’agrège bien dans les tests sur terminal du routeur, donc l’agrégation fonctionne sur le routeur mais pas sur mon réseau du moins c'est l'impression que ça me donne 

Je cherche donc ce qui pourrait provoquer ça.
avant avec des routeurs 4G je n'avais pas ce genre de problème 
La Starlink est en master et la free en activé 

Dans le statut tout est vert
## Specifications

  - OpenMPTCProuter version: [v0.59.1-5.4 r0+16594-ce92de8c8c]
  - OpenMPTCProuter VPS version: Debian 11 / 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: SAPINET
  - OpenMPTCProuter platform: PC fixe HP x86-64

Merci de votre futur aide

iitsmee, 2023-07-27:
Wed Jul 26 22:58:43 2023 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Jul 26 22:58:43 2023 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Jul 26 22:58:43 2023 daemon.notice netifd: Interface 'omrvpn' is now up
Wed Jul 26 22:58:43 2023 daemon.info glorytun[18682]: INITIALIZED tun0
Wed Jul 26 22:58:43 2023 user.notice omr-tracker: Launching...
Wed Jul 26 22:58:44 2023 daemon.err glorytun[18682]: 45.90.160.41.65001: connected
Wed Jul 26 22:58:44 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [SQM_IFB_2f7f2] invalid sysfs path read for net/SQM_IFB_2f7f2
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_2f7f2 not found
Wed Jul 26 22:58:44 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [SQM_IFB_7afbb] invalid sysfs path read for net/SQM_IFB_7afbb
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_7afbb not found
Wed Jul 26 22:58:44 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [SQM_IFB_96198] invalid sysfs path read for net/SQM_IFB_96198
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_96198 not found
Wed Jul 26 22:58:44 2023 daemon.info glorytun[18682]: STARTED tun0
Wed Jul 26 22:58:44 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [SQM_IFB_3823e] invalid sysfs path read for net/SQM_IFB_3823e
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_3823e not found
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [SQM_IFB_a3b3a] invalid sysfs path read for net/SQM_IFB_a3b3a
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_a3b3a not found
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [SQM_IFB_e4f01] invalid sysfs path read for net/SQM_IFB_e4f01
Wed Jul 26 22:58:44 2023 daemon.warn [11597]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_e4f01 not found
Wed Jul 26 22:58:45 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Wed Jul 26 22:58:45 2023 daemon.warn [11597]: <warn>  [SQM_IFB_1e40f] invalid sysfs path read for net/SQM_IFB_1e40f
Wed Jul 26 22:58:45 2023 daemon.warn [11597]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_1e40f not found
Wed Jul 26 22:58:45 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Wed Jul 26 22:58:45 2023 daemon.warn [11597]: <warn>  [SQM_IFB_eac30] invalid sysfs path read for net/SQM_IFB_eac30
Wed Jul 26 22:58:45 2023 daemon.warn [11597]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_eac30 not found
Wed Jul 26 22:58:45 2023 user.notice Shadowsocks: Rules DOWN
Wed Jul 26 22:58:45 2023 daemon.warn [11597]: <warn>  [SQM_IFB_be1e1] invalid sysfs path read for net/SQM_IFB_be1e1
Wed Jul 26 22:58:45 2023 daemon.warn [11597]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_be1e1 not found
Wed Jul 26 22:58:45 2023 daemon.warn [11597]: <warn>  [SQM_IFB_0f194] invalid sysfs path read for net/SQM_IFB_0f194
Wed Jul 26 22:58:45 2023 daemon.warn [11597]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_0f194 not found
Wed Jul 26 22:58:45 2023 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Wed Jul 26 22:58:46 2023 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Wed Jul 26 22:58:46 2023 user.notice Shadowsocks: Rules UP
Wed Jul 26 22:58:46 2023 user.notice Shadowsocks: Reload omr-bypass rules
Wed Jul 26 22:58:46 2023 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jul 26 22:58:47 2023 user.notice omr-bypass: OMR-ByPass is running
Wed Jul 26 22:58:47 2023 user.info : using tcp fast open
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-local[22981]: using tcp fast open
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-redir[22982]: enable TCP no-delay
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-redir[22982]: resolving hostname to IPv6 address first
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-local[22981]: enable TCP no-delay
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-redir[22982]: initializing ciphers... aes-256-gcm
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-local[22981]: initializing ciphers... aes-256-gcm
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-redir[22982]: listening at :::1100
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-redir[22982]: tcp port reuse enabled
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-local[22981]: listening at 127.0.0.1:1111
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-redir[22982]: UDP relay enabled
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-local[22981]: tcp port reuse enabled
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-redir[22982]: udp port reuse enabled
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-local[22981]: udprelay enabled
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-redir[22982]: running from root user
Wed Jul 26 22:58:47 2023 daemon.info /usr/bin/ss-local[22981]: udp port reuse enabled
[... cut ...]
Thu Jul 27 16:12:59 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth0) 192.168.100.181 2c:4d:54:59:fd:00
Thu Jul 27 16:12:59 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(eth0) 192.168.100.181 2c:4d:54:59:fd:00 Serveur-Plex
Thu Jul 27 16:22:35 2023 user.notice post-tracking-post-tracking: Check API configuration...
Thu Jul 27 16:22:44 2023 user.notice post-tracking-post-tracking: Check API configuration... Done
Thu Jul 27 16:34:53 2023 user.info : luci: accepted login on /admin/status/syslog for root from 192.168.100.229

Ysurac, 2023-07-27:
Si c'est bon côté routeur, il faudrait vérifié côté client l'adresse IP et la route par défaut utilisé afin d'être certain qu'il sorte bien via OMR. Sinon simplement en allant sur http://ifconfig.co/

iitsmee, 2023-07-27:
le site http://ifconfig.co/ m'indique que je suis sur le réseau starlink et non sapinet,
mon ordinateur est pourtant connecté en DHCP au OMR

Ysurac, 2023-07-27:
Il récupére donc l'IP via le DHCP de starlink. Essaye de passer en configuration statique et donc de régler IP et gateway manuellement.

iitsmee, 2023-07-27:
Je pense que tu parles de mon PC pour l'adresse statique
que je sois en statique ou en DHCP ça ne change pas j'ai toujours en passerelle 192.168.100.1 et une IP comprise dans ce réseau par défaut starlink attribue des IP en 192.168.1.XX

Si je fais un SPEED TEST il m'indique bien mon IP Publique de Sapinet lors du test 
Par contre sur ifconfig.co  starlink 
(merci pour ton temps)

iitsmee, 2023-07-27:
Alors la je suis peut être sur une piste...
Vu que le routeur STARLINK n'est pas configurable j'ai activé le mode Bypass et je le gère donc depuis un routeur Huawei (qui est bien configuré en statique) sur le quel je viens de désactiver le par-feu et la les connexion ont l'air de bien s’agréger.
Je continuerai mes recherche demain
Merci

iitsmee, 2023-08-03:
Finalement après différents essaie ça semble assez aléatoire à des moments ça agrège bien et d'autre non.
J'ai remarqué que l'upload à encore plus de mal à s’agréger que le download...

Ysurac, 2023-08-03:
Normalement ça marche tout le temps, faut vérifier qu'il n'y a pas d'autres soucis avec le DHCP ou parefeu, et vérifier les logs sur OpenMPTCProuter aussi voir si il n'y a pas un souci niveau proxy (Shadowsocks par défaut) ou VPN (Glorytun TCP par défaut)  .

iitsmee, 2023-08-05:
Dernier test effectué mettre FreeMobile en master plutôt que Starlink, pour l'instant j'en suis plutôt satisfait même si ça reste encore aléatoire j'ai l'impression...
Est ce que tu penses que cela peut venir des services car j'ai actuellement une personne qui utilise ma bibliothèque Plex et l'upload est seulement sur starlink, si au même moment j'envoi une longue vidéo via snap depuis un Iphone ça utilise Free.
Et actuellement si je fais un Speed test sur OOkla ça semble s’agréger je dirais 80% du temps.

iitsmee, 2023-08-11:
Salut je reviens demander conseil je comprend plus rien :'-(
J'ai branché un vieux PC sur mon réseau pour tester avec un appareil qui n'a jamais était branché sur mon routeur depuis la nouvelle configuration avec starlink.
Donc je lance un speed test avec cet ancien appareil je retrouve la bonne connexion agréger comme j'ai par moment, donc content, je lance ensuite un téléchargement de données sur Google un gros fichier de 50Go et la il n'y a que starlink qui est utilisé alors que c'est au même moment avec le même appareil, je comprend plus trop.
voici le journal système : 
Thu Aug 10 18:27:56 2023 kern.info kernel: [   13.651394] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
Thu Aug 10 18:27:56 2023 kern.info kernel: [   13.652130] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Thu Aug 10 18:27:57 2023 daemon.err glorytun[9058]: 45.90.160.41.65001: connected
Thu Aug 10 18:27:57 2023 daemon.info glorytun[9058]: STARTED tun0
Thu Aug 10 18:27:57 2023 user.notice mptcp: Reloading mptcp config due to ifup of wan1 (wan1)
Thu Aug 10 18:27:57 2023 daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Thu Aug 10 18:27:57 2023 daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Thu Aug 10 18:27:57 2023 daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Thu Aug 10 18:27:57 2023 daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Thu Aug 10 18:27:57 2023 daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Thu Aug 10 18:27:57 2023 user.notice omr-tracker: Launching...
Thu Aug 10 18:27:58 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Thu Aug 10 18:27:58 2023 daemon.warn [7929]: <warn>  [SQM_IFB_55d7f] invalid sysfs path read for net/SQM_IFB_55d7f
Thu Aug 10 18:27:58 2023 daemon.warn [7929]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_55d7f not found
Thu Aug 10 18:27:58 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Thu Aug 10 18:27:58 2023 daemon.warn [7929]: <warn>  [SQM_IFB_e1262] invalid sysfs path read for net/SQM_IFB_e1262
Thu Aug 10 18:27:58 2023 daemon.warn [7929]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_e1262 not found
Thu Aug 10 18:27:58 2023 user.notice firewall: Reloading firewall due to ifup of wan2 (wan2)
Thu Aug 10 18:27:58 2023 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Aug 10 18:27:59 2023 daemon.info [7929]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:1f.6': not supported by any plugin
Thu Aug 10 18:28:02 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth0) 00:31:92:26:79:cc
Thu Aug 10 18:28:02 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth0) 192.168.100.194 00:31:92:26:79:cc
Thu Aug 10 18:28:02 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth0) 00:31:92:26:79:cc
Thu Aug 10 18:28:02 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth0) 192.168.100.194 00:31:92:26:79:cc
Thu Aug 10 18:28:02 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth0) 192.168.100.194 00:31:92:26:79:cc
Thu Aug 10 18:28:02 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(eth0) 192.168.100.194 00:31:92:26:79:cc deco-M4R
Thu Aug 10 18:28:03 2023 user.notice mptcp: Reloading mptcp config due to ifup of wan2 (wan2)
Thu Aug 10 18:28:03 2023 user.notice unbound: default memory configuration
Thu Aug 10 18:28:03 2023 user.notice mptcp: Reloading mptcp config due to ifup of loopback (lo)
Thu Aug 10 18:28:03 2023 daemon.notice unbound: [16338:0] notice: init module 0: validator
Thu Aug 10 18:28:03 2023 daemon.notice unbound: [16338:0] notice: init module 1: iterator
Thu Aug 10 18:28:03 2023 daemon.info unbound: [16338:0] info: start of service (unbound 1.15.0).
Thu Aug 10 18:28:03 2023 daemon.info unbound: [16338:0] info: generate keytag query _ta-4f66. NULL IN
Thu Aug 10 18:28:03 2023 daemon.warn [7929]: <warn>  [SQM_IFB_8cb69] invalid sysfs path read for net/SQM_IFB_8cb69
Thu Aug 10 18:28:03 2023 daemon.warn [7929]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_8cb69 not found
Thu Aug 10 18:28:03 2023 daemon.warn [7929]: <warn>  [SQM_IFB_2f4ec] invalid sysfs path read for net/SQM_IFB_2f4ec
Thu Aug 10 18:28:03 2023 daemon.warn [7929]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_2f4ec not found
Thu Aug 10 18:28:03 2023 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Thu Aug 10 18:28:03 2023 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Aug 10 18:28:03 2023 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Thu Aug 10 18:28:06 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth0) 02:75:cc:40:20:9f
Thu Aug 10 18:28:06 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth0) 192.168.100.240 02:75:cc:40:20:9f
Thu Aug 10 18:28:06 2023 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Thu Aug 10 18:28:06 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Aug 10 18:28:06 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Thu Aug 10 18:28:06 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth0) 192.168.68.111 02:75:cc:40:20:9f
Thu Aug 10 18:28:06 2023 daemon.info dnsmasq-dhcp[1]: DHCPNAK(eth0) 192.168.68.111 02:75:cc:40:20:9f wrong server-ID
[... cut ...]
Fri Aug 11 17:00:08 2023 user.notice post-tracking-post-tracking: Check API configuration...
Fri Aug 11 17:00:13 2023 user.notice post-tracking-post-tracking: Check API configuration... Done
Fri Aug 11 17:11:44 2023 daemon.err /usr/bin/ss-redir[4262]: server recv: Operation timed out
Fri Aug 11 17:15:44 2023 daemon.notice ttyd[25924]: WS closed from 192.168.100.25, clients: 0
Fri Aug 11 17:15:44 2023 daemon.notice ttyd[25924]: killing process 7292 with signal: 1 (SIGHUP)
Fri Aug 11 17:15:44 2023 daemon.notice ttyd[25924]: process killed with signal 1 (SIGHUP), pid: 7292

iitsmee, 2023-08-11:
je viens d'essayer avec un autre VPS dans le doute sur Ionos et même constat agrégation aléatoire

Xenon69, 2023-08-12:
Hello,
Comment le routeur Starlink est-il configuré ?

Ysurac, 2023-08-12:
Le comportement de Starlink est souvent un peu étrange en effet, je teste encore afin de savoir si il n'y aurait pas des priorités selon les flux.

iitsmee, 2023-08-13:
D'accord je comprend je fais aussi beaucoup d'essai j'active ou désactive certaines options pour trouver le meilleur compromis mais c'est vrai que j'ai l'impression qu'il réagi moins bien qu'une connexion LTE standard.
Merci Ysurac pour ton travail.

closed 2023-11-17 by github-actions[bot]