Ysurac/openmptcprouter #2905: Spitz AX (GL-X3000) - Wi-Fi 6 Dual-SIM 4G/5G Cellular Gateway

mati0090, 2023-07-30:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
GL.Inet Spitz AX is newly released router with dual sim support for 5G/4G connections, which has preinstalled OpenWRT with custom GUI interface. It makes difficult to build OpenWRT package which would include all parts (at lease for me). 

**Describe why you are asking for this support?**
It's a perfect router for recreational vehicles which makes perfect usage of bounding multi sim connections, especially in rural areas. It has 4x4 MIMO 5G/4G connectors and following hardware:
_MediaTek Dual-core MT7981A, @1.3GHz
DDR4 512 MB / eMMC 8GB
Antennas | 6 x Antennas (4 for 5G NR, 2 for Wi-Fi)_

**Additional context**
Can I have some clarification, how custom build should be prepared? Ideally, if all GL.Inet preinstalled packages would be included. (https://github.com/gl-inet/gl-infra-builder)
Any help with build for some GL.Inet hardware would be very much appreciable.

highfreq, 2023-09-05:
It is a dual sim, single modem. Useless for mptcprouter.

wombat94, 2023-09-14:
> It is a dual sim, single modem. Useless for mptcprouter.

I'll add my request for support of this router.

It is not useless as an mptcprouter (it is at least as useful in my mind as RPi4)... the GL-X3000 has support for up to 4 WAN connections (1 5G/LTE modem internal, 1 USB-tethered internet source - phone tethering or USB LTE modem, 1 Wi-Fi WAN connection and 1 hard wired Ethernet)

A very typical use case would be dual SIMs in the GL-X3000 and Starlink connected via ethernet as a base config.

I don't have Starlink yet, but have the failover/multi-wan working currently with T-Mo/AT&T SIMs in the GL-X3000 and a second T-Mo SIM running in the Cudy P5 I replaced with this router connected via Ethernet.

It all works well for failover/simple load balancing but I am looking for options for connection bonding when I am in areas with less stable/lower speed connections.

closed 2023-12-19 by github-actions[bot]

gustawdaniel, 2025-03-07:
@highfreq can you explain what do you mean by "Useless for mptcprouter."?

Can we reopen this issue and investigate again what is state of support of openmptcprouter for Spitz AX (GL-X3000)?

Connected topic:
https://forum.gl-inet.com/t/openmptcprouter-build-for-gl-x3000/31947
https://www.reddit.com/r/GlInet/comments/15ed77p/openmptcprouter_build_for_glx3000/

Ysurac, 2025-03-07:
GL-X3000 only have one modem, there is nothing to aggregate with only one modem. It's why support was not added.

geekaz01d, 2025-03-31:
The AX3000 is a multi-wan device. The internal radio is only one of the WAN options.

I'm currently testing:
- wireless WAN
- ethernet WAN to an external gateway (*link, M5 loco, mifi... you pick)
- usb tethering (currently testing with a spare phone but a p*plink max would work)

We treat it like a swiss army knife for networking at the edge. Sp**dify is supported, users are interested, so why rule it out?

onnimonni, 2025-04-21:
Hey,

I saw that similiarly specced teltonika rutx50 is also supported nowadays. RUTX50 has only 1 modem and 2 sim card slots similiarly as this router.

I would be really interested in the GL-X3000 support too. How could I try adding a support for it?