Ysurac/openmptcprouter #2906: RUTX12 maximum bandwith

alexisries, 2023-07-30:
I would like to aggregate two 4G connections with each 200Mbps max and a Starlink connection with 200Mbps max.

Which is a total of 500-600Mbps under best conditions, I would like a router that is able to handle that load.

The router that interests me is the Teltonika RUTX12 because it is an all-in-one router (two 4G modems, Wifi, WAN port).

My question is this RUTX12 router capable of providing 500Mbps bandwidth with 3 WAN interfaces and the basic configuration (Shadowshock + Gloryrun) ?

If not, what bandwidth can I get with this router ?

Thank you

suyuan168, 2023-07-31:
The soc is probably ipq40xx. It uses a 717mhz cpu and can probably handle 100mbps of bandwidth.
However, I am going to launch a rm520ngl communication module with hardware with m2 interface, 3 mobile phone card interfaces, 1 2.5g interface, and 2 1G interfaces. You can customize any combination of WAN or LAN, the SOC is IPQ8072A, with a battery of 6800ma.
Probably able to handle 1000mbps bandwidth.
Maybe it will be sold, maybe, because I'm in China, I haven't figured it out yet, maybe.

alexisries, 2023-07-31:
> The soc is probably ipq40xx. It uses a 717mhz cpu and can probably handle 100mbps of bandwidth. However, I am going to launch a rm520ngl communication module with hardware with m2 interface, 3 [...]

OK thanks for your answer .
Your communication module is interesting, don't hesitate to contact me if you decide to sell it in Europe.

closed 2023-07-31 by alexisries