Ysurac/openmptcprouter #2909: How do i actually use OMR on a raspberry pi 4

amarmashruwala, 2023-08-02:
The raspbeery pi 4 only has 1 network port. Now the static IP for OMR is 192.168.100.1. So how do i actually use this? Do i connect my network to a switch and the raspberry pi to the switch as well? then do i connect 4G, 5G modems to the raspberry pi USB ports? and does it tether to an iphone? 

thank you.

Ysurac, 2023-08-02:
On a RPI4 you need to connect all to same switch (sometimes there is a switch on some modems/box). To "isolate" networks macvlan is used and OpenMPTCProuter create some "virtual" interfaces.
USB modems need to be connected on a powered USB hub (one USB modem can sometimes work on RPI4 USB but more than one need a powered USB hub).
iPhone USB tethering is supported.

amarmashruwala, 2023-08-02:
Ok thank you 
So if I understand this correctly, I would connect the raspberry pi to a switch and the WAN network to the switch as well? + additional 4G 5G modems to a powered usb hub + iPhone to the raspberry pi usb port for tethering?

Ysurac, 2023-08-03:
Exactly

amarmashruwala, 2023-08-03:
So far the hardest things has been configuring OMR. Are there any tutorials on how to configure OMR to use with a server on AWS?

Ysurac, 2023-08-03:
Should not be difficult, you only need to enter data in the wizard: https://github.com/Ysurac/openmptcprouter/wiki/OpenMPTCProuter-configuration
Else you can follow videos: https://github.com/Ysurac/openmptcprouter/wiki/Tutorials

hairy-dog, 2023-08-07:
It really isn't difficult. Basically, you simply follow the wizard and it all works. 

All the devices are physically connected to the same network and the PI4 does the magic in keeping them separate.

You need to give the wan devices fixed IPs and you need to switch off any other DHCP servers on the lan so the PI does the DHCP allocations.

Once you have it working with the wizard-recommendef settings, leave it working a while before meddling. Only make one change at a time and give each change 20 minutes to settle in before doing anything else.

closed 2023-11-11 by github-actions[bot]