Ysurac/openmptcprouter #2913: Wifi support

bagou4502, 2023-08-03:
Hello I just want to know if it's possible to add a wifi interface to the aggregated connexion instead of a eth interface?

Ysurac, 2023-08-04:
Hi,Yes.
Interface need to be created via Network->Wireless menu and added to "WAN" firewall zone.

bagou4502, 2023-08-04:
But this is for add a wireless hotspot not for ask opentcprouter to connect to a wifi network and use it for the connexion agregation no?

Ysurac, 2023-08-04:
No, if you scan and connect to a network this will create a new device that you can use with OpenMPTCProuter as connection.

hairy-dog, 2023-09-05:
I am not sure what is being asked for, so here's my overview:

The OMR device (a Pi4 on my system) is the DHCP server for the local network, so it controls what happens on the lan.

When you set things up, each device is connected (by cable or wi-fi) tp te same lan, but the OMR box decides which way to route each request. 

A. You can add a wi-fi device as an access point, allowing other devices to connect to the lan through it.

B. You can add a wi-fi device as a WAN connection (providing it has an internet connection of its own) so that the OMR box can send data through it to the remote VPS (connections combiner) 

Which of these two (A or B) are you asking about?

closed 2023-12-10 by github-actions[bot]