Ysurac/openmptcprouter #2914: Redirecting Outbound Traffic over second VPS not working

iEddie-cmd, 2023-08-04:
I have a current version of openmptcprouter and debian 10 (With install script). Bonding and port forwarding are working now. But I have problems to send the traffic of my client 192.168.100.207, through the tunnel to the VPS over to a second VPS 93.90.100.100 (proxy).

As I said, port forwarding 93.90.100.100 > 45.129.200.200 > (tunnel) > 192.168.100.1 > 192.168.100.207 works.

My VPS looks like this:
###Working portfoward. outbound questionable
sudo iptables -t nat -A PREROUTING -p tcp ! --dport 22 -j DNAT --to-destination 45.129.200.200 &&
sudo iptables -t nat -A POSTROUTING -j MASQUERADE &&
sudo iptables -A FORWARD -d 45.129.200.200 -p tcp ! --dport 22 -j ACCEPT &&
sudo iptables -t nat -A POSTROUTING -p tcp -s 45.129.200.200 -j SNAT --to-source 93.90.100.100 &&
sudo iptables -A FORWARD -s 45.129.200.200 -p tcp -j ACCEPT

I had defined outbound traffic under Firewall at router site - NAT Rules, but this does not work:

Protocol: TCP
Outbound zone: VPN
Source address: 192.168.100.207
SNAT: 93.90.100.100

How do I get the traffic to my VPS 93.90.100.100? 

PS. Nested VPN works conditionally. Neither ovpn nor wireguard had sufficient performance for me.

Ysurac, 2023-08-04:
Firewall configuration on VPS side must be done via Shorewall.
No changes should be needed on router side (router only need to know first VPS).
It may be easier to have VPS OMR part on last proxy 93.90.100.100 and use iptables on 45.129.200.200 to forward all ports > 65000 to 93.90.100.100

iEddie-cmd, 2023-08-06:
Thank you very much for your response. For that I would need a snat to be able to identify traffic coming from a specific client. But I am not able to change it for some reason. Traffic on gt-tun is still coming from 10.255.255.2. 

My Outgoing snat was for Interface vpn, source 192.168.100.207. Rewriting ip also 192.168.100.207. But it seems it doesn't apply to my traffic at all.

Ysurac, 2023-08-06:
VPN traffic will always be from router IP so 10.255.255.2 when GLorytun TCP is used.
It's why I was saying that you can redirect all traffic in the range 65000-65535 to your other VPS. You can filter with your public IP if it's fixed IP, else maybe filter by range.

iEddie-cmd, 2023-08-06:
Thank you. I will try it out and come back with the result. The issue is, that I have 10+ clients at home (Opnsense) Where every Client has to get routed over a specific public vps. And every of my client has to be able to get all ports 1 to 65000. So routing based on ports is unfortunately not an option.

iEddie-cmd, 2023-08-11:
Hi,
no unfortunately not. My problem was that I could not find out which traffic comes from which client. I did not find out by SNAT or packet marks. I then set up a separate instance of OpenMPTCProuter for each IP (VPS). I am testing this right now. 

If you found a solution, it would be great if you could share it with us.