Ysurac/openmptcprouter #2915: Can I use Dynamic DNS like No-IP instead of the IP for the VPS?

vempire-ghost, 2023-08-04:
I am using Google Cloud for my VPS, and since they charge per hour used, whenever I'm not using it, I shut down my VPS. As a result, the IP changes every time I start it again. To make it easier, I set up a DNS on no-ip and in the OMR, I used my DNS instead of the IP, so I don't have to configure the new IP every time I start my VPS.
I would like to know if this can cause any issues with the OMR?

Ysurac, 2023-08-04:
This is not tested. I think I would have to add a domain IP test to refresh IP in configurations.

vempire-ghost, 2023-08-04:
If I start the OMR before the VPS, when the DNS updates to the new IP, it can connect but cannot transmit any data. All connections get stuck on the MPTCP screen. However, if I restart the OMR or start it after the DNS has the new IP of the VPS, then it seems to work without any issues.

Ysurac, 2023-08-05:
ok. I have to add a test to refresh VPS IP from domain when needed. I will work on that.

vempire-ghost, 2023-11-03:
@Ysurac did you made something about this?

Ysurac, 2023-11-04:
Yes, should be done.
I just added xray/v2ray new protocols support to that.

vempire-ghost, 2023-11-06:
When I change the VPS IP while OMR is running, OMR can no longer connect to the VPS even after the no-ip has updated the domain to the new IP. For some reason, OMR keeps giving messages in the log that it couldn't connect to my address. It's as if it doesn't attempt to update the DNS and only looks from the cache. The only way to regain access is by restarting OMR.

vempire-ghost, 2023-11-06:
I always try to use the last snapshot available here https://snapshots.openmptcprouter.com/
Is there a away to see if I need to update my VPS?

closed 2024-02-09 by github-actions[bot]