Ysurac/openmptcprouter #2916: Port Forwarding, how to forward from OMR VPS to a web server behind another LAN Router

ioogithub, 2023-08-06:
I want to open forward port 80 from the `VPN` to `OMR` to a `Web server.` `Web server` is behind another OpenWrt `LAN Router`:
Web site:80 -> VPS:80 -> OMR:80 -> Lan Router:80 -> Web server

## Current Behavior

Packets are forwarded from VPS to the OMR VPS but they do not hit the LAN router (next hop)

- On OMR: `tcpdump -i tun0 port 80` I see the packets here.
- On OMR `tcpdump -i eth0 port 80` no packets here
- On LAN Router: `tcpdump -i eth1 port 80` where eth1 is the port on the LAN router which connects to OMR, no packets.
- One Web server: `tcpdump -i eth0 port 80`, no packets.

- I followed the guide here: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding.
- I setup one Port forward rule and one SNAT rule. 
- This guide only shows how to setup an SNAT to forward when the webserver is on the same subnet as OMR.
- My webserver is on a different subnet.
- I modified the SNAT rule `destination address ` changed from webserver to IP address of the LAN router on the OMR subnet:

```
OMR                         LAN Router                  Web server
192.168.2.1 <-> 192.168.2.6 (WAN)| 192.168.1.1 (LAN) <-> 192.168.1.2

```
Question: Can I use SNAT like this to forward from OMR vps to LAN router interface or do I have to use another method to forward packets to the LAN router where the web server is connected.
- I also added a forward rule from LAN router to web server on LAN router but not

I spent many hours troubleshooting but  packets never make it past the OMR router. Please help.

## Debug:
- Check that the VPN is running in System->OpenMPTCProuter and Status tab. `Yes`
- Check on the VPS in /etc/shorewall/rules that rules you set are available `Yes`
- Check on the VPS in /etc/shorewall/params.vpn that OMR_ADDR is the router side VPN IP (10.255.255.2 for glorytun TCP, the default VPN) `Yes`
- Check on the router if packets are coming on port 8080 from the VPN for example: tcpdump -i tun0 port 8080 `Yes`

If all is ok, then packets come to the router. Set a SNAT rule if not already done else check your firewall rules. `Yes but packets don't make it to the next router.` Please help.

## Specifications
  - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version: don't know how to find this.
  - OpenMPTCProuter VPS provider: vultr
  - OpenMPTCProuter platform: x86_64

Ysurac, 2023-08-06:
As you set it's on another subnet, destination answer to ping from the router ? If not, it must answer and you have to add a new LAN interface in the destination subnet.

ioogithub, 2023-08-06:
Thank you for troubleshooting help. This is what I see from pings:

LAN router is a gateway between OMR router and LAN network. It has two IP addresses:
1. LAN IP address: 192.168.1.1
2. LAN router IP address on OMR network (aka WAN): 192.168.2.6
3. OMR router IP address is: 192.168.2.1

- from OMR router: ping LAN IP address: fail
- from OMR router: ping LAN OMR IP address: ok
- from OMR router: ping destination webserver: fail

> you have to add a new LAN interface in the destination subnet.
Do you mean add a LAN interface on the OMR router for the destination subnet?

The traffic already works going out of network. Computers on the destination subnet can go out to the internet and access websites. This works:
- PC -> LAN Router -> OMR (via VPN) -> VPS -> Website

Also, if I disable VPS and run OMR in direct  access mode this works:
- PC -> LAN router -> OMR -> WAN1 -> Website

Traffic coming in doesn't work:
- PC on Internet -> VPS -> OMR (via VPN) -> xx

I thought if the traffic can go out then it should be an issue with the firewall? If this is wrong thinking please let me know so I can start researching how to add  LAN interface as you said.

Ysurac, 2023-08-06:
In all case OMR must be able to contact what you wxant to reach.
To add a new LAN interface this can be done using the wizard or in Network->Interface menu and set interface in LAN firewall zone.

ioogithub, 2023-08-06:
okay thanks, I also tried to create a static route but this did not work:
- before the change `ping reply "4 packets transmitted, 0 received, 100% packet loss, time 3082ms`
- after the change I get 'From 192.168.2.6 icmp_seq=2 Destination Port Unreachable'
so I guess this approach didn't work.

I have been researching creating the interface.

The LAN router already has an IP address on the OMR network via the OMR DHCP (192.168.2.6). Will this be a conflict?
Do I need a new subnet IP range in the OMR network or do I use the same range of the LAN network?

For example:
- LAN subnet is: 192.168.1.x
- OMR is subnet is: 192.168.2.x
- so for this new interface do I have to make 192.168.3.x?

Also what do I put for the default gateway of this new interface? Should it be the IP address assigned by OMR DHCP for the IP address of the LAN router? 192.168.2.6 or something else?

ioogithub, 2023-08-06:
I can ping the web server (192.168.1.2) form OMR!

I did it with two settings
1. static route: 192.168.1.0/24 to 192.168.2.6.
2. I had to modify wan zone on LAN router to forward traffic to destination WAN (it was blocked).

From OMR documentation I enables 2 rules:
1. Forward port rule, same settings as screenshot
3. SNAT rule with destination address as the www server (192.168.1.2).

Back to the original problem:
- OMR: tcpdump -i tun0 port 80 i see packets
- OMR: tcpdump -i eth0 port 80 no packets here
- LAN router :tcpdump -i eth1 port 80 no packets
- www server: tcpdump -i eth0 port 80 no packets

So the last hop I see packets in tun0 on OMR. If I look at eth0 which i the gateway port between OMR and LAN router no packets and no packets after this.

If I enter the VPN IP:80 into a web browser I get OMR login but really I want to forward the packets to the www server. I do not need OMR available on the internet. Now that I can ping the www how can I forward port 80 to the www server?

ioogithub, 2023-08-06:
@Ysurac: Working on this all day no success.

I can `nc` connect from OMR to www, send messages and trace the packet every hop :

- On OMR: `nc 192.168.1.222 80`, connection established, send message
- On OMR: `tcpdump -i tun0 port 80`, see packet
- On LAN router (WAN): `tcpdump -i eth0 port 80`, see packet
- On LAN router (LAN): `tcpdump -i eth1 port 80`, see packet
- On `www: nc -l -p 80` get message

but when I implement the two rules from this page: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding and do a curl http://domain:80 I can't get from OMR `tun0` to OMR `eth0`. It always stops here and I get served the OMR luci interface. If I change to another port 8000 I never get past tun0. I tried to disable the firewall, didn;t work. 

Why can an nc TCP packet work but curl request for webpage never gets past tun0. What else can I try?

Ysurac, 2023-08-07:
I need the result of ```uci show firewall```, ```ip a``` and ```ip r``` via SSH from the router.

ioogithub, 2023-08-07:
Thank you, here is what I have.
- 
## ip a
```
default via 10.255.255.1 dev tun0 
default metric 1 
        nexthop via 100.64.x.x dev eth1 weight 2 
        nexthop via 10.0.y.y dev eth2 weight 1 
default via 100.64.x.x dev eth1 metric 3 
default via 10.0.y.y dev eth2 metric 4 
default via 10.255.255.1 dev tun0 metric 1200 
10.0.0.0/24 via 10.0.y.y dev eth2 
10.0.0.0/24 dev eth2 scope link metric 4 
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
10.255.255.2 dev tun0 scope link metric 1200 
b.b.b.b dev eth1 proto static scope link src a.a.a.a metric 3 
100.64.0.0/10 dev eth1 scope link metric 3 
127.0.0.0/8 dev lo proto static scope link metric 6 
c.c.c.c metric 1 
        nexthop via 100.64.x.x dev eth1 weight 10 
        nexthop via 10.0.y.y dev eth2 weight 1
192.168.1.2 via 192.168.2.aaa dev eth0 
```
- Note 1: I redacted the ip addresses:
Note 2: OMR: (192.168.2.1), LAN router (192.168.1.1 and 192.168.2.aaa),  www server (192.168.1.2)
- Note2: I set a static route 192.168.1.2 (www server on LAN) via 192.168.2.aaa (LAN router IP assigned by OMR DHCP), it is pingable.

## ip r
```
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: ip6tnl0@NONE: <NOARP,80000> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd ::
3: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
4: gre0@NONE: <NOARP,80000> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
5: gretap0@NONE: <BROADCAST,MULTICAST,80000> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
6: erspan0@NONE: <BROADCAST,MULTICAST,80000> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: ip6gre0@NONE: <NOARP,80000> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd ::
8: bond0: <BROADCAST,MULTICAST,MASTER,80000> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether ee:ee:ee:ee:ee:ee brd ff:ff:ff:ff:ff:ff
9: teql0: <NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
10: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 qdisc mq state UP group default qlen 100
    link/ether dd:dd:dd:dd:dd:dd brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.1/24 brd 192.168.2.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 b1b1:b1b1:b1b1::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 c1c1::c1c1:c1c1:c1c1:c1c1/64 scope link 
       valid_lft forever preferred_lft forever
11: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 100
    link/ether cc:cc:cc:cc:cc:cc brd ff:ff:ff:ff:ff:ff
    inet 100.64.x.x/10 brd 100.x.255.255 scope global eth1
       valid_lft forever preferred_lft forever
12: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1460 qdisc mq state UP group default qlen 100
    link/ether bb:bb:bb:bb:bb:bb brd ff:ff:ff:ff:ff:ff
    inet 10.0.y.y/24 brd 10.0.0.255 scope global eth2
       valid_lft forever preferred_lft forever
13: eth3: <BROADCAST,MULTICAST,80000> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether aa:aa:aa:aa:a:aa brd ff:ff:ff:ff:ff:ff
24: tun0: <POINTOPOINT,MULTICAST,NOARP,PROMISC,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 100
[... 261 lines cut ...]
- OMR: `tcpdump -i tun0 port 80`, packet here <--- stops here
- OMR: `tcpdump -i eth0 port 80`, no packet
- LAN router WAN: no packet
- LAN router LAN: no packet
- www server: no packet
- www apache log: no hit

### 2nd solution try:
If I deactivate the SNAT rule and modify the Port forward from:
`firewall.@redirect[0].dest_ip='192.168.2.1'` to
`firewall.@redirect[0].dest_ip='192.168.1.2'`

I get this result:
- OMR: `tcpdump -i tun0 port 80`, packet here
- OMR: `tcpdump -i eth0 port 80`, packet here <---- stop here
- LAN router WAN `tcpdump -i eth1 port 80`: no packet
- LAN router LAN: no packet
- www server: no packet
- www apache log: no hit

The packet makes it one more hop, passed though OMR router WAN port but `tcpdump -i eth1 port 80` on the Openwrt LAN router never sees the packet, I don't know is it in a format the LAN router cannot accept? I tried setting various combination of port forward, traffic rule etc but it never gets to the www server.

I want to use the first solution from the OMR doc but the packet never leaves OMR.

Thanks for looking this over, I appreciate it.

closed 2023-08-21 by ioogithub