Ysurac/openmptcprouter #2917: VPN is closed and STOPPED tun0  Frequent with one wifi interface

fareign, 2023-08-07:
## Expected Behavior
I use WIFI MT7922AM as a interface to test if it can work well.
[image]

then I find Tun0 stopped frequent.

### routerlog

Aug  7 03:57:02 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Aug  7 03:57:02 OpenMPTCProuter daemon.err glorytun[15754]: 47.106.225.139.65001: connected
Aug  7 03:57:04 OpenMPTCProuter user.notice post-tracking-post-tracking: Set firewall on server vps
Aug  7 03:57:12 OpenMPTCProuter daemon.err glorytun[15754]: 47.106.225.139.65001: key exchange failed
Aug  7 03:57:12 OpenMPTCProuter daemon.info glorytun[15754]: STOPPED tun0
Aug  7 03:57:13 OpenMPTCProuter daemon.err glorytun[15754]: 47.106.225.139.65001: connected
Aug  7 03:57:23 OpenMPTCProuter daemon.err glorytun[15754]: 47.106.225.139.65001: key exchange failed
Aug  7 03:57:23 OpenMPTCProuter daemon.info glorytun[15754]: STOPPED tun0
Aug  7 03:57:24 OpenMPTCProuter daemon.err glorytun[15754]: 47.106.225.139.65001: connected
Aug  7 03:57:25 OpenMPTCProuter daemon.info glorytun[15754]: STARTED tun0
Aug  7 03:57:25 OpenMPTCProuter user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Aug  7 03:58:59 OpenMPTCProuter user.notice post-tracking-post-tracking: omrvpn down because gateway down
Aug  7 03:58:59 OpenMPTCProuter user.notice post-tracking-post-tracking: Glorytun VPN down, restart it
Aug  7 03:58:59 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
Aug  7 03:59:04 OpenMPTCProuter daemon.info procd: Instance glorytun::instance1 pid 15754 not stopped on SIGTERM, sending SIGKILL instead
Aug  7 03:59:04 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Aug  7 03:59:04 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Aug  7 03:59:04 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Aug  7 03:59:04 OpenMPTCProuter daemon.notice ttyd[13749]: [2023/08/07 03:59:04:9482] N: rops_handle_POLLIN_netlink: DELADDR
Aug  7 03:59:04 OpenMPTCProuter daemon.notice ttyd[13749]: [2023/08/07 03:59:04:9492] N: rops_handle_POLLIN_netlink: DELADDR
Aug  7 03:59:04 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Aug  7 03:59:05 OpenMPTCProuter daemon.err glorytun[29797]: MPTCP upstream enabled
Aug  7 03:59:05 OpenMPTCProuter daemon.info glorytun[29797]: INITIALIZED tun0
Aug  7 03:59:05 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Aug  7 03:59:05 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Aug  7 03:59:05 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Aug  7 03:59:05 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Aug  7 03:59:05 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Aug  7 03:59:05 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Aug  7 03:59:05 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Aug  7 03:59:05 OpenMPTCProuter daemon.notice ttyd[13749]: [2023/08/07 03:59:05:8422] N: rops_handle_POLLIN_netlink: DELADDR
Aug  7 03:59:05 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Aug  7 03:59:06 OpenMPTCProuter daemon.err glorytun[29797]: 47.106.225.139.65001: connected
Aug  7 03:59:14 OpenMPTCProuter user.notice post-tracking-post-tracking: Set firewall on server vps
Aug  7 03:59:16 OpenMPTCProuter daemon.err glorytun[29797]: 47.106.225.139.65001: key exchange failed
Aug  7 03:59:16 OpenMPTCProuter daemon.info glorytun[29797]: STOPPED tun0
Aug  7 03:59:17 OpenMPTCProuter daemon.err glorytun[29797]: 47.106.225.139.65001: connected
Aug  7 03:59:27 OpenMPTCProuter daemon.err glorytun[29797]: 47.106.225.139.65001: key exchange failed
Aug  7 03:59:27 OpenMPTCProuter daemon.info glorytun[29797]: STOPPED tun0
Aug  7 03:59:28 OpenMPTCProuter daemon.err glorytun[29797]: 47.106.225.139.65001: connected
Aug  7 03:59:28 OpenMPTCProuter daemon.info glorytun[29797]: STARTED tun0
Aug  7 03:59:28 OpenMPTCProuter user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Aug  7 04:03:47 OpenMPTCProuter user.notice post-tracking-post-tracking: omrvpn down because gateway down
Aug  7 04:03:47 OpenMPTCProuter user.notice post-tracking-post-tracking: Glorytun VPN down, restart it
Aug  7 04:03:47 OpenMPTCProuter daemon.info glorytun[29797]: STOPPED tun0
Aug  7 04:03:47 OpenMPTCProuter daemon.notice ttyd[13749]: [2023/08/07 04:03:47:2686] N: rops_handle_POLLIN_netlink: DELADDR
Aug  7 04:03:47 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Aug  7 04:03:47 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Aug  7 04:03:47 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Aug  7 04:03:47 OpenMPTCProuter daemon.notice ttyd[13749]: [2023/08/07 04:03:47:2695] N: rops_handle_POLLIN_netlink: DELADDR
Aug  7 04:03:47 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
Aug  7 04:03:47 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Aug  7 04:03:47 OpenMPTCProuter daemon.err glorytun[26785]: MPTCP upstream enabled
Aug  7 04:03:47 OpenMPTCProuter daemon.info glorytun[26785]: INITIALIZED tun0
Aug  7 04:03:47 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Aug  7 04:03:47 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Aug  7 04:03:47 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Aug  7 04:03:47 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Aug  7 04:03:47 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Aug  7 04:03:47 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Aug  7 04:03:47 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Aug  7 04:03:48 OpenMPTCProuter daemon.notice ttyd[13749]: [2023/08/07 04:03:48:1462] N: rops_handle_POLLIN_netlink: DELADDR
[... 180 lines cut ...]
wifi:MT7922AM
wifi driver:linux firmware 2023-0625,mt76 driver is 2023-05-13

all of setting is defaut

Push flow is SRT with 1080P 3mb

middle load ping from router
[image]

idle ping from router
[image]

another test:
if I close OMR, I do speedtest. it works well.
if open OMR do speedtest .then wifi will be broken down
[image]

## Specifications

  - OpenMPTCProuter version: 0.59.2 lastest 5.15 kernel
  - OpenMPTCProuter VPS version: 0.1029 upstream
  - OpenMPTCProuter VPS provider:aliyun
  - OpenMPTCProuter platform:  x86_64 
  - Country:cn

Ysurac, 2023-08-07:
You should use 6.1 kernel on this alpha release, 5.15 kernel will not be a release.
Did you enabled SQM with download and upload speed ?

fareign, 2023-08-07:
> You should use 6.1 kernel on this alpha release, 5.15 kernel will not be a release. Did you enabled SQM with download and upload speed ?

enabled SQM but not enable sqm-atuorate ,I speed not settinng is 0.

fareign, 2023-08-07:
I test enable or disable sqm ,enable sqm-autorate or disable . is the same .
I think this may help us to find the reason which cause VPN dislink from VPS.
the ping  packages lost TX = 11059，RX = 11025，lost = 34 (0% ).
最短 = 0ms，最长 = 117ms，average = 4ms

Ysurac, 2023-08-07:
Can be due to saturation of the link, it's what SQM is made for (with download/upload speed value set)
Can be due to driver bug (use 6.1 based alpha release)
Can be due to a bad wifi link

fareign, 2023-08-07:
test with QCA6390 is OK.
 must be something with wifi. but wifi link looks well. 
if I use RTMP, nothing VPN also closed, but every works well .beacuase no traffic use VPN.

closed 2023-08-29 by fareign