Ysurac/openmptcprouter #2920: omr-service[214035]: Error: Nexthop has invalid gateway.

fareign, 2023-08-08:
## Expected Behavior
VPS LOG
Aug  8 15:01:26 iZwz9j49zfsqj6oedh7guqZ omr-service[196742]: Error: Nexthop has invalid gateway.
Aug  8 15:01:37 iZwz9j49zfsqj6oedh7guqZ omr-service[197345]: Error: Nexthop has invalid gateway.
Aug  8 15:01:47 iZwz9j49zfsqj6oedh7guqZ omr-service[197789]: Error: Nexthop has invalid gateway.
Aug  8 15:01:58 iZwz9j49zfsqj6oedh7guqZ omr-service[198220]: Error: Nexthop has invalid gateway.
Aug  8 15:02:05 iZwz9j49zfsqj6oedh7guqZ systemd[1]: Started Session 16 of user root.
Aug  8 15:02:09 iZwz9j49zfsqj6oedh7guqZ omr-service[198736]: Error: Nexthop has invalid gateway.
Aug  8 15:02:19 iZwz9j49zfsqj6oedh7guqZ omr-service[199462]: Error: Nexthop has invalid gateway.
Aug  8 15:02:28 iZwz9j49zfsqj6oedh7guqZ systemd[1]: session-16.scope: Succeeded.
Aug  8 15:02:30 iZwz9j49zfsqj6oedh7guqZ omr-service[199957]: Error: Nexthop has invalid gateway.
Aug  8 15:02:37 iZwz9j49zfsqj6oedh7guqZ systemd[1]: Started Session 17 of user root.
Aug  8 15:02:40 iZwz9j49zfsqj6oedh7guqZ omr-service[200442]: Error: Nexthop has invalid gateway.
Aug  8 15:02:51 iZwz9j49zfsqj6oedh7guqZ omr-service[200960]: Error: Nexthop has invalid gateway.
Aug  8 15:03:02 iZwz9j49zfsqj6oedh7guqZ omr-service[201480]: Error: Nexthop has invalid gateway.
Aug  8 15:03:12 iZwz9j49zfsqj6oedh7guqZ omr-service[201988]: Error: Nexthop has invalid gateway.
Aug  8 15:03:23 iZwz9j49zfsqj6oedh7guqZ omr-service[202477]: Error: Nexthop has invalid gateway.
Aug  8 15:03:34 iZwz9j49zfsqj6oedh7guqZ omr-service[202995]: Error: Nexthop has invalid gateway.
Aug  8 15:03:44 iZwz9j49zfsqj6oedh7guqZ omr-service[203507]: Error: Nexthop has invalid gateway.

## Current Behavior

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.59.2alpha 6.1 kernel
  - OpenMPTCProuter VPS version: 0.1029test
  - OpenMPTCProuter VPS provider:aliyun
  - OpenMPTCProuter platform:  x86_64 
  - Country: cn

Ysurac, 2023-08-08:
Maybe a problem in ```/etc/openmptcprouter-vps-config/omr-admin-config.json``` in vpnremoteip or lanips sections.
Else do you have multiples public IPs on your VPS ?

fareign, 2023-08-08:
is the openvpntcp.but I dont know how to solve it.
                "vpn": "openvpn",
                "vpnremoteip": "10.255.252.6",
                "vpnlocalip": "10.255.252.1",
                "ula": "fdd0:f60f:4061::/48",
                "lanips": [
                    "192.168.100.1/255.255.255.0"
                ],
                "proxy": "shadowsocks"

Ysurac, 2023-08-08:
You are using openvpn as VPN ?  Not tested since a long time, I will test.

fareign, 2023-08-08:
haha , everything have a test. I find if network is not well glorytun offen down.

the ip is10.255.255.2  /255.1 ?

Ysurac, 2023-08-08:
There is issues with Glorytun TCP on 6.1 kernel, I'm working on it.
For now, Glorytun UDP seems to work better but need download and upload speed settings.
For glorytun TCP it's 10.255.255.1 on VPS and .2 on router.

fareign, 2023-08-14:
@Kalimeiro @Ysurac 

10.255.252.0/24 via 10.255.252.2 dev tun0 
10.255.252.2 dev tun0 proto kernel scope link src 10.255.252.1 
10.255.253.0/24 dev mlvpn0 proto kernel scope link src 10.255.253.1 

"vpn": "openvpn",
"vpnremoteip": "10.255.252.6",
"vpnlocalip": "10.255.252.1"

is must be 10.255.252.2,but it will change by itself after a few seconds

Ysurac, 2023-08-14:
IP on router side when you use OpenVPN TCP as VPN is really 10.255.252.6 by default. You can check with ```ip a``` via SSH on router side, and  ```ping 10.255.252.6``` on VPS side.
The NextHop error is fixed in snapshot VPS script (this happen when route is set but VPN link is not up or not responding correctly).

fareign, 2023-08-14:
you are right. so it is not a importent ERR.

> IP on router side when you use OpenVPN TCP as VPN is really 10.255.252.6 by default. You can check with `ip a` via SSH on router side, and `ping 10.255.252.6` on VPS side. The NextHop error is fixed [...]

ioogithub, 2023-10-26:
I am trying to use openvpn tcp but I am getting the next hop error and it does not correct itself:

```
Oct 26 13:10:04 vpstest omr-service[19891]: Error: Nexthop has invalid gateway.
Oct 26 13:10:15 vpstest omr-service[20588]: Error: Nexthop has invalid gateway.
Oct 26 13:10:26 vpstest omr-service[21245]: Error: Nexthop has invalid gateway.
Oct 26 13:10:37 vpstest omr-service[21982]: Error: Nexthop has invalid gateway.
Oct 26 13:10:48 vpstest omr-service[22639]: Error: Nexthop has invalid gateway.
Oct 26 13:10:59 vpstest omr-service[23328]: Error: Nexthop has invalid gateway.
Oct 26 13:11:09 vpstest omr-service[23986]: Error: Nexthop has invalid gateway.
Oct 26 13:11:20 vpstest omr-service[24670]: Error: Nexthop has invalid gateway.
Oct 26 13:11:31 vpstest omr-service[25350]: Error: Nexthop has invalid gateway.
Oct 26 13:11:42 vpstest omr-service[26118]: Error: Nexthop has invalid gateway.
Oct 26 13:11:53 vpstest omr-service[26776]: Error: Nexthop has invalid gateway.
Oct 26 13:12:04 vpstest omr-service[27445]: Error: Nexthop has invalid gateway.
Oct 26 13:12:15 vpstest omr-service[28166]: Error: Nexthop has invalid gateway.
Oct 26 13:12:26 vpstest omr-service[28801]: Error: Nexthop has invalid gateway.
Oct 26 13:12:37 vpstest omr-service[29562]: Error: Nexthop has invalid gateway.
Oct 26 13:12:48 vpstest omr-service[30197]: Error: Nexthop has invalid gateway.
Oct 26 13:12:58 vpstest omr-service[30873]: Error: Nexthop has invalid gateway.
Oct 26 13:13:09 vpstest omr-service[31569]: Error: Nexthop has invalid gateway.
Oct 26 13:13:20 vpstest omr-service[32235]: Error: Nexthop has invalid gateway.
Oct 26 13:13:31 vpstest omr-service[32916]: Error: Nexthop has invalid gateway.
Oct 26 13:13:42 vpstest omr-service[33670]: Error: Nexthop has invalid gateway.
Oct 26 13:13:53 vpstest omr-service[34327]: Error: Nexthop has invalid gateway.
Oct 26 13:14:04 vpstest omr-service[35020]: Error: Nexthop has invalid gateway.
Oct 26 13:14:15 vpstest omr-service[35717]: Error: Nexthop has invalid gateway.
Oct 26 13:14:26 vpstest omr-service[36351]: Error: Nexthop has invalid gateway.
Oct 26 13:14:37 vpstest omr-service[37078]: Error: Nexthop has invalid gateway.
Oct 26 13:14:47 vpstest omr-service[37745]: Error: Nexthop has invalid gateway.
Oct 26 13:14:58 vpstest omr-service[38433]: Error: Nexthop has invalid gateway.
Oct 26 13:15:09 vpstest omr-service[39108]: Error: Nexthop has invalid gateway.
Oct 26 13:15:20 vpstest omr-service[39796]: Error: Nexthop has invalid gateway.
Oct 26 13:15:31 vpstest omr-service[40459]: Error: Nexthop has invalid gateway.
Oct 26 13:15:42 vpstest omr-service[41197]: Error: Nexthop has invalid gateway.
Oct 26 13:15:53 vpstest omr-service[41854]: Error: Nexthop has invalid gateway.
Oct 26 13:16:04 vpstest omr-service[42560]: Error: Nexthop has invalid gateway.
Oct 26 13:16:15 vpstest omr-service[43204]: Error: Nexthop has invalid gateway.
Oct 26 13:16:26 vpstest omr-service[43894]: Error: Nexthop has invalid gateway.
Oct 26 13:16:36 vpstest omr-service[44643]: Error: Nexthop has invalid gateway.
Oct 26 13:16:47 vpstest omr-service[45317]: Error: Nexthop has invalid gateway.
```

which interface does openvpn use, it sounds like this is a known issue, how can i fix it?

[Ysurac](https://github.com/Ysurac) is it fixed with stable or only with beta1? I installed with the script less than a month ago. If not can you give some information on how to fix it manually?

ioogithub, 2023-10-26:
> @Kalimeiro @Ysurac
>
> 10.255.252.0/24 via 10.255.252.2 dev tun0 10.255.252.2 dev tun0 proto kernel scope link src 10.255.252.1 10.255.253.0/24 dev mlvpn0 proto kernel scope link src 10.255.253.1 [...]

What exactly must be 10.255.255.2
Mine is definatly not fixing itself after a few seconds, it always throws the error every few seconds indefinitely.

ioogithub, 2023-10-27:
@Ysurac can you comment on your statement:

"The NextHop error is fixed in snapshot VPS script"

I spend an hour yesterday searching for a commit and looking at all the files mentioned in this thread but I can't find the change I need to make to get openvpn working on stable. Right now I do not believe it will work.

I want to try openvpn because glorytun and dsvpn have connection issues when there is a problem and they do not always recover without intervention. I am seeking a stable VPN.

closed 2024-01-31 by github-actions[bot]