Ysurac/openmptcprouter #2921: how to change this dns?

fareign, 2023-08-09:
## Expected Behavior
[image]

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2023-08-09:
They are used only for OMR-Tracker to check connection status.

fareign, 2023-08-09:
> They are used only for OMR-Tracker to check connection status.

but I change all DNS in OMR-tracker which can't open or PING. not find this.

Ysurac, 2023-08-09:
114dns is also used if you select China in the wizard, this can be modified in Network->DHCP&DNS, it's 114.114.114.114
For the 8.8.8.8, can be used by SQM-Autorate in snapshot.

fareign, 2023-08-09:
I try to ping them,but none of them can reply.

[image]
[image]

Ysurac, 2023-08-09:
makaki.miuku.net and time.neu.edu.cn are NTP time servers.

fareign, 2023-08-09:
OK,thank you!

closed 2023-08-29 by fareign