Ysurac/openmptcprouter #2924: v2ray & openvpntcp work much better than glorytun

fareign, 2023-08-10:
OMR version:0.59.2-kernel6.1-VPS0.1029

after I changed the dns.it may a litter better.

ss+glorytun udp with setting SQM rate. seems a litter better than without.but it may course a bad download rate.about 2-3mb(two LTE card max download maybe 20mb)

v2ray+openvpntcp SQM setting to 0. maybe tcp and udp all use V2ray. but it work very well. the same network I push up about 1 hour nerver broken down.

Ysurac, 2023-08-10:
There is an issue with SQM, I'm already working on this.

thelastdreamer, 2023-08-13:
> There is an issue with SQM, I'm already working on this.

hello @Ysurac thanks a lot for the amazing project! 
when this is fixed should we re-install our vps or the raspberry pi/pc which we use for the client/server?

or is there any way to update to the latest fix via update?

closed 2023-08-29 by fareign