Ysurac/openmptcprouter #2926: I need help for first installation please

DarthVaderFR, 2023-08-11:
## Expected Behavior

I would be able to reach the RPI image IP 192.168.100.1

## Current Behavior

I tried to install different versions, with different softwares on 2 different type of USB Keys, but still have the same issue. The program starts, but not completely because i can't join the admin page, and it seems that the network isn't created. The leds are not nlinking on the RPI, and whatever the time i'm waiting for, the situation remains the same. IP is not pingable here is the picture of what i have on screen : 

[image]

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4-r0 (64 and 32 bits)
  - OpenMPTCProuter VPS version: not tried yet
  - OpenMPTCProuter VPS provider: same
  - OpenMPTCProuter platform: RPI4B

Ysurac, 2023-08-12:
Never tried a screen on RPI. This seems to be started here, press enter and you should see prompt.

DarthVaderFR, 2023-08-12:
Hello,

thank you for the answer. Hitting enter or any other key don’t produce anything. Another symptom, when i connect directly the pi to my computer the leds are fixed (orange/green) when i connect it to a switch (that one is up to 100mbits only) inonly have the orange led up and fixed. I also tried to connect the rpi to a « blank nerwork » where he will be alone with a router dhcp disabled, and my pc, or to connect it on my home nerwork provided by a freebox delta with dhcp « on ». The result is the same. Should i take an older version ? Or do you have any other idea to help me to solve my issue please ?

Ysurac, 2023-08-12:
Latest stable release is working.
Make sure your power supply is powerful, remove all USB device on the RPI, check sd card, try another program to write the image.

DarthVaderFR, 2023-08-12:
I already powered up the rpi with the raspbian os and it worked, plugged with mouse, keyboard and screen. But i also powered up the pi with your image in « standalone » (ethernet, and power plugged only) same problem. But i’m using a usb key. Should it be the problem ? I tried 2 different usb keys

Ysurac, 2023-08-12:
OMR use CPU at full speed it can work on raspbian but not on OMR.
Why do you use an usb key? You should use an SD card to boot, else a powered hub may be needed.

DarthVaderFR, 2023-08-14:
Hello,Well you were right. I plugged the SD card and it worked. Thanks for the advice. Does that mean that i’ll need to buy a selfpowered usb hub to plug an iphone for sharing connection (and charging as well) and plug a little 4g router usb as well ? In this configuration, do you think that the router will be able to identify the both connections on the hub ?Thank again :)

Ysurac, 2023-08-14:
USB on RPI often doesn't give enough power to devices, should be ok for one USB device (work for me with a 4G dongle) but for more than one or anything that need too much power an USB hub is required.
No problem to use any router/usb key/... on an USB powered hub.

ignatsolovey, 2023-10-15:
@DarthVaderFR I highly recommend  TP-Link UH700 or UH720 powered USB hubs. Not because I'm endorsing them or something, it just the best solution for the cause, tried and tested specifically for OMR by many people. 
My own UH720 hub powers and connects 4 USB-stick 4G modems that do the bonding, the Pi 4B itself, and has one more charging port open for other uses (like, to kick some juice into my own phone if not anything else). As well, it's 4 modems only because of their size, thus 3 ports out of 7 are usually unused. If I use extension leads that have a port-wide USB plug each, I can do 7 modems. Which does not make sense because there are only four cellular carriers in my area, and connecting two modems of one carrier that both hook onto one base station (which is mostly the case) does not increase speed or reliability.

closed 2024-01-18 by github-actions[bot]