Ysurac/openmptcprouter #2928: glorytun can't work for srt(UDP) at The lastest version with kernel 6.1

fareign, 2023-08-16:
after push SRT flow a few seconds. VPN adn VPS,broken down.

Ysurac, 2023-08-16:
Please use the template when you open an issue.
Do you have SQM enabled with download and upload speed set ?

Ysurac, 2023-08-16:
Do you mean Glorytun TCP or Glorytun UDP ? If glorytun TCP did you update VPS ?

fareign, 2023-08-16:
is glorytun tcp.  I give up glorytun UDP, because it use more traffic than other VPN.
I find another thing.  glorytun tcp, openvpn tcp which need mptcp ,can not work on multipath,only one interface.
and glorytun udp,mlvpn.v2ray can work on multipath.
so there is something with mptcp?

fareign, 2023-08-16:
[image]

Ysurac, 2023-08-16:
Glorytun TCP is for now really slow and aggregate only sometimes. I need to find why...
For OpenVPN TCP, I didn't patch OpenVPN yet to add MPTCP support.

fareign, 2023-08-16:
so now ss+mlvpn ,v2ray+mlvpn or openvpn. is the better choice.

psfabio, 2023-08-16:
SRT works on UDP, you can try to use v2ray as proxy without any vpn.

fareign, 2023-08-17:
> SRT works on UDP, you can try to use v2ray as proxy without any vpn.

thanks,V2ray now work well,but cant without VPN.must select one ,even if vpn not work.
I have to compare ss+vpn and V2ray.which is more satble.

fareign, 2023-08-17:
mlvpn buffer size is an important setting ,if set 0, it may use more tracffic . set too high(8000) may cant work.
I test ,if buffer size is full. it may use more traffic.so it only use for small stream <3mb.

closed 2023-08-29 by fareign