Ysurac/openmptcprouter #2931: Port forwarding not working with v2ray, works with shadowsocks

ioogithub, 2023-08-21:
## Expected Behavior
Following the documentation here: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding-using-v2ray, forward a TCP port, it should just work.

## Current Behavior
Port forward is working with shadowsocks but it does not work with v2ray. v2ray is working for outgoing traffic.

## Specifications
- OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c](https://github.com/ysurac/openmptcprouter)
- OpenMPTCProuter VPS version: v0.59.1
- OpenMPTCProuter VPS provider: vultr
- OpenMPTCProuter platform: x86_64

## Configs

After I set port to forward using OMR interface and the 'use V2RAY' checkbox I see this config:

1. v2ray route and incoming tags added:

`/etc/v2ray/v2ray-server.json`

```
routing:
{
                "type": "field",
                "inboundTag": [
                    "openmptcprouter_redir_tcp_5555_to_192.168.100.150:19233"
                ],
                "outboundTag": "OMRLan"
            },
}
inbounds:
{
            "tag": "openmptcprouter_redir_tcp_5555_to_192.168.100.150:19233",
            "port": "19233",
            "protocol": "dokodemo-door",
            "settings": {
                "network": "tcp",
                "port": "19233",
                "address": "192.168.100.150"
            }
        },
```
2. Shorewall policy updated with DNAT rule:

`/etc/shorewall/policy`

```
ACCEPT   net    $FW    tcp   19233    # OMR openmptcprouter open router 19233 port tcp --- V2Ray to 192.168.100.150:19233
```

## Test:
From client: `curl domain.com:19233`

Result:
`Failed to connect to host.com port 19233 after 784 ms: Connection refused`

## Troubleshooting:
1. iptables net-fw shows packets arriving.

`iptables -L net-fw -v`
```
Chain net-fw (1 references)
 pkts bytes target     prot opt in     out     source               destination         

45  2700  ACCEPT     tcp  --  any    any     anywhere             anywhere             tcp dpt:19233
```

2. tcpdump shows packets arriving to VPS

`tcpdump -i eth0 port 19233
`
```
03:38:19.715414 IP host.com.55412 > 172.16.0.7.19233: Flags [S], seq 2393207102, win 64240, options [mss 1284,sackOK,TS val 2515860846 ecr 0,nop,wscale 7], length 0
03:38:19.715527 IP 172.16.0.7.19233 > host.com.55412: Flags [R.], seq 0, ack 2393207103, win 0, length 0
```

3. Run tcpdump on OMR

`tcpdump -i eth0 port 19233`

```
packet never arrives on OMR.
```

So packet arrives at VPS, then what? Is v2ray supposed to pick up the packet and forward it to OMR?

I set `"loglevel": "debug"` in both `config.json` and `v2ray-server.json` but no errors. Am I missing something? How do I get v2ray to pick up the packets and forward them to a server via OMR?

ioogithub, 2023-08-21:
Solution found.

The config file is read by v2ray but it is symbolically linked to another file. This was broken on my vps:

/etc/v2ray/config.json -> /etc/v2ray/v2ray-server.json.

This exposed another bug. When you use OMR to add a v2ray port forward, v2ray-server.json is updated however it is updated incorrectly. the port field needs to be an int16 but it is updated with a string.

This is what is updated
"port:" "19322"

This is what is required
"port:" 19322

log shows:
`main: failed to read config files: [/etc/v2ray/config.json] > infra/conf: failed to load inbound detour config. > json: cannot unmarshal string into Go struct field DokodemoConfig.port of type uint16`

The only solution currently is to manually edit the json file and reboot both OMR and VPS. After this it will work however you will need to do this everytime a port forwarded is updated.

[Ysurac](https://github.com/Ysurac): Please switch this tag to "bug" to track this issue.

ioogithub, 2023-08-23:
So this bug is a bigger problem than I thought. I expected that I could just edit the config.json file after making any changes to port forwarding manually but actually, if the vps is rebooted, some how it writes the values out again wrong. This means that even in this stable version is unstable as you are just one reboot away from breaking your system every time. I just experienced an hour of downtime due to these bugs speaking back in.

Other than running the alpha version is there another way to address this issue? @Ysurac: Is there any other work around? As far as I can tell the latest current version is stable and I do not want to run the alpha snapshot which risks introducing other problems.

Is no one else running v2ray, why isn't this a bigger known problem? The port-forward documentation on the wiki doesn't mention it so if a user follows it, they will spend many hours frustrated that it doesn't work.

ioogithub, 2023-08-23:
I'm actually not sure if it was rebooting OMR or rebooting the VPS which rewrote the bad values now. I think it might have been rebooting OMR. I am struggling to login to the VPS because this bug doesn't just break port forwarding, it breaks the proxy because v2ray will not start afterwards.

ioogithub, 2023-08-23:
Okay so it seems to be that rebooting the VPS will write the bad values and break the proxy but now I don't know how to fix it. I can edit config.json manually and restart v2ray but I can't get the traffic to return to the proxy once v2ray is restarted. I thought that if I restarted the vpn again it would shift the traffic back to the proxy but it doesn't work. Traffic that runs over the VPN is so slow it is unusable-n 200kbps. I really need the proxy to work and keep working. Any ideas?

Ysurac, 2023-08-23:
I "backported" latest omr-admin script to VPS stable script. To force traffic back to V2Ray, you can try to validate wizard or firewall again. Or do a ```/etc/init.d/openmptcprouter-vps restart``` via SSH on the router.

Ysurac, 2023-08-23:
The openmptcprouter-vps script synchronize the router and VPS config.
In anycase you shouldn't have port forwarding using VPN when V2ray checkbox is used, the only case the checkbox is ignored is when the proxy set in the wizard is not v2ray.

ioogithub, 2023-08-23:
Also, I am not using port forwarding to vpn and v2ray proxy at the same time. It wouldn't work because DNAT rules are given a higher priority in shorewall so it would always forward to the VPN which definitely not what you want if you are trying to use v2ray.

closed 2023-08-23 by ioogithub