Ysurac/openmptcprouter #2933: new VPS update yesterday can work

fareign, 2023-08-23:
## Expected Behavior
1.23100474922259d00a8c0c4b00a0c8de89202cf9 is not a tree can't install
2.I change it back . but CPU always 100%. python3 used all of them.

[image]

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: VPs the latest update

Ysurac, 2023-08-23:
Which VPS script ? What is the command line used ? You have some errors on install ?

Ysurac, 2023-08-23:
I fixed issue with Glorytun-UDP install.
I'm not able to reproduce issue with CPU usage, what is the script ? omr-admin ? Can you provide the output of ```journalctl -u omr-admin``` ?

fareign, 2023-08-23:
> I fixed issue with Glorytun-UDP install. I'm not able to reproduce issue with CPU usage, what is the script ? omr-admin ? Can you provide the output of `journalctl -u omr-admin` ? (omr-admin can have high CPU if speedtests are enabled in wizard during these speedtests)

I Released the VPs.tomorrow install a new one.

fareign, 2023-08-23:
Aug 24 01:38:23 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Stopped OMR-Admin.
Aug 24 01:38:23 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Consumed 1.852s CPU time.
Aug 24 01:38:23 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Started OMR-Admin.
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[8848]: Traceback (most recent call last):
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[8848]:   File "/usr/local/bin/omr-admin.py", line 2641, in <module>
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[8848]:     async def speedtestul(file: UploadFile | None = None):
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[8848]: TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Consumed 1.656s CPU time.
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 4.
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Stopped OMR-Admin.
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Consumed 1.656s CPU time.
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Started OMR-Admin.
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Stopping OMR-Admin...
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Succeeded.
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Stopped OMR-Admin.
Aug 24 01:38:27 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Started OMR-Admin.
Aug 24 01:38:31 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[9183]: Traceback (most recent call last):
Aug 24 01:38:31 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[9183]:   File "/usr/local/bin/omr-admin.py", line 2641, in <module>
Aug 24 01:38:31 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[9183]:     async def speedtestul(file: UploadFile | None = None):
Aug 24 01:38:31 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[9183]: TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'
Aug 24 01:38:31 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Aug 24 01:38:31 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Aug 24 01:38:31 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Consumed 1.892s CPU time.
Aug 24 01:38:32 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 1.
Aug 24 01:38:32 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Stopped OMR-Admin.
Aug 24 01:38:32 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Consumed 1.892s CPU time.
Aug 24 01:38:32 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Started OMR-Admin.
Aug 24 01:38:35 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[9258]: Traceback (most recent call last):
Aug 24 01:38:35 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[9258]:   File "/usr/local/bin/omr-admin.py", line 2641, in <module>
Aug 24 01:38:35 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[9258]:     async def speedtestul(file: UploadFile | None = None):
Aug 24 01:38:35 iZj6c24tb3w7akvsoh5np6Z omr-admin.py[9258]: TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'
Aug 24 01:38:35 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Aug 24 01:38:35 iZj6c24tb3w7akvsoh5np6Z systemd[1]: omr-admin.service: Failed with result 'exit-code'.

Ysurac, 2023-08-23:
Should be fixed in latest VPS script.

psfabio, 2023-08-23:
> Should be fixed in latest VPS script.

which is the correct place to download the latest vps script?

https://www.openmptcprouter.com/server/debian-x86_64.sh
or
http://www.openmptcprouter.com/server-test/debian.sh
or something else?

Thanks a lot

kinkyninja, 2023-08-23:
Default VPS installation on Debian 10 doesn't work.
[image]

Last n rows from system logs:
Wed Aug 23 20:14:35 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: connected
Wed Aug 23 20:14:35 2023 daemon.err summator_agent[598]: 2023/08/23 20:14:35 Skipping 25 more seconds to let everything start up...
Wed Aug 23 20:14:35 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: key exchange failed
Wed Aug 23 20:14:35 2023 daemon.info glorytun[30491]: STOPPED tun0
Wed Aug 23 20:14:36 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: connected
Wed Aug 23 20:14:36 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: key exchange failed
Wed Aug 23 20:14:36 2023 daemon.info glorytun[30491]: STOPPED tun0
Wed Aug 23 20:14:36 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: connected
Wed Aug 23 20:14:36 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: key exchange failed
Wed Aug 23 20:14:36 2023 daemon.info glorytun[30491]: STOPPED tun0
Wed Aug 23 20:14:37 2023 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, server API answer)
Wed Aug 23 20:14:37 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: connected
Wed Aug 23 20:14:37 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: key exchange failed
Wed Aug 23 20:14:37 2023 daemon.info glorytun[30491]: STOPPED tun0
Wed Aug 23 20:14:38 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: connected
Wed Aug 23 20:14:38 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: key exchange failed
Wed Aug 23 20:14:38 2023 daemon.info glorytun[30491]: STOPPED tun0
Wed Aug 23 20:14:38 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: connected
Wed Aug 23 20:14:38 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: key exchange failed
Wed Aug 23 20:14:38 2023 daemon.info glorytun[30491]: STOPPED tun0
Wed Aug 23 20:14:39 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: connected
Wed Aug 23 20:14:39 2023 daemon.err glorytun[30491]: 185.251.88.172.65001: key exchange failed
Wed Aug 23 20:14:39 2023 daemon.info glorytun[30491]: STOPPED tun0

Ysurac, 2023-08-23:
@psfabio If you use stable image: ```wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | sh``` but the issue was about snapshot image script so using server-test URL as indicated in the doc

@kinkyninja It's not related to this issue and you seems to use very old release, so update both VPS and router.

kinkyninja, 2023-08-23:
Thanks. Is my old v57 router incompatible with new VPS updates? 

A week ago I successfully installed and ran new VPS with my v57 router. But now there is an error, first time in 2 years. (I need this 57 version because I use kinda forked version for my own purposes)

Ysurac, 2023-08-23:
@kinkyninja at least it's not tested with old releases, so no idea if this can work or not. Here you seems to not be able to join API on VPS, first check log on VPS about omr-admin, if you have same log as this issue, run VPS script again.