Ysurac/openmptcprouter #2937: Can't forward ports

DarthVaderFR, 2023-08-25:
## Expected Behavior
Hello, again, (and sorry if my english is not perfect, i'm French)

I'm trying to forward some ports to my NAS, and i tried to follow different steps of your tutorials, in different ways, i'm not managing to forward them.

First i tried to directly redirect them from the port forwardings straight from the VPS to the device : NOK
Then i tried to use the SNAT on the same ports : NOK
I decided to restart the router, and it crashed completely.
I reinstalled and reconfigured everything from scratch because i didn't save anything
I retried the same thing : NOK
I tried the ports from the VPS to the router ip : NOK
Then i recreated the SNAT with th router ip : NOK

I tried to check the conf fil on the VPS you mentionned in the tutorial : /etc/shorewall/rules
Rules are not written.
I tried to switch all ports to the router : nothing happened neither
I rechecked that file, and the lines mentionned all the ports below 65000 redirected to the router still are in comments mode.

I have the feeling that the file can't be written, but no error on the router interface when i save and apply the configuration

## Current Behavior

I can't reach the web interface of my NAS, and my different web services on it are down.

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4.r0
  - OpenMPTCProuter VPS version: Debian 11
  - OpenMPTCProuter VPS provider: Firstheberg
  - OpenMPTCProuter platform: RPI4B

Ysurac, 2023-08-25:
All is screen in status page without errors ?
What is the result of ```uci show firewall``` from the router via SSH ?

DarthVaderFR, 2023-08-25:
Hello and thanks for your help.

Here is the result of the command :

oot@OpenMPTCProuter:~# uci show firewall
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].input='REJECT'
firewall.@defaults[0].output='REJECT'
firewall.zone_lan=zone
firewall.zone_lan.name='lan'
firewall.zone_lan.input='ACCEPT'
firewall.zone_lan.output='ACCEPT'
firewall.zone_lan.forward='ACCEPT'
firewall.zone_lan.mtu_fix='1'
firewall.zone_lan.auto_helper='1'
firewall.zone_lan.network='lan'
firewall.zone_wan=zone
firewall.zone_wan.name='wan'
firewall.zone_wan.input='REJECT'
firewall.zone_wan.output='ACCEPT'
firewall.zone_wan.forward='REJECT'
firewall.zone_wan.masq='1'
firewall.zone_wan.mtu_fix='1'
firewall.zone_wan.auto_helper='1'
firewall.zone_wan.network='wan1' 'iPhoneAlexandre'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
firewall.@rule[0].family='ipv4'
firewall.@rule[1]=rule
firewall.@rule[1].name='Allow-Ping'
firewall.@rule[1].src='wan'
firewall.@rule[1].proto='icmp'
firewall.@rule[1].icmp_type='echo-request'
firewall.@rule[1].family='ipv4'
firewall.@rule[1].target='ACCEPT'
firewall.@rule[2]=rule
firewall.@rule[2].name='Allow-IGMP'
firewall.@rule[2].src='wan'
firewall.@rule[2].proto='igmp'
firewall.@rule[2].family='ipv4'
firewall.@rule[2].target='ACCEPT'
firewall.@rule[3]=rule
firewall.@rule[3].name='Allow-DHCPv6'
firewall.@rule[3].src='wan'
firewall.@rule[3].proto='udp'
firewall.@rule[3].dest_port='546'
firewall.@rule[3].family='ipv6'
firewall.@rule[3].target='ACCEPT'
firewall.@rule[4]=rule
firewall.@rule[4].name='Allow-MLD'
firewall.@rule[4].src='wan'
firewall.@rule[4].proto='icmp'
firewall.@rule[4].src_ip='fe80::/10'
firewall.@rule[4].icmp_type='130/0' '131/0' '132/0' '143/0'
firewall.@rule[4].family='ipv6'
firewall.@rule[4].target='ACCEPT'
firewall.@rule[5]=rule
firewall.@rule[5].name='Allow-ICMPv6-Forward'
firewall.@rule[5].src='wan'
firewall.@rule[5].dest='*'
firewall.@rule[5].proto='icmp'
[... 365 lines cut ...]
firewall.@nat[6].snat_ip='192.168.100.1'
firewall.@nat[6].dest_ip='192.168.100.139'
firewall.@nat[7]=nat
firewall.@nat[7].name='NAS'
firewall.@nat[7].proto='tcp' 'udp'
firewall.@nat[7].src='lan'
firewall.@nat[7].src_port='443'
firewall.@nat[7].dest_port='443'
firewall.@nat[7].target='SNAT'
firewall.@nat[7].snat_ip='192.168.100.1'
firewall.@nat[7].dest_ip='192.168.100.139'
firewall.@nat[8]=nat
firewall.@nat[8].name='NAS'
firewall.@nat[8].proto='tcp' 'udp'
firewall.@nat[8].src='lan'
firewall.@nat[8].src_port='6001-6010'
firewall.@nat[8].dest_port='6001-6010'
firewall.@nat[8].target='SNAT'
firewall.@nat[8].snat_ip='192.168.100.1'
firewall.@nat[8].dest_ip='192.168.100.139'
root@OpenMPTCProuter:~# 

Here is the screen of the status. Seems no problem there

[image]

Ysurac, 2023-08-27:
So not any related rule in /etc/shorewall/rules ?
What do you have in /var/log/daemon.log on the VPS related to omr-admin ?
What is the result of ```/etc/init.d/openmptcprouter-vps set_firewall``` on the router, via SSH ?

DarthVaderFR, 2023-08-27:
Hello, and thanks for helping me.

Here are the answers for the questions :

So not any related rule in /etc/shorewall/rules ?
Nop ☹

[image]

What do you have in /var/log/daemon.log on the VPS related to omr-admin ?

Start/stop service excepted i only have this : 
 
[image]

What is the result of /etc/init.d/openmptcprouter-vps set_firewall on the router, via SSH ?

[image]
 

Thank you again

Ysurac, 2023-08-27:
ok, the problem is on VPS side. It seems that you have a problem with ```/etc/openmptcprouter-vps-admin/omr-admin-config.json``` file. I think it's empty, can you check ?
If it's the case you will have to remove this file, run script again using "OMR_ADMIN_PASS=xxx" parameter where xxx is your current VPS key.
I will add automatic backup of this file, I think API was not able to write file to disk correctly.

DarthVaderFR, 2023-08-27:
actually it's not empty, but is it complete ? :

[image]
 (i removed what i thought sensitive)

Ysurac, 2023-08-28:
Yes it's correct but missing vpn and proxy information.
What do you have when you run via SSH on the VPS ```systemctl restart omr-admin``` and on the router via SSH the correct command (sorry) ```/etc/init.d/openmptcprouter-vps set_vps_firewall``` ? and then check end of log on VPS using ```journalctl -u omr-admin``` ?

DarthVaderFR, 2023-08-29:
I think that i was at the end :

[image]

Ysurac, 2023-08-29:
There is something strange... Can you give me the result of ```mount``` via SSH on the VPS ?

DarthVaderFR, 2023-08-29:
here is the result :

[image]

Ysurac, 2023-08-29:
partition "/" is read only. This explain all problems.
I've had these problems on FirstHeberg before...
You can reboot the server or maybe a ```mount -o remount,rw /``` may work.

DarthVaderFR, 2023-08-29:
reboot has solved my problem to access to my NAS from the exterior. BIG THANK YOU AGAIN.

But, (always a but ^^) it seems that download station on my NAS, is unable to download anything via HTTP(S) any idea ? I'm guessing that it uses 80/443 forwarded but... Still not connecting. And i think that the problem maybe be the same for my rustdesk server. I can't remote access to/from the internet.

DarthVaderFR, 2023-08-31:
Hello,

I've solved my problem, i don't know how really i've managed but one of the problem came from a third party provider who doesn't accept ip adress from VPS.

Ysurac, 2023-08-31:
Yes, it's often the case... They use some block list that have most VPS ranges.

DarthVaderFR, 2023-09-03:
Hello,

This is me again, stil trying to bypass that restriction, and i maybe found a way. I have a VPN plan with ExpressVPN, that allows me to get an openvpn fil to import inside of a router or linux etc... I've added to the router, i've managed to activate it, but i'm lonsing the connection. I'm trying the doc about the VPN but i don't. Could you help me with please ? What should i configure in the Services->Shadowsocks-libev->Redir rules->Destinations settings if i have anything to configure there ?

Thanks

Ysurac, 2023-09-04:
You can also services->OMR-ByPass to exit directly via a connection based on IP/domain/ASN/...

DarthVaderFR, 2023-09-04:
But if i do that, i'm losing the agreagation right ?

Ysurac, 2023-09-05:
Yes.
Else you need to find a VPS that doesn't have IP blocked, not really easy...

closed 2023-12-09 by github-actions[bot]