Ysurac/openmptcprouter #2938: V2ray no longer works after 25/08 snapshot.

vempire-ghost, 2023-08-26:
## Current Behavior
After installing today's snapshot on 08/25, V2ray no longer works in any mode, be it VLESS, MLESS, TROJAN, or SOCKS. If I select it, when I check the log, it shows this error on Aug 26 at 02:06:08:

"OpenMPTCProuter daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41.212.162, 1.0.0.1, 212.27.48.10)
Aug 26 02:06:09 OpenMPTCProuter user.notice v2ray: Rules DOWN."

Shadowsocks is working normally. I've tried restarting and selecting each of the protocols, but the error remains the same.

## Possible Solution
Do I need to update the VPS to support these new protocols?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59.2alpha-5.4 r0+16862-170
  - OpenMPTCProuter VPS version: openmptcprouter v0.59.2alpha-5.4 r0+16862-170
  - OpenMPTCProuter VPS provider: Google Cloud
  - OpenMPTCProuter platform: Virtual Box x86_x64
  - Country:

Ysurac, 2023-08-26:
Yes VPS need to be updated but "V2RAY VLESS" should work with previous config as it was the already used protocol.
If it doesn't work, can you run ```/etc/init.d/v2ray start``` via SSH on router part ?

vempire-ghost, 2023-08-26:
Indeed VLESS worked today.
Aug 26 13:42:06 OpenMPTCProuter daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)

I gona update the VPS to see if the others protocols start working, thanks.

vempire-ghost, 2023-08-26:
Updated the VPS and stoped to work again, when I run /etc/init.d/v2ray start give me this message.
root@OpenMPTCProuter:~# /etc/init.d/v2ray start
[info] DNS disabled: main_dns
[info] Setting transparent proxy on port: 1897
[info] Transparent proxy mode: default
Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'

Back to this message again.
Aug 26 14:15:00 OpenMPTCProuter daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164)
Aug 26 14:15:01 OpenMPTCProuter user.notice v2ray: Rules DOWN

vempire-ghost, 2023-08-26:
And after the VPS update, even the old snapshot stopped to work with V2Ray completely.
openmptcprouter v0.59.2alpha-5.4 r0+16819-1b6
Aug 26 14:19:11 OpenMPTCProuter daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41.212.162)
Aug 26 14:19:11 OpenMPTCProuter user.notice v2ray: Rules DOWN

root@OpenMPTCProuter:~# /etc/init.d/v2ray start
[info] DNS disabled: main_dns
[info] Setting transparent proxy on port: 1897
[info] Transparent proxy mode: default
Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'

Ysurac, 2023-08-26:
What do you have using ```journalctl -u v2ray``` at the end of the log on the VPS ?

vempire-ghost, 2023-08-26:
Aug 26 13:46:33 instance-3 v2ray[783]: 2023/08/26 13:46:33 186.249.199.xxx:51158 rejected  proxy/vless/encoding: invalid request user id
Aug 26 13:46:35 instance-3 v2ray[783]: 2023/08/26 13:46:35 186.249.199.xxx:52895 rejected  proxy/vless/encoding: invalid request user id
Aug 26 13:46:37 instance-3 v2ray[783]: 2023/08/26 13:46:37 186.249.199.xxx:50880 rejected  proxy/vless/encoding: invalid request user id
Aug 26 13:46:37 instance-3 v2ray[783]: 2023/08/26 13:46:37 186.249.199.xxx:52575 rejected  proxy/vless/encoding: invalid request user id
Aug 26 13:46:39 instance-3 v2ray[783]: 2023/08/26 13:46:39 186.249.199.xxx:50978 rejected  proxy/vless/encoding: invalid request user id
Aug 26 13:46:41 instance-3 v2ray[783]: 2023/08/26 13:46:41 186.249.199.xxx:50338 rejected  proxy/vless/encoding: invalid request user id
Aug 26 13:46:42 instance-3 v2ray[783]: 2023/08/26 13:46:42 186.249.199.xxx:50403 rejected  proxy/vless/encoding: invalid request user id
Aug 26 13:46:43 instance-3 v2ray[783]: 2023/08/26 13:46:43 186.249.199.xxx:50655 rejected  proxy/vless/encoding: invalid request user id
Aug 26 13:46:45 instance-3 v2ray[783]: 2023/08/26 13:46:45 186.249.199.xxx:50599 rejected  proxy/vless/encoding: invalid request user id
Aug 26 13:46:47 instance-3 v2ray[783]: 2023/08/26 13:46:47 186.249.199.xxx:53040 rejected  proxy/vless/encoding: invalid request user id
Aug 26 13:46:47 instance-3 v2ray[783]: 2023/08/26 13:46:47 186.249.199.xxx:49563 rejected  proxy/vless/encoding: invalid request user id

Ysurac, 2023-08-26:
You have same result in ```uci show v2ray.omrout.s_vless_user_id``` on router and in ```grep id /etc/v2ray/v2ray-server.json``` on server ?

vempire-ghost, 2023-08-26:
No, its different id in the router and in the vps.

Ysurac, 2023-08-27:
Replace the id on the VPS /etc/v2ray/f2ray-server.json file. I'm fixing VPS script.

vempire-ghost, 2023-08-27:
Thanks this solved the issue.

closed 2023-08-28 by Ysurac