Ysurac/openmptcprouter #2939: The latest update V2ray socks still not change

fareign, 2023-08-28:
## Expected Behavior
config outbound 'omrout'
        option tag 'omrout_tunnel'
        option protocol 'socks'
        option s_vmess_port '65230'
        option s_vmess_user_security 'aes-128-gcm'
        option s_vmess_user_alter_id '0'
        option s_vless_port '65228'
        option s_vless_user_security 'aes-128-gcm'
        option s_vless_user_encryption 'none'
        option s_vless_user_alter_id '0'
        option s_trojan_port '65229'
        option s_trojan_user_security 'aes-128-gcm'
        option s_trojan_user_encryption 'none'
        option s_trojan_user_alter_id '0'
        option s_socks_port '65229'

I use snapshots and buid by myself .is the same .

## Specifications

  - OpenMPTCProuter version: 0.59.2 6.1 latest 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

Ysurac, 2023-08-28:
If you mean for V2RAY socks port, it was fixed in this commit: https://github.com/Ysurac/openmptcprouter-feeds/commit/844807f25a434629aeab9985c2079567d985764c
You can do a ```uci set v2ray.omrout.s_socks_port='65231' && uci commit``` to fix your current config.

fareign, 2023-08-28:
@Ysurac yes, I know you fix it,but I update new firmware without old setting,it still not change. so I don't know why.

fareign, 2023-08-28:
not effection . when I wizard setting . it change to 65229

Ysurac, 2023-08-28:
Sorry but should be fixed by https://github.com/Ysurac/openmptcprouter-feeds/commit/54403bd3e2d8f0953f313f59b1a4bf0ea622f82a
There is no others places this port is modified.

fareign, 2023-08-28:
I use uci set v2ray.omrout.s_socks_port='65231' && uci commit reboot  ,is OK.
so maybe luci-app -openmptcprouter.
I will try later,the new fix.thanks.

fareign, 2023-08-28:
@Ysurac is oK now. thanks.