Ysurac/openmptcprouter #2940: Wireguard Client on Openmptcrouter not work

MTSmhas, 2023-08-28:
I have the following problem. I am using OpenMPTCProuter for site-to-site networking. It works fine when I use one OpenMPTCProuter as a server and a normal OpenWRT device as a client. However, when I try to connect two OpenMPTCProuters, the client does not initiate the connection. The wg0 adapter only turns on once and then turns off again. Can someone help me?
With OpenVPN everything works normally, but it's unfortunately too slow.

Ysurac, 2023-08-28:
Please use template when you open an issue.

wg0 interface is used internally, you should use another interface.

MTSmhas, 2023-08-28:
Okay i have test with tun5 as name but don't work. It is the OpenMPTCProuter v0.59.1 x86 - 5.4

MTSmhas, 2023-08-28:
The First Router Open the Server. That's no problem. But the client with mptcp don't connect. I found a problem from wirguard with mwan3... i think the probelm is the similar: https://forum.openwrt.org/t/cant-get-wireguard-to-work-with-mwan3/80198

MTSmhas, 2023-08-28:
The mptcpovervpn option cant break a second client? that would be an idea that the script overwrites the client.

MTSmhas, 2023-08-28:
https://github.com/Ysurac/openmptcprouter/issues/2606  This is the error

MTSmhas, 2023-08-28:
Mon Aug 28 11:54:48 2023 daemon.notice netifd: Interface 'tun5' is setting up now
Mon Aug 28 11:54:48 2023 daemon.warn [8841]: <warn>  [tun5] invalid sysfs path read for net/tun5
Mon Aug 28 11:54:48 2023 daemon.warn [8841]: <warn>  [base-manager] couldn't handle kernel event: device net/tun5 not found
Mon Aug 28 11:54:49 2023 daemon.notice netifd: Interface 'tun5' is now down

MTSmhas, 2023-08-28:
okay i found the problem the omr-tracker del the wireguard connection.

MTSmhas, 2023-08-28:
@Ysurac 
now we have a working connection.
But one error ;) 
Mon Aug 28 19:01:44 2023 user.notice OMR-VPS: Setting server config error, try again
Mon Aug 28 19:01:48 2023 user.notice OMR-VPS: Setting server config error, try again

Can you tell me where a this command?

MTSmhas, 2023-08-28:
Is this message from vps or router?

Ysurac, 2023-08-29:
Can you make a doc or changes in default config you made ? Maybe I can make this changes as default and this may help

MTSmhas, 2023-08-29:
We Build a luci app for your Router. That makes it easier. But give me some more time. There are too many changes and too difficult for normal users. To write a doc. ^^

closed 2023-12-02 by github-actions[bot]