Ysurac/openmptcprouter #2943: Openmptcprouter-VPS Bug

MTSmhas, 2023-08-31:
[nf_conntrack-Full](https://github.com/Ysurac/openmptcprouter/assets/76729306/e7fb8962-2b45-4be4-9179-f36d4a89b429)

Hi i found a bug. If the nf_conntrack table is full you lost the connection to the vps.
I found a way to stop this bug.

Please add this to the VPS:  /etc/sysctl.conf 

net.netfilter.nf_conntrack_max = 524288
net.netfilter.nf_conntrack_tcp_timeout_close_wait = 60
net.netfilter.nf_conntrack_tcp_timeout_fin_wait = 60
net.netfilter.nf_conntrack_tcp_timeout_time_wait = 60

MTSmhas, 2023-08-31:
[image]

If the nf_conntrack table is full this bullshit starts.

I think that helps a lot of users.

Ysurac, 2023-08-31:
I modified nf_conntrack_max and nf_conntrack_buckets on VPS server test script, will be in next release.
I never reached that limit, you are doing P2P ?

MTSmhas, 2023-08-31:
We used multiple routers for connections between multiple data centers. And they are a lot of connection ^^