Ysurac/openmptcprouter #2944: VPS upstream6.1 erro

fareign, 2023-08-31:
2023-08-31T17:28:10.026040+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:28:10.029922+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:28:10.030499+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:28:10.031036+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:28:10.031619+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:28:10.032105+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:28:10.032642+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:28:10.033132+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:28:10.033653+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:28:10.037165+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/omr-admin.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:28:10.037746+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/omr-admin-ipv6.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:28:10.039811+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/iperf3.service is marked world-inaccessible. This has no effect as

### ##
the latest snapshot 6.1kernel  must use the same kernel VPS?

net.ipv4.tcp_available_congestion_control = reno cubic bbr  only three.

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

fareign, 2023-08-31:
2023-08-31T17:19:52.997015+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-firstboot.service:34: Invalid syntax, ignoring: passwd.shell.root
2023-08-31T17:19:52.997142+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-firstboot.service:35: Invalid syntax, ignoring: firstboot.locale
2023-08-31T17:19:52.997299+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-firstboot.service:36: Invalid syntax, ignoring: firstboot.locale-messages
2023-08-31T17:19:52.997457+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-firstboot.service:37: Invalid syntax, ignoring: firstboot.keymap
2023-08-31T17:19:52.997598+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-firstboot.service:38: Invalid syntax, ignoring: firstboot.timezone
2023-08-31T17:19:53.888560+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Reloading.
2023-08-31T17:19:54.141909+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-tmpfiles-clean.service:24: Invalid syntax, ignoring: tmpfiles.extra
2023-08-31T17:19:54.161570+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-tmpfiles-setup.service:26: Invalid syntax, ignoring: tmpfiles.extra
2023-08-31T17:19:54.161956+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-tmpfiles-setup.service:27: Invalid syntax, ignoring: login.motd
2023-08-31T17:19:54.162112+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-tmpfiles-setup.service:28: Invalid syntax, ignoring: login.issue
2023-08-31T17:19:54.162243+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-tmpfiles-setup.service:29: Invalid syntax, ignoring: network.hosts
2023-08-31T17:19:54.162386+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-tmpfiles-setup.service:30: Invalid syntax, ignoring: ssh.authorized_keys.root
2023-08-31T17:19:54.163217+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-tmpfiles-setup-dev.service:25: Invalid syntax, ignoring: tmpfiles.extra
2023-08-31T17:19:54.164107+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: /lib/systemd/system/systemd-sysusers.service:21: Unknown key name 'ConditionCredential' in section 'Unit'

fareign, 2023-08-31:
2023-08-31T17:24:57.720162+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/iperf3.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:24:57.831298+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Reloading.
2023-08-31T17:24:58.261921+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/iperf3.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:24:58.437192+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Reloading.
2023-08-31T17:24:58.912922+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/omr-admin.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:24:58.914848+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/iperf3.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:24:59.072137+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Reloading.
2023-08-31T17:24:59.520657+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/omr-admin.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:24:59.521874+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/omr-admin-ipv6.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:24:59.523065+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/iperf3.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:24:59.872870+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Reloading.
2023-08-31T17:25:00.298724+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/omr-admin.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:25:00.299256+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/omr-admin-ipv6.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
2023-08-31T17:25:00.300408+08:00 iZj6c24tb3w7akvsoh5np6Z systemd[1]: Configuration file /lib/systemd/system/iperf3.service is marked world-inaccessible. This has no effect as

Ysurac, 2023-08-31:
I fixed chmod on system file I install. For the "Invalid Syntax" errors, it's not file I modify, what is the Linux version used ?
If router use 6.1 version, VPS should have kernel >= 6.1.0
There is less congestion control available on VPS side as a default distribution kernel is used (in any case there is less congestion control than in 5.4).

fareign, 2023-08-31:
debian 11.7

fareign, 2023-08-31:
when I use debian12 ,the "Invalid Syntax" errors gone.

but also has many warning.

Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/omr-admin.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/omr-admin-ipv6.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/iperf3.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Reloading.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/dsvpn-server@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: /lib/systemd/system/v2ray.service:15: Unknown key name 'StartLimitIntervalSec' in section 'Service', ignoring.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: /etc/systemd/system/tat_agent.service:7: PIDFile= references a path below legacy directory /var/run/, updating /var/run/tat_agent.pid → /run/tat_agent.pid; please update the unit file accordingly.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:14 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
[... cut ...] strictions. Proceeding anyway.
Aug 31 19:43:15 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.
Aug 31 19:43:15 VM-16-11-debian systemd[1]: Configuration file /lib/systemd/system/openvpn@.service is marked world-inaccessible. This has no effect as configuration data is accessible via APIs without restrictions. Proceeding anyway.

fareign, 2023-08-31:
Aug 31 19:43:15 VM-16-11-debian systemd[1]: /etc/systemd/system/tat_agent.service:7: PIDFile= references a path below legacy directory /var/run/, updating /var/run/tat_agent.pid → /run/tat_agent.pid; please update the unit file accordingly.

fareign, 2023-08-31:
Aug 31 19:44:00 VM-16-11-debian systemd[1]: iperf3.service: Failed with result 'exit-code'.

fareign, 2023-08-31:
Aug 31 19:44:04 VM-16-11-debian systemd-modules-load[245]: Failed to find module 'mptcp_olia'
Aug 31 19:44:04 VM-16-11-debian systemd-modules-load[245]: Failed to find module 'mptcp_wvegas'
Aug 31 19:44:04 VM-16-11-debian systemd-modules-load[245]: Failed to find module 'mptcp_balia'

fareign, 2023-08-31:
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: using tcp fast open
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: UDP relay enabled
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: enable TCP no-delay
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: initializing ciphers... chacha20-ietf-poly1305
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: tcp server listening at [::0]:65101
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: tcp port reuse enabled
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: fast open is not supported on this platform
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: tcp server listening at 0.0.0.0:65101
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: tcp port reuse enabled
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: udp server listening at [::0]:65101
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: udp port reuse enabled
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: udp server listening at 0.0.0.0:65101
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: udp port reuse enabled
Aug 31 19:46:08 VM-16-11-debian ss-server[1041]: running from root user
Aug 31 19:46:08 VM-16-11-debian ovpn-bonding7[899]: DEPRECATED OPTION: --cipher set to 'AES-256-CBC' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations.
Aug 31 19:46:08 VM-16-11-debian ovpn-bonding8[900]: DEPRECATED OPTION: --cipher set to 'AES-256-CBC' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations.
Aug 31 19:46:08 VM-16-11-debian ovpn-bonding8[900]: Note: dev-type not tun, disabling data channel offload.
Aug 31 19:46:08 VM-16-11-debian ovpn-bonding8[900]: OpenVPN 2.6.3 x86_64-pc-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [PKCS11] [MH/PKTINFO] [AEAD] [DCO]
Aug 31 19:46:08 VM-16-11-debian ovpn-bonding8[900]: library versions: OpenSSL 3.0.9 30 May 2023, LZO 2.10
Aug 31 19:46:08 VM-16-11-debian ovpn-bonding8[900]: DCO version: N/A
Aug 31 19:46:08 VM-16-11-debian ovpn-bonding8[900]: net_route_v4_best_gw query: dst 0.0.0.0
Aug 31 19:46:08 VM-16-11-debian ovpn-bonding8[900]: net_route_v4_best_gw result: via 172.19.16.1 dev eth0
Aug 31 19:46:08 VM-16-11-debian ovpn-bonding3[889]: DEPRECATED OPTION: --cipher set to 'AES-256-CBC' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations.

fareign, 2023-08-31:
> What is the result of `uname -a` and `multipath` on the VPS ? Router is also using 6.1 ? What do you have in status page ? Proxy and VPN using default values ? SQM enabled or not ?

router is 6.1 kernel. every mode is the same. change to 5.4 kernel VPS  everything is OK.now I will install another VPS on aliyun. and have a test.
[image]
[image]
[image]

fareign, 2023-08-31:
sometimes Wan1 become to yellow. wan1 is not 4G or 5G。is home fiber optic network。

fareign, 2023-08-31:
I dont like SQM with download and upload speed setting ,because it never work good . I do some test early.
[image]
if diffrent interface may cause this promblem.
[image]
if I set SQM value,it may slow slow slow to the same .
[image]
set to BBR.may have a hight speed.
[image]
set back to cubic test again.

Ysurac, 2023-08-31:
There is now a speedtest checkbox in the wizard to help setting SQM, then after few minutes SQM is configured (still not perfect, but a bit better). 
So it's working correctly with BBR and not with cubic ? Did you test BBR2 ?

fareign, 2023-08-31:
I forget , not record. BBR2 maybe not high than bbr, a little stable.

closed 2023-12-04 by github-actions[bot]