Ysurac/openmptcprouter #2947: Changing OMR default IP address (Starlink related)

h45h3ncryp73d, 2023-09-01:
Before I start experimenting and mess things up ... Do I need to change default OMR address if Starlink is being used as one of the WAN routers? Has someone perhaps tried to put it in bridge mode with OMR?

Thanks.

Ysurac, 2023-09-01:
I use Starlink but with the old wifi router, so it's on 192.168.1.1. But in any case you can change default OpenMPTCProuter IP, 192.168.100.1 to anything you want and use Starlink without the router they provides.
You can see in the issues multiples users using Starlink without the provided router.

h45h3ncryp73d, 2023-09-01:
I guess OMR IP can be changed through Network -> Interfaces LAN/eth0 settings, right?

Ysurac, 2023-09-01:
Exactly or via System->OpenMPTCProuter in the wizard.

h45h3ncryp73d, 2023-09-01:
Thank you.

Regarding Starlink I'll have to check it out as from posted comments I've seen some have resolved it by using USB/Eth adapters which was my initial setup but I had problems with interfaces switching upon reboot so I gave up on it and went to switch/MacVLAN solution ... I still don't get it if dish itself has DHCP server or is it part of  their router ... I know I should try myself but in most cases a good advice can save a lot of time ...

Ysurac, 2023-09-01:
There is a DHCP server on the dish .

h45h3ncryp73d, 2023-09-01:
Yeah, seems to be just ONE IP address (enough for me and OMR) but it seems to be NAT one.
I found a good explanation at https://www.starlink-user.com/2022/01/18/bypass-mode-first-impressions

If I'll need to setup a static route Network -> Statc routes -> Add
I guess Interface would be a Starlink interface and IP address the one dish provided
Do I need a netmask and gateway or not?

I do understand basic networking but in most things I just a n00b ;-)

closed 2023-12-07 by github-actions[bot]

JeedHome44, 2024-03-17:
Hello,

If I understood correctly, it is therefore impossible to connect the StarLink adapter to the home LAN network?

Do we have to take an Ethernet to USB adapter to put directly on a USB port of the Raspberry Pi ORM?

And what bothers me the most is that I also have to change the OMR IP! But can I leave all my devices at 192.168.100.X and only change the OMR IP?

Ysurac, 2024-03-17:
I use Starlink with the provided ethernet router (old version) without issues.
If you use USB with a RPI you often need to use also a powered USB hub.
You should be able to use Starlink on 192.168.100.x using VLAN or using a subnet that only have Starlink in it.

JeedHome44, 2024-03-17:
> I use Starlink with the provided ethernet router (old version) without issues. If you use USB with a RPI you often need to use also a powered USB hub. You should be able to use Starlink on 192.168.100.x using VLAN or using a subnet that only have Starlink in it.

I did not understand very well. can we use a MacVLan for StarLink ?
Do I only need to connect StarLink to the home LAN network ?

Then change only the OMR IP to 192.168.X.1 and keep all my network devices in 192.168.100.X ?
So I could access the StarLink interface in 192.168.100.1 while keeping the configurations of all my devices in 192.168.100.X, right ?

Do you just have to put the StarLink router in By-Pass mode?

Ysurac, 2024-03-17:
Starlink have one IP with the router and one without. 192.168.1.1 is with the Starlink router, in this case you can connect it to the network and use it directly but Starlink have a DHCP that is can't be disabled (and I don't think there is a bypass mode)
Else you can use VLAN if you a manageable switch with VLAN support.
In 192.168.100.1, you should be able to use it directly, only set it in a small subnet /30 for example if possible.
Try to set it normally, in most case it should already work, then you can still adapt configuration.

JeedHome44, 2024-03-17:
There is a by-pass mode !

[image]

JeedHome44, 2024-03-18:
Hello,

You didn't answer my question

I have a bypass mode in my StarLink routeur.

Can I only change the IP of OMR to 192.168.101.1 for example and leave all devices in 192.168.100.X ?

There will only be OMR in DHCP on the LAN since StarLink will be in bypass !

And like that, 192.168.100.1 will be for the StarLink web interface ! is it ok like that ?