Ysurac/openmptcprouter #2948: OpenMPTCProuter daemon.err glorytun[31165]: 47.243.244.8.65001: key exchange failed

fareign, 2023-09-01:
## Expected Behavior
need to reboot VPS to solve  this erro
but when reboot route also have this erro. 

## Specifications

  - OpenMPTCProuter version: 0.59.2alhpa 6.1
  - OpenMPTCProuter VPS version: 0.1029test upstream ubuntu
  - OpenMPTCProuter VPS provider:aliyun
  - OpenMPTCProuter platform:  x86_64 
  - Country: cn

fareign, 2023-09-01:
feedback:
now glorytun works very well .

Ysurac, 2023-09-01:
When there is this error, you only need to wait a little, "omr-service" script will restart Glorytun on VPS.
I maybe increased the timeout a bit too much..

fareign, 2023-09-01:
Sep  1 12:15:11 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sep  1 12:15:13 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: key exchange failed
Sep  1 12:15:13 OpenMPTCProuter daemon.info glorytun[21476]: STOPPED tun0
Sep  1 12:15:14 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: connected
Sep  1 12:15:14 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPCNT'
Sep  1 12:15:14 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPIDLE'
Sep  1 12:15:14 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPINTVL'
Sep  1 12:15:14 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep  1 12:15:24 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: key exchange failed
Sep  1 12:15:24 OpenMPTCProuter daemon.info glorytun[21476]: STOPPED tun0
Sep  1 12:15:25 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: connected
Sep  1 12:15:25 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPCNT'
Sep  1 12:15:25 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPIDLE'
Sep  1 12:15:25 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPINTVL'
Sep  1 12:15:25 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep  1 12:15:36 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: key exchange failed
Sep  1 12:15:36 OpenMPTCProuter daemon.info glorytun[21476]: STOPPED tun0
Sep  1 12:15:37 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: connected
Sep  1 12:15:37 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPCNT'
Sep  1 12:15:37 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPIDLE'
Sep  1 12:15:37 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPINTVL'
Sep  1 12:15:37 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep  1 12:15:47 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: key exchange failed
Sep  1 12:15:47 OpenMPTCProuter daemon.info glorytun[21476]: STOPPED tun0
Sep  1 12:15:48 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: connected
Sep  1 12:15:48 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPCNT'
Sep  1 12:15:48 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPIDLE'
Sep  1 12:15:48 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPINTVL'
Sep  1 12:15:48 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep  1 12:15:58 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: key exchange failed
Sep  1 12:15:58 OpenMPTCProuter daemon.info glorytun[21476]: STOPPED tun0
Sep  1 12:15:59 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: connected
Sep  1 12:15:59 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPCNT'
Sep  1 12:15:59 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPIDLE'
Sep  1 12:15:59 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPINTVL'
Sep  1 12:15:59 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep  1 12:16:09 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: key exchange failed
Sep  1 12:16:09 OpenMPTCProuter daemon.info glorytun[21476]: STOPPED tun0
Sep  1 12:17:19 OpenMPTCProuter daemon.err glorytun[21476]: 47.243.244.8.65001: connected
Sep  1 12:17:19 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPCNT'
Sep  1 12:17:19 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPIDLE'
Sep  1 12:17:19 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_KEEPINTVL'
Sep  1 12:17:19 OpenMPTCProuter daemon.err glorytun[21476]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep  1 12:17:19 OpenMPTCProuter daemon.info glorytun[21476]: STARTED tun0
Sep  1 12:17:19 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0

fareign, 2023-09-01:
YES,it can relink itself.a bout 2 minute.

Ysurac, 2023-09-01:
I changed the ping test to lower values. Will be available in next CDN cache refresh.

closed 2023-12-06 by github-actions[bot]