Ysurac/openmptcprouter #2951: StarLink + Freebox

bagou4502, 2023-09-04:
## Expected Behavior

Hello, I try to use OpenMpTCPRouter with a Freebox and a Starlink connexion.
Both are connected to my vps but we loose 88mbits (download) at the end.
For explain here are the speedtest with OpenMpTCPRouter:
[image]
Here is the speedtest with Starlink:
[image]
Here is the connexion with the Freebox:
[image]
And also the speedtest from the VPS:
[image]

but 56+125=181 and 181-93=88mbits.
This is a problem and I try to understand from where this can come.
[image]

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1028
  - OpenMPTCProuter VPS provider: HostMyServers
  - OpenMPTCProuter platform: RPI4

Ysurac, 2023-09-07:
Can you test using ```omr-test-speed eth2```, ```omr-test-speed wan2``` and ```omr-test-speed``` (for at least 1 minute, Ctrl+C to stop) ? You can also try ```omr-test-speed``` on the VPS.
Starlink doesn't provide a connection with a stable speed/latency and sometimes seems to give some strange results in speedtests.
You loose also some speed with the encryption (but not as much).

Xenon69, 2023-09-13:
Hello,
I experienced same problem, not yet resolved.
What device do you use for OMR (Raspberry pi, or other)?
If Raspberry pi, how did you connect starlink ethernet?

closed 2023-12-25 by github-actions[bot]