Ysurac/openmptcprouter #2953: OMR-Bypass domain openai.com doesn't fix ChatGPT App from being blocked

adamb, 2023-09-09:
## Expected Behavior

Setting the domain openai.com to be bypassed should allow the ChatGPT iOS app to connect directly and avoid any errors based on VPN IP detection.

## Current Behavior

I would expect that the app would work after telling OMR to directly connect to openai.com domains.

The website works fine with and without the bypass.  I'm using Linode for my vps.  My guess is there is another domain that the iOS app uses, but I can't figure out what it might be.  Or maybe I don't know how domain bypass works and I'm using it incorrectly.

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: Linode
  - OpenMPTCProuter platform: RPi4

Ysurac, 2023-09-12:
You can try, via SSH, ```tcpdump -vvv port 53``` to check what DNS requests are made. Check also that iOS use OMR IP are DNS.

adamb, 2023-09-12:
found it! 

`ios.chat.openai.com.cdn.cloudflare.net`

Thanks, you're the best!

closed 2023-09-12 by adamb