Ysurac/openmptcprouter #2954: new VPS upstream6.1 kernel erro log

fareign, 2023-09-10:
### proc/sys/net
Sep 10 08:06:02 iZj6cbfmkgkw74ryinmu3uZ omr-service[782]: sysctl: cannot stat /proc/sys/net/netfilter/nf_conntrack_max: No such file or directory
Sep 10 08:06:02 iZj6cbfmkgkw74ryinmu3uZ omr-service[782]: sysctl: cannot stat /proc/sys/net/netfilter/nf_conntrack_buckets: No such file or directory
Sep 10 08:06:02 iZj6cbfmkgkw74ryinmu3uZ omr-service[782]: sysctl: cannot stat /proc/sys/net/netfilter/nf_conntrack_tcp_timeout_established: No such file or directory

### net/netfilter

 Couldn't write '524288' to 'net/netfilter/nf_conntrack_max', ignoring: No such file or directory
Sep 10 08:05:58 iZj6cbfmkgkw74ryinmu3uZ systemd-sysctl[259]: Couldn't write '131072' to 'net/netfilter/nf_conntrack_buckets', ignoring: No such file or directory
Sep 10 08:05:58 iZj6cbfmkgkw74ryinmu3uZ systemd-sysctl[259]: Couldn't write '86400' to 'net/netfilter/nf_conntrack_tcp_timeout_established', ignoring: No such file or directory

### file is exist cat it
/proc/sys/net/netfilter/nf_conntrack_max
/proc/sys/net/nf_conntrack_max
root@iZj6cbfmkgkw74ryinmu3uZ:~# cat /proc/sys/net/nf_conntrack_max
6656
root@iZj6cbfmkgkw74ryinmu3uZ:~# cat /proc/sys/net/netfilter/nf_conntrack_max
6656
root@iZj6cbfmkgkw74ryinmu3uZ:~# 

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 0.1029test upstream 6.1
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

fareign, 2023-09-10:
### also some other erro log

Sep 09 01:24:50 iZj6cbfmkgkw74ryinmu3uZ omr-service[2083011]: Error: Nexthop device required for onlink.
Sep 09 01:24:50 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2083041]: /bin/sh: 1: dig: not found
Sep 09 01:24:54 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2083095]: /bin/sh: 1: dig: not found
Sep 09 01:24:55 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2083131]: /bin/sh: 1: dig: not found
Sep 09 01:24:56 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2083167]: /bin/sh: 1: dig: not found
Sep 09 01:24:58 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2083203]: /bin/sh: 1: dig: not found
Sep 09 01:24:59 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2083239]: /bin/sh: 1: dig: not found
Sep 09 01:25:01 iZj6cbfmkgkw74ryinmu3uZ CRON[2083312]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Sep 09 01:25:01 iZj6cbfmkgkw74ryinmu3uZ CRON[2083313]: (root) CMD (command -v debian-sa1 > /dev/null && debian-sa1 1 1)
Sep 09 01:25:01 iZj6cbfmkgkw74ryinmu3uZ CRON[2083312]: pam_unix(cron:session): session closed for user root
Sep 09 01:25:05 iZj6cbfmkgkw74ryinmu3uZ omr-service[2083921]: Error: Nexthop device required for onlink.
Sep 09 01:25:19 iZj6cbfmkgkw74ryinmu3uZ omr-service[2084594]: Error: Nexthop device required for onlink.
Sep 09 01:25:30 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:25:31 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:25:34 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:25:34 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:25:34 iZj6cbfmkgkw74ryinmu3uZ omr-service[2085267]: Error: Nexthop device required for onlink.
Sep 09 01:25:34 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:25:34 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:25:42 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:25:45 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:25:45 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:25:45 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:25:49 iZj6cbfmkgkw74ryinmu3uZ omr-service[2085940]: Error: Nexthop device required for onlink.
Sep 09 01:26:04 iZj6cbfmkgkw74ryinmu3uZ omr-service[2086613]: Error: Nexthop device required for onlink.
Sep 09 01:26:18 iZj6cbfmkgkw74ryinmu3uZ omr-service[2087286]: Error: Nexthop device required for onlink.
Sep 09 01:26:34 iZj6cbfmkgkw74ryinmu3uZ omr-service[2087972]: Error: Nexthop device required for onlink.
Sep 09 01:26:48 iZj6cbfmkgkw74ryinmu3uZ omr-service[2088645]: Error: Nexthop device required for onlink.
Sep 09 01:26:52 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:26:54 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:26:55 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:00 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:01 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:02 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:03 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:03 iZj6cbfmkgkw74ryinmu3uZ omr-service[2089331]: Error: Nexthop device required for onlink.
Sep 09 01:27:03 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:04 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:04 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:04 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:04 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:05 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:05 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:05 iZj6cbfmkgkw74ryinmu3uZ ss-server[265244]: remote recv: Connection reset by peer
Sep 09 01:27:18 iZj6cbfmkgkw74ryinmu3uZ omr-service[2090017]: Error: Nexthop device required for onlink.
Sep 09 01:27:33 iZj6cbfmkgkw74ryinmu3uZ omr-service[2090690]: Error: Nexthop device required for onlink.
Sep 09 01:27:47 iZj6cbfmkgkw74ryinmu3uZ omr-service[2091363]: Error: Nexthop device required for onlink.
Sep 09 01:28:02 iZj6cbfmkgkw74ryinmu3uZ omr-service[2092036]: Error: Nexthop device required for onlink.
Sep 09 01:28:17 iZj6cbfmkgkw74ryinmu3uZ omr-service[2092710]: Error: Nexthop device required for onlink.
Sep 09 01:28:31 iZj6cbfmkgkw74ryinmu3uZ omr-service[2093383]: Error: Nexthop device required for onlink.
Sep 09 01:28:46 iZj6cbfmkgkw74ryinmu3uZ omr-service[2094056]: Error: Nexthop device required for onlink.
Sep 09 01:29:01 iZj6cbfmkgkw74ryinmu3uZ omr-service[2094729]: Error: Nexthop device required for onlink.
Sep 09 01:29:15 iZj6cbfmkgkw74ryinmu3uZ omr-service[2095402]: Error: Nexthop device required for onlink.
Sep 09 01:29:30 iZj6cbfmkgkw74ryinmu3uZ omr-service[2096075]: Error: Nexthop device required for onlink.
[... cut ...]
Sep 09 01:33:44 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2107594]: /bin/sh: 1: dig: not found
Sep 09 01:33:48 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2107635]: /bin/sh: 1: dig: not found
Sep 09 01:33:49 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2107722]: /bin/sh: 1: dig: not found
Sep 09 01:33:50 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2107758]: /bin/sh: 1: dig: not found
Sep 09 01:33:52 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2107857]: /bin/sh: 1: dig: not found
Sep 09 01:33:53 iZj6cbfmkgkw74ryinmu3uZ omr-admin.py[2108431]: /bin/sh: 1: dig: not found

fareign, 2023-09-10:
sysctl -p /etc/sysctl.d/90-shadowsocks.conf 2>&1 >/dev/null  

can work
,but reboot VPS，The value doesnt change.

fareign, 2023-09-10:
### /bin/sh: 1: dig: not found

root@iZj6cbfmkgkw74ryinmu3uZ:~# apt-get install dnsutils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 bind9-libs : Depends: libuv1 (>= 1.40.0) but 1.34.2-1 is to be installed
E: Unable to correct problems, you have held broken packages.

fareign, 2023-09-10:
> ### /bin/sh: 1: dig: not found
> root@iZj6cbfmkgkw74ryinmu3uZ:~# apt-get install dnsutils Reading package lists... Done Building dependency tree... Done Reading state information... Done Some packages [...]

I have solved this ,install by deb package.

fareign, 2023-09-10:
### systemd start
Sep 10 15:16:36 iZj6cbfmkgkw74ryinmu3uZ systemd-sysctl[261]: Couldn't write '524288' to 'net/netfilter/nf_conntrack_max', ignoring: No such file or directo
### omr-server start
Sep 10 15:16:39 iZj6cbfmkgkw74ryinmu3uZ omr-service[742]: sysctl: cannot stat /proc/sys/net/netfilter/nf_conntrack_max: No such file or directory

I think this start earlyer than netfilter. so there are No such file?

Ysurac, 2023-09-10:
I fixed dig issue with omr-admin by removing dig dependency.
For sysctl issue, omr-service is now started later.
For the onlink error, was is the VPN you use ?

fareign, 2023-09-10:
glorytun tcp

fareign, 2023-09-10:
[image]
there also some SS link erro.

closed 2023-12-14 by github-actions[bot]