Ysurac/openmptcprouter #2955: OMR setup on x86 (EFI) firewall appliance

caselmann, 2023-09-11:
Hi,
I installed OMR on some 3rd party Intel based firewall appliance with 7 network ports. As the BIOS is EFI compatible, I selected ext4 image (EFI): openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined-efi.img.gz

After starting the machine, I was able to log in thru console but none of the network ports were configured. I tested them all for some DHCP signaling but to no avail. Are there any additional tasks to be done?

Installing it on my Linksys WRT3200 was no problem, the thruput was just miserable thus I wanted to test it with some different hardware.

Rgs
Chris

Ysurac, 2023-09-11:
Please use the template when you open an issue.

What you mean by not configured ? You have web access to the interface ? All interfaces are detected ? If not a driver may be missing.

caselmann, 2023-09-11:
I can't access the interface. When I check with ifconfig, it only shows lo (127.0.0.1). Would I need to setup a regular Ubuntu first which would automatically configure all interfaces? But the OMR would then delete everything when being flashed.

Ysurac, 2023-09-11:
No, you can try a snapshot image to check if the driver will be available in next release: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
Else you will have to find the used network chipset so that support can be added.

closed 2023-12-15 by github-actions[bot]