Ysurac/openmptcprouter #2957: Next Release when?

ccmks, 2023-09-13:
Hi,

When would be the next release for OMR? I see that the latest release was on August 30, 2022 last year and there has no update ever since

Thank you

Ysurac, 2023-09-13:
Work is made in develop branch and snapshots are made after each changes.
A beta for next release should be out this week.

fareign, 2023-09-13:
ss is not statible.status offen become yellow.

Ysurac, 2023-09-13:
@fareign can you open an issue ? Add system log and screenshot of status page when it's yellow.

fareign, 2023-09-13:
[image]

fareign, 2023-09-13:
I'M not sure is one of interface cause this or some other settings. V2ray sometime have socket not connected. but network still 
you know me test a long long time. there also a lot of  "Connection reset by peer" I know it is not real erro.but 5.4 kernel VPS may have less this erro. 

Sep 14 02:38:04 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: getpeername: Transport endpoint is not connected
Sep 14 02:38:02 iZj6cbfmkgkw74ryinmu3uZ omr-service[18265]: Error: Nexthop device required for onlink.
Sep 14 02:37:59 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:59 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:55 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:54 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:53 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:51 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:51 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:51 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:51 iZj6cbfmkgkw74ryinmu3uZ omr-service[17602]: Error: Nexthop device required for onlink.
Sep 14 02:37:51 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:50 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:50 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:46 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:43 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:43 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:40 iZj6cbfmkgkw74ryinmu3uZ omr-service[16939]: Error: Nexthop device required for onlink.
Sep 14 02:37:40 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:31 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:29 iZj6cbfmkgkw74ryinmu3uZ omr-service[16276]: Error: Nexthop device required for onlink.
Sep 14 02:37:26 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:25 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:24 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:23 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:23 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer
Sep 14 02:37:18 iZj6cbfmkgkw74ryinmu3uZ ss-server[874]: remote recv: Connection reset by peer

Ysurac, 2023-09-13:
Please open a new issue, it's not related to the original question.

closed 2023-12-18 by github-actions[bot]