Ysurac/openmptcprouter #2960: V2RAY UDP Issue

ccmks, 2023-09-15:
Is the UDP on V2RAY is working on last build?

OpenMPTCProuter version: v0.59.1-5.4
OpenMPTCProuter VPS version: 0.1028 5.4.207
OpenMPTCProuter VPS provider: melbicom SG
OpenMPTCProuter platform: x86 Hyper-V

I tried to make it to work for UDP aggregation under V2RAY and it seemed to be working OK, but after some random time later, it will stop working until I have to restart the OMR

Thank you

Ysurac, 2023-09-21:
What is the problem exactly ? V2Ray crash, no aggregation, something else ?

ccmks, 2023-09-21:
I am not sure if it is crash, but when try to use UDP traffic through OMR, it will not go through under V2RAY until I have to reboot the OMR

ccmks, 2023-09-21:
I guess you also mention on your next release that V2RAY doesn't work well on UDP

ioogithub, 2023-10-12:
> I tried to make it to work for UDP aggregation under V2RAY and it seemed to be working OK, but after some random time later, it will stop working until I have to restart the OMR.

This is very similar to my issue where after I restart omr or v2ray or tun0, aggregate works for a period of time but then stops. Restarting always restores it for a short time.

Where are you generating the udp traffic from? If I do this:

mtr -u www.domain.com from omr it will always use the tunnel rather than v2ray as the first hop:
`
1  10.255.251.1 (10.255.251.1)  42.387 ms  29.134 ms  45.914 ms`

Is this expected? I am not sure how to test if UDP traffic from a client uses v2ray rather than tun0.

closed 2024-01-16 by github-actions[bot]