Ysurac/openmptcprouter #2962: lot of "Error: Nexthop device required for onlink."

fareign, 2023-09-17:
## Expected Behavior

2023-09-17T09:58:51.917398+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[548060]: Error: Nexthop device required for onlink.
2023-09-17T09:59:13.470476+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[548785]: Error: Nexthop device required for onlink.
2023-09-17T09:59:34.588423+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[549701]: Error: Nexthop device required for onlink.
2023-09-17T09:59:55.580700+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[550558]: Error: Nexthop device required for onlink.
2023-09-17T10:00:17.884923+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[551336]: Error: Nexthop device required for onlink.
2023-09-17T10:00:38.905238+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[552145]: Error: Nexthop device required for onlink.
2023-09-17T10:00:45.801562+08:00 iZj6cbfmkgkw74ryinmu3uZ ss-server[544878]: getpeername: Transport endpoint is not connected
2023-09-17T10:00:59.884654+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[552837]: Error: Nexthop device required for onlink.
2023-09-17T10:01:20.905288+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[553529]: Error: Nexthop device required for onlink.
2023-09-17T10:01:42.242555+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[554221]: Error: Nexthop device required for onlink.
2023-09-17T10:02:03.328083+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[554926]: Error: Nexthop device required for onlink.
2023-09-17T10:02:03.898063+08:00 iZj6cbfmkgkw74ryinmu3uZ ss-server[544878]: remote recv: Connection reset by peer
2023-09-17T10:02:09.770427+08:00 iZj6cbfmkgkw74ryinmu3uZ ss-server[544878]: getpeername: Transport endpoint is not connected
2023-09-17T10:02:24.403155+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[555618]: Error: Nexthop device required for onlink.
2023-09-17T10:02:45.339592+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[556310]: Error: Nexthop device required for onlink.
2023-09-17T10:03:06.293434+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[557002]: Error: Nexthop device required for onlink.
2023-09-17T10:03:27.275213+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[557695]: Error: Nexthop device required for onlink.
2023-09-17T10:03:47.390602+08:00 iZj6cbfmkgkw74ryinmu3uZ OMR-Service: No answer from VPN client end, restart Glorytun-TCP
2023-09-17T10:03:47.397541+08:00 iZj6cbfmkgkw74ryinmu3uZ systemd[1]: Stopping glorytun-tcp@tun0.service - Glorytun TCP on tun0...
2023-09-17T10:03:47.397769+08:00 iZj6cbfmkgkw74ryinmu3uZ glorytun-tcp-run[545346]: INITIALIZED gt-tun0
2023-09-17T10:03:47.398506+08:00 iZj6cbfmkgkw74ryinmu3uZ glorytun-tcp-run[545346]: STARTED gt-tun0
2023-09-17T10:03:47.398556+08:00 iZj6cbfmkgkw74ryinmu3uZ glorytun-tcp-run[545346]: STOPPED gt-tun0
2023-09-17T10:03:47.400643+08:00 iZj6cbfmkgkw74ryinmu3uZ systemd-networkd[502]: gt-tun0: Link DOWN
2023-09-17T10:03:47.400697+08:00 iZj6cbfmkgkw74ryinmu3uZ systemd-networkd[502]: gt-tun0: Lost carrier
2023-09-17T10:03:47.411104+08:00 iZj6cbfmkgkw74ryinmu3uZ systemd[1]: glorytun-tcp@tun0.service: Deactivated successfully.
2023-09-17T10:03:47.411179+08:00 iZj6cbfmkgkw74ryinmu3uZ systemd[1]: Stopped glorytun-tcp@tun0.service - Glorytun TCP on tun0.
2023-09-17T10:03:47.418758+08:00 iZj6cbfmkgkw74ryinmu3uZ systemd[1]: Starting glorytun-tcp@tun0.service - Glorytun TCP on tun0...
2023-09-17T10:03:47.427173+08:00 iZj6cbfmkgkw74ryinmu3uZ (udev-worker)[557785]: Network interface NamePolicy= disabled on kernel command line.
2023-09-17T10:03:47.439398+08:00 iZj6cbfmkgkw74ryinmu3uZ systemd-networkd[502]: gt-tun0: Link UP
2023-09-17T10:03:47.439476+08:00 iZj6cbfmkgkw74ryinmu3uZ systemd-networkd[502]: gt-tun0: Gained carrier
2023-09-17T10:03:47.439956+08:00 iZj6cbfmkgkw74ryinmu3uZ systemd-networkd[502]: gt-tun0: Gained IPv6LL
2023-09-17T10:03:47.447342+08:00 iZj6cbfmkgkw74ryinmu3uZ systemd[1]: Started glorytun-tcp@tun0.service - Glorytun TCP on tun0.
2023-09-17T10:03:48.426160+08:00 iZj6cbfmkgkw74ryinmu3uZ glorytun-tcp-run[557789]: ::ffff:120.228.113.34.8098: connected
2023-09-17T10:03:48.426243+08:00 iZj6cbfmkgkw74ryinmu3uZ glorytun-tcp-run[557789]: couldn't set socket option `TCP_KEEPCNT'
2023-09-17T10:03:48.426274+08:00 iZj6cbfmkgkw74ryinmu3uZ glorytun-tcp-run[557789]: couldn't set socket option `TCP_KEEPIDLE'
2023-09-17T10:03:48.426295+08:00 iZj6cbfmkgkw74ryinmu3uZ glorytun-tcp-run[557789]: couldn't set socket option `TCP_KEEPINTVL'
2023-09-17T10:03:48.426314+08:00 iZj6cbfmkgkw74ryinmu3uZ glorytun-tcp-run[557789]: couldn't set socket option `TCP_USER_TIMEOUT'
2023-09-17T10:04:01.390241+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[558604]: Error: Nexthop device required for onlink.
2023-09-17T10:04:15.335031+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[559295]: Error: Nexthop device required for onlink.
2023-09-17T10:04:29.300297+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[559981]: Error: Nexthop device required for onlink.
2023-09-17T10:04:43.258584+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[560655]: Error: Nexthop device required for onlink.
2023-09-17T10:04:57.290684+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[561328]: Error: Nexthop device required for onlink.
2023-09-17T10:05:01.889007+08:00 iZj6cbfmkgkw74ryinmu3uZ CRON[561345]: (root) CMD (command -v debian-sa1 > /dev/null && debian-sa1 1 1)
2023-09-17T10:05:11.248531+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[562004]: Error: Nexthop device required for onlink.
2023-09-17T10:05:25.217461+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[562677]: Error: Nexthop device required for onlink.
2023-09-17T10:05:39.204682+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[563350]: Error: Nexthop device required for onlink.
2023-09-17T10:05:53.170345+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[564023]: Error: Nexthop device required for onlink.
2023-09-17T10:06:07.169998+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[564696]: Error: Nexthop device required for onlink.
2023-09-17T10:06:21.160314+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[565369]: Error: Nexthop device required for onlink.
2023-09-17T10:06:35.182336+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[566042]: Error: Nexthop device required for onlink.
2023-09-17T10:06:49.131983+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[566728]: Error: Nexthop device required for onlink.
2023-09-17T10:07:03.262149+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[567414]: Error: Nexthop device required for onlink.
2023-09-17T10:07:17.367001+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[568087]: Error: Nexthop device required for onlink.
2023-09-17T10:07:31.389997+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[568773]: Error: Nexthop device required for onlink.
2023-09-17T10:07:45.465234+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[569459]: Error: Nexthop device required for onlink.
2023-09-17T10:07:59.500403+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[570132]: Error: Nexthop device required for onlink.
2023-09-17T10:08:13.511106+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[570805]: Error: Nexthop device required for onlink.
2023-09-17T10:08:27.595452+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[571479]: Error: Nexthop device required for onlink.
2023-09-17T10:08:41.767801+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[572152]: Error: Nexthop device required for onlink.
2023-09-17T10:08:55.958433+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[572825]: Error: Nexthop device required for onlink.
2023-09-17T10:09:09.921167+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[573498]: Error: Nexthop device required for onlink.
2023-09-17T10:09:23.886654+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[574171]: Error: Nexthop device required for onlink.
2023-09-17T10:09:38.003509+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[574844]: Error: Nexthop device required for onlink.
2023-09-17T10:09:52.057984+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[575517]: Error: Nexthop device required for onlink.
2023-09-17T10:10:05.996155+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[576190]: Error: Nexthop device required for onlink.
2023-09-17T10:10:20.003128+08:00 iZj6cbfmkgkw74ryinmu3uZ omr-service[576863]: Error: Nexthop device required for onlink.
2023-09-17T10:10:40.117828+08:00 iZj6cbfmkgkw74ryinmu3uZ OMR-Service: No answer from VPN client end, restart Glorytun-TCP
2023-09-17T10:10:40.124732+08:00 iZj6cbfmkgkw74ryinmu3uZ systemd[1]: Stopping glorytun-tcp@tun0.service - Glorytun TCP on tun0...
[... 109 lines cut ...]
2023-09-17T10:32:50.672763+08:00 iZj6cbfmkgkw74ryinmu3uZ v2ray[2934462]: 2023/09/17 10:32:50 120.228.113.34:7708 accepted tcp:163.123.192.55:443 [direct] email: openmptcprouter
[... cut ...]
2023-09-17T10:32:56.353417+08:00 iZj6cbfmkgkw74ryinmu3uZ v2ray[2934462]: 2023/09/17 10:32:56 120.228.113.34:7729 accepted tcp:129.28.240.139:443 [direct] email: openmptcprouter
2023-09-17T10:32:56.509590+08:00 iZj6cbfmkgkw74ryinmu3uZ v2ray[2934462]: 2023/09/17 10:32:56 120.228.113.34:7730 accepted tcp:129.28.240.139:443 [direct] email: openmptcprouter

## Steps to Reproduce the Problem

every prox and every VPN all has this erro

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.61beta
  - OpenMPTCProuter VPS version: 0.1029test upstream
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country:cn

fareign, 2023-09-17:
### vpS ip r
default via 172.18.223.253 dev eth0 
10.255.246.0/24 dev client-wg0 proto kernel scope link src 10.255.246.1 
10.255.247.0/24 dev wg0 proto kernel scope link src 10.255.247.1 
10.255.248.2 dev omr-bonding scope link src 10.255.248.1 
10.255.250.0/24 via 10.255.250.2 dev tun1 
10.255.250.2 dev tun1 proto kernel scope link src 10.255.250.1 
10.255.251.2 dev dsvpn0 proto kernel scope link src 10.255.251.1 
10.255.252.0/24 via 10.255.252.2 dev tun0 
10.255.252.2 dev tun0 proto kernel scope link src 10.255.252.1 
10.255.253.0/24 dev mlvpn0 proto kernel scope link src 10.255.253.1 
10.255.254.0/30 dev gt-udp-tun0 proto kernel scope link src 10.255.254.1 
10.255.255.0/30 dev gt-tun0 proto kernel scope link src 10.255.255.1 
172.18.208.0/20 dev eth0 proto kernel scope link src 172.18.219.191 

### router ip r
default via 10.255.254.1 dev tun0 
default via 192.168.50.1 dev eth1 metric 1 
default via 192.168.50.1 dev eth1 metric 7 
default via 10.255.254.1 dev tun0 metric 1500 
10.255.254.1 dev tun0 proto kernel scope link src 10.255.254.2 
10.255.254.2 dev tun0 scope link metric 1500 
112.74.39.209 via 192.168.50.1 dev eth1 metric 1 
127.0.0.0/8 dev lo proto static scope link metric 5 
192.168.50.0/24 dev eth1 scope link metric 7 

ss+glorytun upd

fareign, 2023-09-17:
### VPS ip r
default via 172.18.223.253 dev eth0 
10.255.246.0/24 dev client-wg0 proto kernel scope link src 10.255.246.1 
10.255.247.0/24 dev wg0 proto kernel scope link src 10.255.247.1 
10.255.248.2 dev omr-bonding scope link src 10.255.248.1 
10.255.250.0/24 via 10.255.250.2 dev tun1 
10.255.250.2 dev tun1 proto kernel scope link src 10.255.250.1 
10.255.251.2 dev dsvpn0 proto kernel scope link src 10.255.251.1 
10.255.252.0/24 via 10.255.252.2 dev tun0 
10.255.252.2 dev tun0 proto kernel scope link src 10.255.252.1 
10.255.253.0/24 dev mlvpn0 proto kernel scope link src 10.255.253.1 
10.255.254.0/30 dev gt-udp-tun0 proto kernel scope link src 10.255.254.1 
10.255.255.0/30 dev gt-tun0 proto kernel scope link src 10.255.255.1 
172.18.208.0/20 dev eth0 proto kernel scope link src 172.18.219.191 

### router
# ip r
default via 172.18.223.253 dev eth0 
10.255.246.0/24 dev client-wg0 proto kernel scope link src 10.255.246.1 
10.255.247.0/24 dev wg0 proto kernel scope link src 10.255.247.1 
10.255.248.2 dev omr-bonding scope link src 10.255.248.1 
10.255.250.0/24 via 10.255.250.2 dev tun1 
10.255.250.2 dev tun1 proto kernel scope link src 10.255.250.1 
10.255.251.2 dev dsvpn0 proto kernel scope link src 10.255.251.1 
10.255.252.0/24 via 10.255.252.2 dev tun0 
10.255.252.2 dev tun0 proto kernel scope link src 10.255.252.1 
10.255.253.0/24 dev mlvpn0 proto kernel scope link src 10.255.253.1 
10.255.254.0/30 dev gt-udp-tun0 proto kernel scope link src 10.255.254.1 
10.255.255.0/30 dev gt-tun0 proto kernel scope link src 10.255.255.1 
172.18.208.0/20 dev eth0 proto kernel scope link src 172.18.219.191 

ss+openVPN

fareign, 2023-09-17:
is VPS IP dosenot match link IP?

Ysurac, 2023-09-18:
I removed the not working onlink for now, it was added on route to lan network without the VPN network interface.

closed 2023-12-22 by github-actions[bot]