Ysurac/openmptcprouter #2963: Multipath TCP path-manager does not aggregate connections.

vempire-ghost, 2023-09-17:
## Expected Behavior

## Current Behavior
Multipath TCP path-manager Binders dont work.

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)
Using the Binder path manager does not aggregate connections; when selected, it only utilizes the Master connection, even when using the Redundant scheduler.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta1-5.4 r0+16862-170d9
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1029-test
  - OpenMPTCProuter VPS provider: Google Cloud
  - OpenMPTCProuter platform: x86_64
  - Country: Brazil

Ysurac, 2023-09-18:
Binder path manager require to have "Loose Source and Record Route" (LSRR) supported on the local network. When it's not the case all packets are routed over the same initially chosen path.

vempire-ghost, 2023-09-18:
Thank you, its something physical on my routers or a configuration? How can I see and enable this in my network?

Ysurac, 2023-09-18:
It's routers configuration on all network path, it's disabled on most networks for security reason (an attacker can spoof an IP receive spoofed packets for this IP) and not useful for general operation. It's only for some specific cases like when we want to implement mobility in IP networks.

closed 2023-09-21 by Ysurac