Ysurac/openmptcprouter #2965: rock 3A rk3568 R5S support

fareign, 2023-09-20:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
I test R5s on rock 3a ,but it can not work.
https://github.com/mj22226/openwrt 
there is the openwrt which support rock 3A.if I use this openwrt tree build the image, it can work?

**Describe why you are asking for this support?**

**Additional context**

Ysurac, 2023-09-21:
I can find only Rock PI 4A support in the github, where is 3a support ?
4A is also available in OpenMPTCProuter, but need to be compiled manually: I don't support platform I don't own.

kpma1985, 2023-10-27:
@fareign for its working perfect
R5S, R5C and R4SE

closed 2024-01-31 by github-actions[bot]