Ysurac/openmptcprouter #2967: UDP traffice may cause high time delay

fareign, 2023-09-21:
## RTMP push and pull
 
[image]

## SRT Push
 
[image]

## Possible Solution
 
every mode is  about the same result.
I think it may something relate to 4G or 5G . or some other reason

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.6beta 6.1 kernel
  - OpenMPTCProuter VPS version: 0.1029test upstream
  - OpenMPTCProuter VPS provider:aliyun
  - OpenMPTCProuter platform:  x86_64 
  - Country: CN

Ysurac, 2023-09-21:
What is the VPN used ? The default VPN, Glorytun TCP doesn't have a very stable latency when it's used.

fareign, 2023-09-21:
every VPN try. V2ray ,try. almost the same result.

Ysurac, 2023-09-21:
Then this can be due to bandwidth saturation, you can use SQM that try to solve this.

fareign, 2023-09-22:
@Ysurac is V2ray default UDP use openvpn? I found v2ray config file key is openvpn

"certificateFile":"/etc/openvpn/ca/pki/issued/server.crt"
keyFile":"/etc/openvpn/ca/pki/private/server.key"

fareign, 2023-09-22:
I think the main reason is telecom operators speed limit for UDP traffic,UDP2Raw-tunnel and updspeed.may  do something help?

Ysurac, 2023-09-22:
Glorytun-TCP, and V2Ray if enabled for UDP in advanced settings tab, both fo UDP over TCP. So even if operators limit UDP traffic this will work.
V2Ray and OpenVPN share same keys, because it's easier to have a single key pair.

closed 2023-12-26 by github-actions[bot]