Ysurac/openmptcprouter #2968: keepalived not installable in 0.60 beta1 (missing dependency)

Steltek, 2023-09-21:
## Expected Behavior

Installing keepalived should install the package and make keepalived available to use.

## Current Behavior

Installation fails due to missing dependency:

> Collected errors:
>  * pkg_hash_check_unresolved: cannot find dependency libip4tc2 for keepalived
>  * pkg_hash_check_unresolved: cannot find dependency libip6tc2 for keepalived
>  * pkg_hash_fetch_best_installation_candidate: Packages for keepalived found, but incompatible with the architectures configured
>  * opkg_install_cmd: Cannot install package keepalived.

 
## Steps to Reproduce the Problem

  1. Install OpenMPTCProuter v0.60beta1-6.1 r0+23789-ce6ad123e7 (x86_64 variant)
  2. Update software.
  3. Try to install keepalived.

## Context (Environment)

I use keepalived to keep my routers redundant (if OpenMPTCPRouter fails, the VRRP IP can fail over to a regular OpenWRT router).

## Specifications

  - OpenMPTCProuter version: v0.60beta1-6.1 r0+23789-ce6ad123e7
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter VPS provider: N/A
  - OpenMPTCProuter platform: x86_64
  - Country: Luxembourg

Ysurac, 2023-09-22:
This will be fixed in snapshot and next beta.

Steltek, 2023-09-24:
Can confirm this works in the latest snapshot (keepalived is installed and running).

Steltek, 2023-10-31:
Unfortunately, the issue of keepalived not being installable appears to be back in [v0.60beta1-6.1 r0+24041-74e7f8ebbd] with it now complaining about kernel incompatibility:

```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-lib-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-nf-log
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-nf-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-ipt-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-nf-ipvs
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-ipvs found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-macvlan
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-nfnetlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf) for kmod-ipt-ipset
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for keepalived:
 * 	kernel (= 6.1.46-1-3e10a79115765fc4caa4898d560a40cf)
 * opkg_install_cmd: Cannot install package keepalived.
```

closed 2023-10-31 by Ysurac