Ysurac/openmptcprouter #2969: VPS side on a local PC

open4user, 2023-09-22:
Is it possible to turn your local PC into a server for OMR?

I have just installed debian-11.7.0-amd64-DVD-1.iso, then "apt-get update && apt-get upgrade".
But "sh debian-x86_64.sh" drops some dependencies error. Am I using an incorrect Debian image/version? Or what I am doing is just not possible?

[image]

Ysurac, 2023-09-22:
It's possible but you will need to install missing packages manually.
Client and VPS shouldn't be on same network else you will get same output speed.

open4user, 2023-09-29:
@Ysurac For some reason I can't ping my static IP (the one I requested on provider's side for my PC server) from the OMR. At the same time OMR has access to web and can access other sites. Maybe you know if I need additional configurations on OMR to access my PC server's static IP?

"ip r" on the OMR looks ok
"systemctl status shadowsocks-libev-manager@manager.service" on the server (my PC) looks running
"systemctl status omr-admin.service" on the server (my PC) looks running
"ping [my static IP]" from my server itself looks ok

Ysurac, 2023-09-29:
I would need ```ip a```, ```ip r``` and the IP you want to ping from OMR router part.

open4user, 2023-10-01:
@Ysurac I guess I found a critical missing part. 

I need to set up my static IP (the one from my ISP) either on my ASUS router, something like this:
"Input the MAC address of my Debian PC and assign the static IP address from my ISP"

or directly from the Debian PC (the server for OMR), something like this:
"ifconfig etho1 [static IP from my ISP]"

Could you please confirm if this is the right direction? And if yes - do I need just the ASUS router configuration, or both ASUS and Debian PC? I will try all option later anyway (don't have access to the Debian PC now).

Ysurac, 2023-10-01:
I don't know where is your asus router on the path. In ideal case, the OMR server should have it's own public IP but can work with a private IP that have internet access.

closed 2024-01-04 by github-actions[bot]