Ysurac/openmptcprouter #2970: OpenMPTCPRouter and VoWiFi

frandj99, 2023-09-22:
Hello,
I'm trying to make work VoWiFi over my LAN.
My italian ISP (TIM) supports VoWiFi abd should work fine over any network.

## Expected Behavior
- Adding an OMR Bypass rule for port 500 and 4500 and to the domain ( `epdg.epc.mnc001.mcc222.pub.3gppnetwork.org` )that my ISP should use for VoWiFI, everything should just work fine.
I've read the similar issue https://github.com/Ysurac/openmptcprouter/issues/2109 but trying the solution they gave, it didn't work for me.
Also I've checked this option:
System > Settings Wizard > Show Advanced Settings > Default Proxy --> **Shadowsocks**

## Current Behavior

VoWiFi is not working on OpenMPTCPRouter LAN, but is working fine if I connect my phone to one of the WAN, without OpenMPTCPRouter in the middle.

## Specifications

  - OpenMPTCProuter version: [0.59.1-5.4 r0+16594-ce92de8c8c](https://github.com/ysurac/openmptcprouter) 
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1028
  - OpenMPTCProuter VPS provider: https://www.serverplan.com
  - OpenMPTCProuter platform: x86_64

Let me know if any other information is needed. I wish I could find some log or trace this port, but I can't find a way or where to look.

aldochip, 2023-09-28:
Hi,
can you confirm that your public IP is in an Italian assigned  addresses pool ?

frandj99, 2023-10-03:
Hello,
yeah, I can confirm the VPS has an italian IP address and also I never had issue with other italian services like TV streaming.

closed 2024-01-07 by github-actions[bot]