Ysurac/openmptcprouter #2972: OMR doesn't return routes to same state after restarting an interface, duplicate route and missing default metric 1

ioogithub, 2023-09-24:
## Expected Behavior
routes as show by` ip r` will be restored to the same state after an interface is restarted.

## Current Behavior
`ip r` shows different routes and sometimes duplicate routes after omr-tracker restarts an interface

## Possible Solution

## Steps to Reproduce the Problem

  1. Start OMR is a known good state, reboot router and wait for , green on status page, do an `ip r`
  2. Wait for omr-tracker to restart an interface due to ping timeout.
  3. After status page is green again, do another `ip r` and compare result to the known good state. The routes are different now. 

## Context (Environment)
- he 'default metric 1' route is never restored.
- Sometime duplicate default addresses are also added.

## Example 1: good ip r after reboot of OMR
```
default via 10.255.255.1 dev tun0 
default metric 1 
        nexthop via x.x.x.x dev eth1 weight 2 
        nexthop via y.y.y.y dev eth2 weight 1 
default via x.x.x.x dev eth1 metric 3 
default via y.y.y.y dev eth2 metric 4 
default via 10.255.255.1 dev tun0 metric 1200 
10.b.b.b/24 via y.y.y.y dev eth2 
10.b.b.b/24 dev eth2 scope link metric 4 
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
10.255.255.2 dev tun0 scope link metric 1200 
100.c.c.c/10 dev eth1 scope link metric 3 
127.0.0.0/8 dev lo proto static scope link metric 6 
a.a.a.a metric 1 
        nexthop via x.x.x.x dev eth1 weight 10 
        nexthop via y.y.y.y dev eth2 weight 1 
```

## Example 2: After om-tracker has restarted  eth1 end eth2 a few times:

```
default via 10.255.255.1 dev tun0 
default via x.x.x.x dev eth1 metric 1 
default via x.x.x.x dev eth1 metric 3 
default via y.y.y.y dev eth2 metric 4 
10.b.b.b/24 via y.y.y.y dev eth2 
10.b.b.b/24 dev eth2 scope link metric 4 
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
10.255.255.2 dev tun0 scope link metric 1200 
100.c.c.c/10 dev eth1 scope link metric 3 
127.0.0.0/8 dev lo proto static scope link metric 6 
a.a.a.a metric 1 
        nexthop via x.x.x.x dev eth1 weight 10 
        nexthop via y.y.y.y dev eth2 weight 1 
```

Two differences:
1. the `default metric 1` section was not restored
2. There are duplicate routes, these two lines are the same route, they have different metrics but the address is the same:
```default via x.x.x.x dev eth1 metric 1 
default via x.x.x.x dev eth1 metric 3 ```

Example 3: I tried to recover by rebooting OMR, this is the result:

```
default via 10.255.255.1 dev tun0 
default via x.x.x.x dev eth1 metric 1 
default via y.y.y.y dev eth2 metric 4 
10.b.b.b/24 via y.y.y.y dev eth2 
10.b.b.b/24 dev eth2 scope link metric 4 
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
10.255.255.2 dev tun0 scope link metric 1200 
100.c.c.c/10 dev eth1 scope link metric 3 
127.0.0.0/8 dev lo proto static scope link metric 6 
a.a.a.a metric 1 
        nexthop via x.x.x.x dev eth1 weight 10 
        nexthop via y.y.y.y dev eth2 weight 1 
```
1. Duplicate is done but default metric not restored.

Example 4: Second reboot,
Same as example 1, router back in normal state again.

I also tried validating the wizard but this did not change the routes.

One of the script is not properly deleting the routes and it does not restore the default metric. Could this be causing some of the issues I am seeing?

## Specifications
   - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
   - OpenMPTCProuter VPS version: VPS 0.1028
   - OpenMPTCProuter VPS provider:
   - OpenMPTCProuter platform: x86_64)

ioogithub, 2023-09-24:
I just caught the bug while I was online. Here is the route table in a good state:

```
default via 10.255.255.1 dev tun0 
default metric 1 
        nexthop via x.x.x.x dev eth1 weight 2 
        nexthop via y.y.y.y dev eth2 weight 1 
default via x.x.x.x dev eth1 metric 3 
default via y.y.y.y dev eth2 metric 4 
default via 10.255.255.1 dev tun0 metric 1200 
10.b.b.b/24 via y.y.y.y dev eth2 
10.b.b.b/24 dev eth2 scope link metric 4 
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
10.255.255.2 dev tun0 scope link metric 1200 
100.c.c.c/10 dev eth1 scope link metric 3 
127.0.0.0/8 dev lo proto static scope link metric 6 
a.a.a.a metric 1 
        nexthop via x.x.x.x dev eth1 weight 10 
        nexthop via y.y.y.y dev eth2 weight 1
```

Here is the log:

```
Sun Sep 24 13:36:53 2023 user.notice post-tracking-post-tracking: wan2 (eth2) switched off because wan2 may have ip issues, interface have no IPv4, interface have no IPv4 gateway
Sun Sep 24 13:36:53 2023 user.notice post-tracking-post-tracking: Delete default route to a.a.a.a via  dev eth2
Sun Sep 24 13:36:54 2023 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Sep 24 13:36:54 2023 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)
Sun Sep 24 13:36:55 2023 daemon.notice netifd: wan2 (21972): udhcpc: sending discover
Sun Sep 24 13:36:55 2023 daemon.notice netifd: Network device 'eth2' link is up
Sun Sep 24 13:36:55 2023 daemon.notice netifd: Interface 'wan2' has link connectivity
Sun Sep 24 13:36:55 2023 kern.info kernel: [49339.456623] igc 0000:04:00.0 eth2: igc: eth2 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Sun Sep 24 13:36:56 2023 user.notice post-tracking-post-tracking: Restart wan2
Sun Sep 24 13:36:56 2023 daemon.notice netifd: wan2 (21972): udhcpc: received SIGTERM
Sun Sep 24 13:36:56 2023 daemon.notice netifd: wan2 (21972): Command failed: Permission denied
Sun Sep 24 13:36:56 2023 daemon.notice netifd: Interface 'wan2' is now down
Sun Sep 24 13:36:56 2023 kern.info kernel: [49339.722238] igc 0000:04:00.0: changing MTU from 1460 to 1500
Sun Sep 24 13:36:56 2023 daemon.notice netifd: Interface 'wan2' is disabled
Sun Sep 24 13:36:56 2023 daemon.notice netifd: Network device 'eth2' link is down
Sun Sep 24 13:36:56 2023 daemon.notice netifd: Interface 'wan2' has link connectivity loss
Sun Sep 24 13:37:01 2023 kern.info kernel: [49344.632167] igc 0000:04:00.0: changing MTU from 1500 to 1460
Sun Sep 24 13:37:01 2023 daemon.notice netifd: Interface 'wan2' is enabled
Sun Sep 24 13:37:01 2023 daemon.notice netifd: Interface 'wan2' is setting up now
Sun Sep 24 13:37:01 2023 daemon.notice netifd: wan2 (23929): udhcpc: started, v1.33.2
Sun Sep 24 13:37:01 2023 daemon.notice netifd: wan2 (23929): udhcpc: sending discover
Sun Sep 24 13:37:04 2023 daemon.notice netifd: wan2 (23929): udhcpc: sending discover
Sun Sep 24 13:37:04 2023 daemon.notice netifd: Network device 'eth2' link is up
Sun Sep 24 13:37:04 2023 daemon.notice netifd: Interface 'wan2' has link connectivity
Sun Sep 24 13:37:04 2023 kern.info kernel: [49348.256298] igc 0000:04:00.0 eth2: igc: eth2 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Sun Sep 24 13:37:07 2023 daemon.notice netifd: wan2 (23929): udhcpc: sending discover
Sun Sep 24 13:37:10 2023 daemon.notice netifd: wan2 (23929): udhcpc: sending select for y.y.y.a
Sun Sep 24 13:37:10 2023 daemon.notice netifd: Interface 'wan2' is now up
Sun Sep 24 13:37:10 2023 kern.info kernel: [49354.145648] igc 0000:04:00.0 eth2: igc: eth2 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Sun Sep 24 13:37:10 2023 kern.info kernel: [49354.209225] igc 0000:04:00.0 eth2: igc: eth2 NIC Link is Down
Sun Sep 24 13:37:10 2023 user.notice firewall: Reloading firewall due to ifup of wan2 (eth2)
Sun Sep 24 13:37:10 2023 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sun Sep 24 13:37:10 2023 user.notice mptcp: Reloading mptcp config due to ifup of wan2 (eth2)
Sun Sep 24 13:37:10 2023 daemon.notice netifd: Network device 'eth2' link is down
Sun Sep 24 13:37:10 2023 daemon.notice netifd: Interface 'wan2' has link connectivity loss
Sun Sep 24 13:37:14 2023 daemon.notice netifd: Network device 'eth2' link is up
Sun Sep 24 13:37:14 2023 daemon.notice netifd: Interface 'wan2' has link connectivity
Sun Sep 24 13:37:14 2023 kern.info kernel: [49357.775977] igc 0000:04:00.0 eth2: igc: eth2 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Sun Sep 24 13:37:29 2023 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Sep 24 13:37:29 2023 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)
Sun Sep 24 13:38:05 2023 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Sep 24 13:38:06 2023 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)
Sun Sep 24 13:38:44 2023 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Sep 24 13:38:45 2023 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)
Sun Sep 24 13:39:22 2023 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Sep 24 13:39:23 2023 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)
[... 5 lines cut ...]
Sun Sep 24 13:41:13 2023 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)
Sun Sep 24 13:41:47 2023 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Sep 24 13:41:48 2023 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)
[... cut ...]
but it doesn't actually delete the route:
`        nexthop via y.y.y.y dev eth2 weight 1`

The` ip r `shows the same routes as before the issue, it never changes. Also, there is no error or log message indicating that it could not delete the route.

- What can be the root cause of this problem:
`Sun Sep 24 13:36:53 2023 user.notice post-tracking-post-tracking: wan2 (eth2) switched off because wan2 may have ip issues, interface have no IPv4, interface have no IPv4 gateway` it happens several times a day.

Ysurac, 2023-09-24:
These issues should already be fixed in latest beta release.

ioogithub, 2023-09-24:
> These issues should already be fixed in latest beta release.

Can you recall where the patches were made?

I don't think I can run the beta now, I need a stable release for this application. The main upload aggregate problem was still present in the 6.1 kernel version I tried but there were much bigger problems, I could not recover from this single aggregate state without restarting OMR. In the stable version I now have a test and script which monitors and fixes the problem.

What is the main script involved for this bug: (post-tracking-post-tracking netifd or omr-tracker). Can I find the commits in the github and see if I can find a fix.

I guess these are the two main problems:
- Tries to run a command and fails:
`Sun Sep 24 13:36:46 2023 daemon.notice netifd: wan2 (6262): Command failed: Permission denied`

- Isn't parsing the route string properly:
`Sun Sep 24 13:36:53 2023 user.notice post-tracking-post-tracking: Delete default route to a.a.a.a via  dev eth2`

ioogithub, 2023-09-24:
I looked at all commits here back to Dec 2020: https://github.com/Ysurac/openmptcprouter/commits/master but I did not see any changes to those scripts. Are these type of changes made somewhere else?

ioogithub, 2023-09-24:
The last stable version 0.59.1 was released aug 29, 2002. I looked at all all changes to omr-tracker here: https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/omr-tracker/files/bin/omr-tracker from today back to Aug 29, 2022, there were only two fixes but they were for ipv4 issues, not related to this problem.

Is this the right place to look for the fixes?

ioogithub, 2023-09-24:
I foudn a patch that looks promising, it changes several route issues, it was made after the 0.59.1 release and it is the post-tracker: https://github.com/Ysurac/openmptcprouter-feeds/commit/88dcdd748b1ab6673d7b211ae4496af5e23057b7.

Is it possible to take this file and run omr version 5.1 with it as a test or will this definitely not work?

ioogithub, 2023-09-24:
I reviewed all of the changed to https://github.com/Ysurac/openmptcprouter-feeds/blob/93fb75da176870fa2be7fd710ccf905e0598c816/mptcp/files/usr/share/omr/post-tracking.d/post-tracking#L611

Here is the log line, it actually runs twice:
`Sun Sep 24 13:36:53 2023 user.notice post-tracking-post-tracking: Delete default route to a.a.a.a via  dev eth2`

and I believe here is the code on line 616

```
del_server_route() {
	local server=$1
	remove_route() {
		local serverip="$1"
		[ -n "$serverip" ] && serverip="$(resolveip -4 -t 5 $serverip | head -n 1 | tr -d '\n')"
		[ -n "$serverip" ] && _log "Delete default route to $serverip via $OMR_TRACKER_DEVICE_GATEWAY dev $OMR_TRACKER_DEVICE"
		local metric
		if [ -z "$OMR_TRACKER_INTERFACE" ]; then
			metric=0
		else
			metric=$(uci -q get network.$OMR_TRACKER_INTERFACE.metric)
		fi
		[ -n "$metric" ] && [ -n "$OMR_TRACKER_DEVICE" ] && [ -n "$serverip" ] && [ -n "$(ip route show $serverip dev $OMR_TRACKER_DEVICE metric $metric)" ] && ip route del $serverip dev $OMR_TRACKER_DEVICE metric $metric >/dev/null 2>&1
		[ -n "$OMR_TRACKER_DEVICE" ] && [ -n "$serverip" ] && [ -n "$(ip route show $serverip dev $OMR_TRACKER_DEVICE)" ] && ip route del $serverip dev $OMR_TRACKER_DEVICE >/dev/null 2>&1
		[ -n "$OMR_TRACKER_DEVICE" ] && [ -n "$serverip" ] && [ -n "$(ip route show $serverip | grep $OMR_TRACKER_DEVICE)" ] && ip route del $serverip dev $OMR_TRACKER_DEVICE >/dev/null 2>&1
	}
```
but it didn't work, it didn't delete the route. I did an ip r before and after and the routing table was unchanged. If I look at the other changes to this file made to the new vesions:
https://github.com/Ysurac/openmptcprouter-feeds/commits/93fb75da176870fa2be7fd710ccf905e0598c816/mptcp/files/usr/share/omr/post-tracking.d/post-tracking
there is nothing else that changes this part of the code so it may still apply to the new beta version.

ioogithub, 2023-09-25:
What does this mean:

Common:
    V2Ray doesn't work well on UDP

closed 2023-12-29 by github-actions[bot]