Ysurac/openmptcprouter #2973: omr-bypass not working after adding many entries. All domains now hop to 10.255.255.1

ioogithub, 2023-09-25:
## Expected Behavior
omr bypass will direct a domain on the list to the interface I specify

## Current Behavior
All bypassed, domains asn and protocol go out 10.255.255.1

## Possible Solution

## Steps to Reproduce the Problem

OMR-bypass was working when I had a few entries as a test (10). Yesterday I added a lot of entries (100) and now all entries are not bypassed.

1. Add several domains to bypass (10)
2. domain.com eth2  
3. on the omr router: traceroute www.domain.com
4. first hop is eth2 interface- working
5. Next day, add a lot of domains (100)
6. repeat step 3
7. now first hop is always 10.255.255.1
8. reboot router
8. repeat try again, alwayas 10.255.255.1

## Context (Environment)
This needs to work as many websites do not like VPS IP addresses.

## More info.

Verify domain is in this list:
```
uci show omr-bypass
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com' 'canalplus.com' 'canalplus-cdn.net' 'canalplus.pro' 'canal-plus.net'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.amazonvideo=proto
omr-bypass.amazonvideo.url='cloudfront.net' 'llnw.net'
omr-bypass.lo=interface
omr-bypass.lo.id='6'
omr-bypass.eth0=interface
omr-bypass.eth0.id='7'
omr-bypass.eth1=interface
omr-bypass.eth1.id='3'
omr-bypass.eth2=interface
omr-bypass.eth2.id='4'
omr-bypass.eth3=interface
omr-bypass.eth3.id='11'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1200'
omr-bypass.wgwan1=interface
omr-bypass.wgwan1.id='8'
omr-bypass.tun1=interface
omr-bypass.tun1.id='9'
omr-bypass.wan3=interface
omr-bypass.wan3.id='14'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='domain.com'
omr-bypass.@domains[0].interface='eth2'
omr-bypass.@domains[0].note='domain.com'
```

Verify domain is in the ipset list:
ipset list
```
Name: omr_dst_bypass_eth2
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 576
References: 5
Number of entries: 2
Members:
13.57.130.190
18.221.195.49
```

ip rule
```
0:      from all lookup local
0:      from all fwmark 0x1 lookup 100
0:      from all oif eth1 lookup 3
0:      from all oif eth2 lookup 4
0:      from all oif tun0 lookup 1200
0:      from wan1ip lookup 3
0:      from wan2ip lookup 4
0:      from 10.255.255.2 lookup 1200
1:      from all fwmark 0x539 lookup 991337
1:      from all fwmark 0x5396 lookup 6
1:      from all fwmark 0x5393 lookup 3
1:      from all fwmark 0x5394 lookup 4
1:      from all fwmark 0x5391200 lookup 1200
1:      from all fwmark 0x5397 lookup 7
100:    from all lookup lan
10000:  from lanip lookup lan
20000:  from all to lanip/24 lookup lan
32766:  from all lookup main
32767:  from all lookup default
90010:  from all iif lo lookup lan
```
ip r
```
default via 10.255.255.1 dev tun0 
default metric 1 
        nexthop via eth1ip dev eth1 weight 2 
        nexthop via eth2ip dev eth2 weight 1 
default via eth1ip dev eth1 metric 3 
default via eth2ip dev eth2 metric 4 
default via 10.255.255.1 dev tun0 metric 1200 
x.x.x.x/24 via eth2ip dev eth2 
x.x.x.x/24 dev eth2 scope link metric 4 
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
10.255.255.2 dev tun0 scope link metric 1200 
a.a.a.a/10 dev eth1 scope link metric 3 
127.0.0.0/8 dev lo proto static scope link metric 6 
vpsip metric 1 
        nexthop via eth1ip dev eth1 weight 10 
        nexthop via eth2ip dev eth2 weight 1 
```

ip rule show table 4
```

0:      from all oif eth2 lookup 4
0:      from wan2ip lookup 4
1:      from all fwmark 0x5394 lookup 4
```

iptables save
```
-A omr-bypass-local -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A omr-bypass-local -m mark --mark 0x5394 -j RETURN
```

traceroute domain.com (on omr)
```
traceroute to domain.com (18.221.195.49), 30 hops max, 46 byte packets
 1  10.255.255.1 (10.255.255.1)  25.638 ms  45.452 ms  28.889 ms
```

I cleared the table and rebooted omr several times, no entries on the omr-list will go out eth2 any longer, everything jumps to 10.255.255.1 as the first hop. I cleared the list and rebooted and tried again with 1 single entry: domain.com. Broken now.

I see this error in the logs:

`Sun Sep 24 19:50:59 2023 daemon.notice procd: /etc/rc.d/S98omr-bypass: sed: to: No such file or directory`
Not sure if this is related or not.

## Specifications
    OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
    OpenMPTCProuter VPS version: VPS 0.1028
    OpenMPTCProuter VPS provider:
    OpenMPTCProuter platform: x86_64)

ioogithub, 2023-09-25:
Because omr-bypass is not working on my stable omr. I installed a fresh 6.1beta and fresh vps to test.

Result: omr-bypass is still not working.

1. enter a domain "domain.com" on the bypass screen, save and apply
2. verify domain is in the ipset bypass list:

```
Name: omr_dst_bypass_eth2
Type: hash:net
Revision: 7
Header: family inet hashsize 64 maxelem 65536 bucketsize 12 initval 0xebad8330
Size in memory: 792
References: 5
Number of entries: 7
Members:
13.57.130.190
18.221.195.49
```

3. from omr router: `traceroute 18.221.195.49`
4. result: `1  10.255.255.1 (10.255.255.1)  36.109 ms  143.549 ms  35.995 ms`
5. next hop is to vps ip

so out of the box it isn't working? I noticed that ip rule is very different, I don't see the tables that mark the pallets.

Does omr-bypass work in beta 6.1, how does it work if it isn't marking the packets? When it was working I could simply add a domain, eth2 and traceroute from omr and see the first hop to eth2. Now it always going to vps. I would like to get this working. Any suggestions?

ioogithub, 2023-09-25:
here is the log after I enter a domain on the omr-bypass screen:

```
Sep 25 19:18:50 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Sep 25 19:18:53 OpenMPTCProuter user.notice omr-bypass: Reload dnsmasq...
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Sep 25 19:18:53 OpenMPTCProuter daemon.err procd: Got unexpected signal 1
Sep 25 19:18:53 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCP, IP range x.x.x.x -- x.x.x.y, lease time 12h
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5353
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Sep 25 19:18:53 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
```

Ysurac, 2023-09-26:
Did you try a traceroute from a client computer ?

ioogithub, 2023-09-26:
> Did you try a traceroute from a client computer ?
yes a traceroute from the client computer first goes to my lan router, after that every hop is * * * so not that helpful.

traceroute from the omr router should be the most best test right? Any idea what is happening here? omr-bypss worked great for a while on stable but now it never direct packets to the wan interface. I posted the ip rule, ip route, ip set, ip tables etc. everything looks good to me, am i missing something? What can cause omr-bypass to stop working?

It is easy to test with real works test, open netflix on client, it was working before omr-bypass, and i could traceroute from omr to netflix.com and first hop would be wan2, now its 10.255.255.1 and i get the vpn error.

Ysurac, 2023-09-26:
No, traceroute from router doesn't use same rules.
OMR-ByPass is working without any know issues on 0.59.1 release, the problem can be related to a manually added route, a wrong configuration,...
To test you should bypass a domain like ```ifconfig.me```, ```ipconfig.me```,... and check the IP in a navigator.
Check that your DNS is still OpenMPTCProuter IP.

Else, if you don't make a mistake here:
```
ip route get 13.57.130.190 mark 0x5394

13.57.130.190 via eth1wanip dev eth2 table 4 src eth2ip mark 0x5394 uid 0 
    cache 
```
Then there is a problem with the via, should not be related to eth1 at all.

ioogithub, 2023-09-26:
> No, traceroute from router doesn't use same rules.

Okay, when I first added domains, i tested by traceroute domain.com from omr and the first hop was always the wan i specified.

> manually added route, a wrong configuration,...

I don't have any manual routes and I tried to test configuration by booting a clean image but the efi bug in the current image prevents me from testing. Is there a work around for this?

> Check that your DNS is still OpenMPTCProuter IP.

DNS is set to OMR for every client yes assigned by dns on lan router, has not changed.

ip route get 13.57.130.190 mark 0x5394

>13.57.130.190 via eth1wanip dev eth2 table 4 src eth2ip mark 0x5394 uid 0 
    cache 

yes i posted it wrong, it should say eth2wanip, if it says this, it should be okay right?

> Then there is a problem with the via, should not be related to eth1 at all.

I will check this, thanks.

> To test you should bypass a domain like `ifconfig.me`, `ipconfig.me`,... and check the IP in a navigator.

This sounds interesting, a good idea but what do you mean navigator?

ioogithub, 2023-09-26:
okay here is my test, thanks for the update:

1. on pc: `tcpdump -i eth0 -s 0 'dst port 53''
2. load ifconfig.me in a web browser, two results:
- ifconfig.me.
- i.scdn.co.
3. add these two domains to omr-bpass, use eth2
4. close browser, empty cookies
5. open ifconfig.me in a browser.
6. page shows eth2 ip, it worked!
7. test again, change ome-bpass to use eth1
8. close browser, empty cookies
9. open ifconfig.me in a browser
10. website still shows eth2 :(

It didn't update the entry.

uci omr-bypass shows eth1:
```
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].note='test'
omr-bypass.@domains[0].name='ifconfig.me'
omr-bypass.@domains[0].interface='eth1'
omr-bypass.@domains[1]=domains
omr-bypass.@domains[1].name='i.scdn.co'
omr-bypass.@domains[1].note='test'
omr-bypass.@domains[1].interface='eth1'
``

iplist on omr:

```
Name: omr_dst_bypass_eth2
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 1024
References: 5
Number of entries: 10
Members:
23.43.242.9
23.43.242.97
23.43.242.59
23.43.242.8
23.43.242.11
34.160.111.145
23.43.242.16
23.43.242.34
23.43.242.104
23.43.242.35
```

```
Name: omr_dst_bypass_eth1
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 1088
References: 5
Number of entries: 10
Members:
23.43.242.59
23.43.242.9
23.43.242.34
23.43.242.16
23.43.242.104
34.160.111.145
23.43.242.11
23.43.242.35
23.43.242.97
23.43.242.8
```

omr-bypass script did not delete the entries from eth2 ipset list it just added them to eth1. This seems like a bug.

ioogithub, 2023-09-26:
I am testing omr-bpyass further am seeing this in the logs when omr starts, could this be cause inconsistant issues or is this normal:

```
Tue Sep 26 16:33:00 2023 user.notice omr-bypass: Starting OMR-ByPass...
Tue Sep 26 16:32:46 2023 daemon.notice procd: /etc/rc.d/S98omr-bypass: iptables-restore: line 2 failed
Tue Sep 26 16:32:46 2023 daemon.notice procd: /etc/rc.d/S98omr-bypass: iptables-restore: line 2 failed
Tue Sep 26 16:32:46 2023 daemon.notice procd: /etc/rc.d/S98omr-bypass: iptables-restore: line 2 failed
Tue Sep 26 16:32:46 2023 daemon.notice procd: /etc/rc.d/S98omr-bypass: iptables-restore: line 2 failed
Tue Sep 26 16:32:46 2023 daemon.notice procd: /etc/rc.d/S98omr-bypass: iptables-restore: line 2 failed
Tue Sep 26 16:32:46 2023 daemon.notice procd: /etc/rc.d/S98omr-bypass: iptables-restore: line 2 failed
Tue Sep 26 16:33:00 2023 user.notice omr-bypass: Starting OMR-ByPass...
Tue Sep 26 16:33:00 2023 daemon.notice procd: /etc/rc.d/S99v2ray: iptables-restore: line 2 failed
Tue Sep 26 16:33:00 2023 daemon.notice procd: /etc/rc.d/S99v2ray: iptables-restore: line 2 failed
Tue Sep 26 16:33:00 2023 daemon.notice procd: /etc/rc.d/S99v2ray: iptables-restore: line 2 failed
Tue Sep 26 16:33:00 2023 daemon.notice procd: /etc/rc.d/S99v2ray: iptables-restore: line 2 failed
Tue Sep 26 16:33:00 2023 daemon.notice procd: /etc/rc.d/S99v2ray: iptables-restore: line 2 failed
```

closed 2024-01-01 by github-actions[bot]